open62541/open62541 #154: additional header

Stasik0, 2015-01-01:
hey gang,

can someone show me how i can put a DateTime into the additionalHeader? I'd like to send some additiona ldata with stateless requests?

jpfr, 2015-01-03:
> Reserved for future use.
Applications that do not understand the header should ignore it

I guess you can put whatever you want into the additionalHeader (encoded as an extensionobject).
Applications will just ignore it.

Otherwise, UA_ResponseHeader already contains a ```datetime``` field.
For reads, UA_ReadResponse returns a UA_DataValue for every read. There you can also put timestamp information down to picosecond precision.

Stasik0, 2015-01-03:
Yea, i need a second timestamp... Searching for a code snipplet :)

commit 6a547139 (2015-01-04): adding an additional dateTime header to stateless read requests fixes #154
closed 2015-01-04 by Stasik0