open62541/open62541 #156: Include from C++ fails due to keyword usage

jpfr, 2015-01-10:
We use C++ keywords in the relevant header files.
So g++ just fails when we include our libs somewhere.
Especially, there are function pointers called ```new``` and ```delete``` in the vtable.

Any creative ideas for better names?

Stasik0, 2015-01-12:
`neu` and `loeschen`.... just kidding.. maybe something with underscores? are keywords case-sensitive?

FlorianPalm, 2015-01-12:
What about "create" and "destroy", would be delphi style.

Stasik0, 2015-01-12:
are "alloc" and "free" also reserved? they are actually no keywords, just stdc function names?

uleon, 2015-01-13:
"new" and "delete" are keywords that are new to c++. "alloc" and "free" aren't keywords but c-function-names, so they could be candidates. Personally I'd prefer "create" and "destroy" as proposed by @FlorianPalm

jpfr, 2015-02-25:
Solved on master.

We can have sth like

```c++
#include <iostream>

#include "ua_server.h"
#include "logger_stdout.h"
#include "networklayer_tcp.h"

using namespace std;

int main()
{
    cout << "Hello World!" << endl;

    UA_Server *server = UA_Server_new();
    UA_Server_addNetworkLayer(server, ServerNetworkLayerTCP_new(UA_ConnectionConfig_standard, 16664));

    //add a node to the adresspace
    UA_Int32 *myInteger = UA_Int32_new();
    *myInteger = 42;
    UA_Variant *myIntegerVariant = UA_Variant_new();
    UA_Variant_setValue(myIntegerVariant, myInteger, UA_TYPES_INT32);
    UA_QualifiedName myIntegerName;
    UA_QUALIFIEDNAME_ASSIGN(myIntegerName, "the answer");
    UA_NodeId nullnode, objects, organizes;
    UA_NODEID_ASSIGN(nullnode, 0, 0);
    UA_NODEID_ASSIGN(objects, UA_NS0ID_OBJECTSFOLDER, 0);
    UA_NODEID_ASSIGN(organizes, UA_NS0ID_ORGANIZES, 0);

    UA_Server_addVariableNode(server, myIntegerVariant, &nullnode, &myIntegerName,
                              &objects, &organizes);

    UA_Boolean running = UA_TRUE;
    UA_StatusCode retval = UA_Server_run(server, 1, &running);
    UA_Server_delete(server);

    return retval;
}
```

However, the lib itself and the networklayer need to be compiled with gcc and linked in.

closed 2015-02-25 by jpfr

Stasik0, 2015-02-25:
Can you add an example .cpp server? Would be great!

jpfr, 2015-02-25:
done
d913c602e3c5efc3d85d7def41beadf8ffc4d964

destogl, 2015-02-25:
Great! :+1: 
I will look in the next days to implement some example client based on this to use it trough ROS.

jpfr, 2015-02-25:
Hey Denis!

have a look at the "manual" client in ```/examples/client.c```.
There is also work underway to build proper client support in the library.
But that is not very complete so far.

https://github.com/acplt/open62541/blob/client/include/ua_client.h
https://github.com/acplt/open62541/blob/client/src/client/ua_client.c

Stasik0, 2015-02-25:
do we want to add server.cpp to CMake? With an option?

destogl, 2015-02-25:
@jpfr I just looked at it, it looks good... When I implement something, I send my changes/wishes :)