open62541/open62541 #157: Getting rid of 70% binary size

jpfr, 2015-01-11:
Hey team,

much of the binary size comes from the create-new-delete-encode-etc. functions we generate.
Imho we can get "nearly completely" rid of them.

- The definition of the UA_ structs is for free, as the compiler uses them to compile RAM offsets at compile time.
- The structure of the complex datatypes is always the same. It's a list of built-in types/other structs or arrays of them.

```c
#define UA_MAX_MEMBERS 20 // Assumption: The maximum number of members per struct in UA

struct UA_ComplexTypeDescription {
    UA_UInt16 memSize; // size in memory including padding
    UA_Boolean dynamicMembers; // are members of the struct dynamically sized (e.g. array)
    UA_Byte membersSize;
    UA_UInt16 members[UA_MAX_MEMBERS]; // for each entry:
                                       // [0-14: member-index in UA_TYPEDESCRIPTIONS,
                                       // 15: is it an array?]
   // The offset of the members in the struct can be known from the padding rules
   // http://en.wikipedia.org/wiki/Data_structure_alignment#Computing_padding.
}

UA_ComplexTypeDescription * UA_TYPEDESCRIPTIONS = { ... }

// example functions
void * UA_ComplexType_new(void *value, UA_ComplexTypeDescrption *description);
void UA_ComplexType_deleteMembers(void *value, UA_ComplexTypeDescrption *description);
UA_UInt32 UA_ComplexType_calcSizeBinary(void *value, UA_ComplexTypeDescrption *description);
```

The above UA_ComplexTypeDescription has just **43 Byte per type** and is complete.

The binaries would get a lot smaller, bringing us close to the C-implementations of CoAP.
And since more code fits into the L1-cache we get _faster_ even with an additional layer of indirection for the de/encoding.

What to you think? Please tell me if I'm completely out of whack here. :smirk: 
@ichrispa You have done something similar for the hardware implementation, didn't you?

Stasik0, 2015-01-12:
+1 here, however i am not sure that it will save so much... what is an encoding function? it is just a list of 6-7 (regarding to the struct's) function calls... so we get kinda 7*4 bytes per function... hmmm then there are these safety-macros etc... maybe you are right... go for a prototype! ;)

destogl, 2015-01-12:
Are this also solves #156 ?

Stasik0, 2015-01-12:
as far as i understood #156, the problem there is in the name of vtable entries.... this issue would change the target of the vtable pointer but not its name.... however, i'm not really sure

uleon, 2015-01-13:
@destogl actually this does not solve #156 unless we give up the vtable approach for the 26 something basic types as well, doesn't it?
@jpfr In my opinion we would gain the most benefit from such a scheme if we generated memory layouts for the complex UA_ structs that are identical to the layout on the wire. See http://www.catb.org/esr/structure-packing/ for a how-to on c structure packing.

Stasik0, 2015-01-13:
@uleon we have already considered it, this would make in-situ encoding possible... the problem is the padding of the structs... padding control is compiler-depended...

jpfr, 2015-01-13:
@uleon 's idea is really neat. It would enable zero-copy decoding where just point into the received bytestream.

There is a zero-copy parser generator for protocols defined with ASN.1.
http://ttsiodras.github.io/asn1.html

The only difference is that messages in ASN.1 have a defined size.
This breaks for UA as arrays are dynamically sized.

Imagine you have an array of strings.
So we have a pointer to the first UA_String and a size. But we can't do pointer arithmetic to get to the second element since the strings are dynamically sized. And we cannot just *overlay* a struct to the bytestream since the members after the array would be off.

So we could either do
- Store a pointer into the bytestream and _walk_ the structure every time we need random access to a member (of a member of a ...)
- Have zero-copying only for structs with a fixed size (no pointers inside). Then we could e.g. have UA_ReadRequest point to an array of UA_ReadItems that are still in the original bytestream

Both options should be straightforward to implement once we have those typedescriptions.
Bute the first option seems to be more invasive for the code that actually uses the data.

Additionally we need to:
   1. #pragma pack around the structures that shall be zero-copyable. Otherwise compilers can (and will) add padding
   2. Add a compiler flag IS_LITTLE_ENDIAN (cannot be determined by the preprocessor), so we don't have to walk the bytestream to shuffle bytes if the machine is little endian anyway

jpfr, 2015-01-26:
Status Update:
We now have the python code that writes the structure layouts. See the typedescription branch.

The format for storing structure layouts currently needs exactly **30 bytes per structure**, i.e. 7.5kb for all types.
So we can fit the type handling/encoding code + all type layouts into the L1 cache and have some space left for the bytestream we operate on.

```c
#define UA_MAX_MEMBERS 13 // Maximum number of members per complex type
// Increase max_members if there are user-defined types that need more than 13 members

typedef struct {
    UA_UInt16 memberTypeIndex : 9; ///< Index of the member in the datatypelayout table
    UA_Boolean nameSpaceZero : 1; ///< The type of the member is defined in namespace zero
    UA_Byte padding : 5; ///< How much padding is there before this member element?
    UA_Boolean isArray : 1; ///< The member is an array if the given type
} UA_DataTypeMember;

typedef struct {
    UA_UInt16 memSize; ///< Size of the struct in memory
    UA_Boolean zeroCopyable; ///< Can the type be copied directly off the stream?
    UA_Byte membersSize; ///< How many members does the struct have?
    UA_DataTypeMember members[UA_MAX_MEMBERS];
} UA_DataTypeLayout;
```

The generic methods for new/delete/calcSize/encodeBinary/... will follow in the next days.

Stasik0, 2015-01-26:
@jpfr great progress! Waiting to test the first prototype

jpfr, 2015-01-26:
Ok. I have the _init function for the type descriptions.

Branching is *bad* for pipelining in the processor.
So the idea is to compute an index into a table of functions instead of if-elses or switch.

Hopefully this will be fast!
Hard numbers follow when there are alternatives to measure against..

```c
typedef void (*UA_TypeInit)(void *p);
typedef void (*UA_TypeInitJump)(void *p, UA_UInt16 typeIndex);

static const UA_TypeInit *builtinInit = (UA_TypeInit[])
{(UA_TypeInit)&UA_Boolean_init, (UA_TypeInit)&UA_SByte_init, (UA_TypeInit)&UA_Byte_init,
 (UA_TypeInit)&UA_Int16_init, (UA_TypeInit)&UA_UInt16_init, (UA_TypeInit)&UA_Int32_init,
 (UA_TypeInit)&UA_UInt32_init, (UA_TypeInit)&UA_Int64_init, (UA_TypeInit)&UA_UInt64_init,
 (UA_TypeInit)&UA_Float_init, (UA_TypeInit)&UA_Double_init, (UA_TypeInit)&UA_String_init,
 (UA_TypeInit)&UA_DateTime_init, (UA_TypeInit)&UA_Guid_init, (UA_TypeInit)&UA_ByteString_init,
 (UA_TypeInit)&UA_XmlElement_init, (UA_TypeInit)&UA_NodeId_init, (UA_TypeInit)&UA_ExpandedNodeId_init,
 (UA_TypeInit)&UA_StatusCode_init, (UA_TypeInit)&UA_QualifiedName_init, (UA_TypeInit)&UA_LocalizedText_init,
 (UA_TypeInit)&UA_ExtensionObject_init, (UA_TypeInit)&UA_DataValue_init, (UA_TypeInit)&UA_Variant_init,
 (UA_TypeInit)&UA_DiagnosticInfo_init
};

static inline void initJump(void *p, UA_UInt16 typeIndex) {
    builtinInit[typeIndex](p);
}

static const UA_TypeInitJump *initJumpTable = (UA_TypeInitJump[])
{&initJump, &initJump, &initJump, &initJump, &initJump, &initJump, &initJump, &initJump,
 &initJump, &initJump, &initJump, &initJump, &initJump, &initJump, &initJump, &initJump,
 &initJump, &initJump, &initJump, &initJump, &initJump, &initJump, &initJump, &initJump,
 &initJump, /* 25 times the jump to the builtin function */
 &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init,
 &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init,
 &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init,
 &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init,
 &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init,
 &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init,
 &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init,
 &UA_DataType_init, &UA_DataType_init, &UA_DataType_init, &UA_DataType_init
 /* 39 times the (recursive) jump to the datatype function */ };

void UA_DataType_init(void *p, UA_UInt16 typeIndex) {
    UA_Byte *ptr = (UA_Byte *)p; // for pointer arithmetic

    /* Do not check if the index is a builtin-type here. Builtins will be called
       with their very own _init functions normally. In the off case, that the
       generic function is called with the index of a builtin, their layout
       contains a single member of the builtin type, that will be inited in the
       for loop. */

    const UA_DataTypeLayout *layout = (const UA_DataTypeLayout*)&UA_TYPES[typeIndex];
    for(int i=0;i<layout->membersSize; i++) {
        const UA_DataTypeMember *member = &layout->members[i];
        if(member->isArray) {
            /* Padding contains bit-magic to split into padding before and after
               the length integer */
            ptr += (member->padding >> 3);
            *((UA_Int32*)ptr) = -1;
            ptr += sizeof(UA_Int32) + (member->padding & 0x000fff);
            *((void**)ptr) = UA_NULL;
            ptr += sizeof(void*);
            continue;
        }
        const UA_DataTypeLayout *memberLayout =
            (const UA_DataTypeLayout*)&UA_TYPES[member->memberTypeIndex];
        ptr += member->padding;
        /* If the type is not builtin, we will hit an index > 32 and the call is recursive. */
        UA_Byte jumpIndex = ((member->memberTypeIndex > 24) << 5) + (member->memberTypeIndex & 0xff);
        initJumpTable[jumpIndex](ptr, member->memberTypeIndex);
        ptr += memberLayout->memSize;
    }
}
```

Stasik0, 2015-01-27:
@jpfr why having 39 copies of UA_DataType_init pointers if you already check for `typeIndex > 24`?

jpfr, 2015-01-27:
Yes!
That *could* be the faster version.

There's a difference between branch prediction (do we branch?) and branch **target** prediction (where will we jump to) and their implementations are trade-secrets of the processor vendors.

Normally, we shouldn't care and just take the easiest solution.
But since we will burn a good portion of our cycles in that loop (or similar ones), let's just measure everything and see then. :-)

Stasik0, 2015-01-27:
Okay, good plan! Experiments, like in real science! (http://cdn.meme.am/instances/500x/41601882.jpg)

jpfr, 2015-02-04:
The typedescription branch (https://github.com/acplt/open62541/compare/typedescription?expand=1) contains the latest changes.
The API for type handling and encoding did not change at all. Only the underlying implementation.

All the files in the /src directory and the generated files compile cleanly.
The /src/server directory is not ready yet.

Our lib (without the server part) now has a **stripped binary size of 44kb**. With the server stuff added back in, this will roughly double.

Stasik0, 2015-02-04:
Did you try compiling with -Os?

Wow, this means we could get below 100k for the whole server...

jpfr, 2015-02-04:
With -Os and stripping we are down to 40kb for GCC and 37kb for MSVC.

Anybody willing to turn on the unit tests and see what happens? ;-)

FlorianPalm, 2015-02-04:
Just awesome :+1:

Stasik0, 2015-02-04:
@jpfr did you forget to commit come compile flags? I get smth like this

```
[ 25%] Building C object CMakeFiles/open62541-objects.dir/src/ua_types.c.o
In file included from /home/sten/Desktop/open62541/src/ua_util.h:20:0,
                 from /home/sten/Desktop/open62541/src/ua_types.c:14:
/home/sten/Desktop/open62541/include/ua_types.h:411:5: error: type of bit-field ‘memberTypeIndex’ is a GCC extension [-Werror=pedantic]
     UA_UInt16 memberTypeIndex : 9; ///< Index of the member in the datatypetable
     ^
/home/sten/Desktop/open62541/include/ua_types.h:415:5: error: type of bit-field ‘padding’ is a GCC extension [-Werror=pedantic]
     UA_Byte padding : 5; /**< How much padding is there before this member element? For arrays this
     ^
/home/sten/Desktop/open62541/include/ua_types.h:423:5: error: type of bit-field ‘typeIndex’ is a GCC extension [-Werror=pedantic]
     UA_UInt16 typeIndex : 13; ///< Index of the type in the datatytypetable
     ^
cc1: all warnings being treated as errors
```

this is gcc 4.8.1

jpfr, 2015-02-04:
Hum. The usual differences between gcc 4.8 and 4.9.
Just removed pedantic. Seems to be good for 4.8 now.

Stasik0, 2015-02-05:
@jpfr have a look here https://travis-ci.org/acplt/open62541/builds/49579491#L313 there is a series of these errors ```error: initialized field overwritten``` for mingw32 builds... probably it is because it uses a 4.6 compiler... is it hard to fix?

jpfr, 2015-02-05:
@Stasik0 Um. That seems to be a 4.6 cross-compilation thing. Google didn't help with that message either... Since neither gcc on linux nor MSVC complain, can we just use 4.8 for cross-compilation also?

Just pushed to the branch.
**The relevant test suites ```check_builtin``` and ```check_memory``` now run without errors and without any valgrind messages.**

Now, if you want to help bringing the server-part back in, just say so.
By the way, that's also the chance to make the code nice for primetime.
@Stasik0 is already mentioning something like a "release".

Stasik0, 2015-02-05:
@jpfr it is also a problem with 4.6 on Linux..... hmmmm

Stasik0, 2015-02-05:
the one issue seems to be fixed

Stasik0, 2015-02-05:
@jpfr: I got 1 more problem: 
```
==5094== Using Valgrind-3.7.0 and LibVEX; rerun with -h for copyright info
==5094== Command: ./check_memory
==5094== 
Running suite(s): testMemoryHandling
==5094== Stack overflow in thread 1: can't grow stack to 0xbe6fcfec
==5094== 
==5094== Process terminating with default action of signal 11 (SIGSEGV)
==5094==  Access not within mapped region at address 0xBE6FCFEC
==5094==    at 0x804EA5B: UA_init (ua_types.c:757)
==5094==  If you believe this happened as a result of a stack
==5094==  overflow in your program's main thread (unlikely but
==5094==  possible), you can try to increase the size of the
==5094==  main thread stack using the --main-stacksize= flag.
==5094==  The main thread stack size used in this run was 8388608.
==5094== Stack overflow in thread 1: can't grow stack to 0xbe6fcfe8
==5094== 
```
is there too much alloca?

jpfr, 2015-02-05:
[image]
Wow. That shouldn't have happened.
The condition in the if-loop seems to be broken.

Debug builds are running fine.

jpfr, 2015-02-05:
Fixed for release builds.

jpfr, 2015-02-10:
Ok. I consider the typedescription branch done now.

It produces an executable of 93kb (the statically linked server including everything), passes all tests and has the same functionality as the current master.

Can you have a look? I'd then clean up the commit history for merging.

Stasik0, 2015-02-11:
@jpfr try to turn on stateless extensions, i fixed the server, however have problems with statelessClient, would you have a look into? whats about UDP layer?

jpfr, 2015-02-11:
Hum yeah. I'll have a look.

Stasik0, 2015-02-11:
i see there are much changes in synax, we need a green light from @LEvertz and @FlorianPalm for merging

Stasik0, 2015-02-12:
@jpft hey, we may need to add compiling client in the tests, it is still broken, i can not find out how to fix it ;(

jpfr, 2015-02-12:
Done. Compiling the client enabled benchmarking in the example server.

I am not so happy with the current user API for adding and interacting with nodes (relevant for the last commit).

How about we have
- addVariableNode(nodeid, variant, parent, parentreference, ...)
- addVariableNodeFromType(nodeid, variabletype-nodeid, parent, parentreference, ...)
- addObjectNode(nodeid, objecttype-nodeid, parent, parentreference, ...)
- etc.

What else do we need for interacting with the nodes? Can we afford to hide the actual node structures from userland?

Stasik0, 2015-02-12:
Can we make some universal macros/functions?

FlorianPalm, 2015-02-24:
I have looked into the typedescription branch. I think the effort to update our used function signatures isn't that big. I agree to merge the typedescription branch into master.

jpfr, 2015-02-25:
And done

closed 2015-02-25 by jpfr