open62541/open62541 #158: Alignment problem when compiling for ARM

Stasik0, 2015-01-14:
compiling for RPI gives me a following problem

```
/home/pi/open62541/src/server/ua_services_attribute.c: In function ‘readValue’:
/home/pi/open62541/src/server/ua_services_attribute.c:102:37: error: cast increases required alignment of target type [-Werror=cast-align]
         retval |= UA_Variant_copy(&((const UA_VariableNode *)node)->value, &v->value); // todo: zero-copy
                                     ^
/home/pi/open62541/src/server/ua_services_attribute.c:131:45: error: cast increases required alignment of target type [-Werror=cast-align]
                                           &((const UA_VariableNode *)node)->accessLevel);
                                             ^
/home/pi/open62541/src/server/ua_services_attribute.c:138:45: error: cast increases required alignment of target type [-Werror=cast-align]
                                           &((const UA_VariableNode *)node)->userAccessLevel);
                                             ^
/home/pi/open62541/src/server/ua_services_attribute.c:145:45: error: cast increases required alignment of target type [-Werror=cast-align]
                                           &((const UA_VariableNode *)node)->minimumSamplingInterval);
                                             ^
/home/pi/open62541/src/server/ua_services_attribute.c:152:45: error: cast increases required alignment of target type [-Werror=cast-align]
                                           &((const UA_VariableNode *)node)->historizing);
                                             ^
/home/pi/open62541/src/server/ua_services_attribute.c: In function ‘writeValue’:
/home/pi/open62541/src/server/ua_services_attribute.c:276:34: error: cast increases required alignment of target type [-Werror=cast-align]
             UA_VariableNode_copy((const UA_VariableNode*)node, (UA_VariableNode *)newNode);
                                  ^
/home/pi/open62541/src/server/ua_services_attribute.c:276:64: error: cast increases required alignment of target type [-Werror=cast-align]
             UA_VariableNode_copy((const UA_VariableNode*)node, (UA_VariableNode *)newNode);
                                                                ^
/home/pi/open62541/src/server/ua_services_attribute.c:381:71: error: cast increases required alignment of target type [-Werror=cast-align]
                 retval |= UA_Variant_copy(&writeValue->value.value, &((UA_VariableNode *)newNode)->value); // todo: zero-copy
                                                                       ^
/home/pi/open62541/src/server/ua_services_attribute.c:451:36: error: cast increases required alignment of target type [-Werror=cast-align]
             UA_VariableNode_delete((UA_VariableNode *)newNode);
                                    ^
cc1: all warnings being treated as errors
```

how to solve it? what is it?

jpfr, 2015-01-14:
To enforce standard-compatibility and good practices I added some compiler flags and fixed all the warnings a few commits ago:
```-Wshadow -Wcast-align -Wcast-qual -Wmissing-prototypes -Wstrict-prototypes```

What compiler is cc1?
We compile clean on gcc 4.9 and 4.6, clang 3.5 and MSVC 2013.

As a quick-fix, just remove the -Wcast-align flag..

Stasik0, 2015-01-14:
i have removed the warning to compile...

the error log is from gcc 4.9.1... 

as far as I understood '-Wcast-align' warnings are not possible on x86, therefore it is only visible when compiling or cross-compiling to an alignment sensible platfrom e.g. for ARM

closed 2015-02-21 by jpfr