open62541/open62541 #159: steps towards mingw cross-compiling

Stasik0, 2015-01-15:
hi gang,

i have started to look into cross-compiling windows binaries on the linux machine

jpfr, 2015-01-15:
Hey Sten,
can you post your Toolchain-mingw32.cmake?

jpfr, 2015-01-15:
On my box, I had to do the following change:
```
-set(CMAKE_C_COMPILER /opt/mingw/usr/bin/i686-pc-mingw32-gcc)
-set(CMAKE_CXX_COMPILER /opt/mingw/usr/bin/i686-pc-mingw32-g++)
-set(CMAKE_RC_COMPILER /opt/mingw/usr/bin/i686-pc-mingw32-windres)
-set(CMAKE_FIND_ROOT_PATH /opt/mingw/usr/i686-pc-mingw32)
+set(CMAKE_C_COMPILER i686-w64-mingw32-gcc)
+set(CMAKE_CXX_COMPILER i686-w64-mingw32-g++)
+set(CMAKE_RC_COMPILER i686-w64-mingw32-windres)
+set(CMAKE_FIND_ROOT_PATH i686-w64-mingw32)
```

Stasik0, 2015-01-15:
okay, i got cross compiling working on travis, had to workeround a number of weird bugs, its seems to be a combunation of windows headers and gcc 4.6

redefinitions e.g. in queue.h
https://travis-ci.org/acplt/open62541/builds/47111760#L307
or for "writeWalue" 
https://travis-ci.org/acplt/open62541/builds/47113343#L320
and some signness
https://travis-ci.org/acplt/open62541/builds/47113733#L334
https://travis-ci.org/acplt/open62541/builds/47114421#L334

the caviat is that is compiles now, however networklayer_udp is still broken, could someone make it compile please?
https://github.com/acplt/open62541/blob/master/examples/networklayer_udp.c#L110

commit 761caf15 (2015-01-15): fix #159
closed 2015-03-01 by Stasik0