open62541/open62541 #160: Add client functionality to the lib

jpfr, 2015-01-15:
Hey all,

our example client does a good job but does many things "by hand".
- Handshake, SecureChannel, Session
- Storing the details in a ```ConnectionDetails``` struct

We could pull many of these things into the library as it's pretty generic functionality.
And users are asking about client functionality...

destogl, 2015-01-15:
:+1: I can also help a bit....

jpfr, 2015-01-15:
Hey @destogl,

Jep. Working on C++ interop right now.

Imagine we can fire up a client like this:
```c
UA_Client *c = UA_BinaryClient_new();
UA_Client_connect(c, "opc.tcp://localhost:16664");
....
UA_Client_disconnect(c);
```
what functions do we need to interact with the server?

Stasik0, 2015-01-15:
@jpfr let us start with a simple connect (with predefined no-security profile etc.) an endpoint discovery, browse and read

destogl, 2015-01-15:
I agree with @Stasik0!

So for starters I would like to have following (pseudo code):

connect(endpoint)
objects = client.getObjectsNode
object.getChildren

node = getNode(nodeID) - nodeID in some form
node.readValue
node.setValue

disconnect

After that the keep alive thread should be made to keep Connection and Channel opened....

FlorianPalm, 2015-01-16:
I suggest we should first introduce a service layer similar to the server's one, to build up requests. On top of that we can define a more suitable user API, which orchestrates these requests.

l29ah, 2015-04-03:
Does UA_Client_browse work? Should i add it to the example client?

Stasik0, 2015-04-03:
it should, but is not tested, though... surely you can do it!

jpfr, 2015-04-07:
done

closed 2015-04-07 by jpfr