open62541/open62541 #161: Multythreading compiling problem on Ubuntu 14.04

IronCaterpillar, 2015-01-27:
Hello! I'm trying to build multithreading version and getting a lot of undefined references from urcu(liburcu-dev is installed). Without multithreading it builds ok. Here is the error part of the build log:

*CMakeFiles/open62541-objects.dir/src/server/ua_server.c.o: In function `UA_Server_delete':
/home/ironcaterpillar/src/open62541-master/src/server/ua_server.c:67: undefined reference to `rcu_barrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_server.c.o: In function `UA_Server_new':
/home/ironcaterpillar/src/open62541-master/src/server/ua_server.c:78: undefined reference to `rcu_init_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_server_worker.c.o: In function `workerLoop':
/home/ironcaterpillar/src/open62541-master/src/server/ua_server_worker.c:88: undefined reference to `rcu_register_thread_memb'
/home/ironcaterpillar/src/open62541-master/src/server/ua_server_worker.c:106: undefined reference to `rcu_unregister_thread_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `cds_lfht_new':
/usr/local/include/urcu/rculfhash.h:166: undefined reference to `rcu_flavor_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_lock_memb':
/usr/local/include/urcu/static/urcu.h:222: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_lock_update':
/usr/local/include/urcu/static/urcu.h:201: undefined reference to `rcu_gp_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `smp_mb_slave':
/usr/local/include/urcu/static/urcu.h:107: undefined reference to `rcu_has_sys_membarrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `UA_NodeStore_delete':
/home/ironcaterpillar/src/open62541-master/src/server/ua_nodestore_concurrent.c:134: undefined reference to `call_rcu_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_unlock_memb':
/usr/local/include/urcu/static/urcu.h:254: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `smp_mb_slave':
/usr/local/include/urcu/static/urcu.h:107: undefined reference to `rcu_has_sys_membarrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_unlock_update_and_wakeup':
/usr/local/include/urcu/static/urcu.h:238: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `wake_up_gp':
/usr/local/include/urcu/static/urcu.h:168: undefined reference to `rcu_gp_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_lock_memb':
/usr/local/include/urcu/static/urcu.h:222: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_lock_update':
/usr/local/include/urcu/static/urcu.h:201: undefined reference to `rcu_gp_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `smp_mb_slave':
/usr/local/include/urcu/static/urcu.h:107: undefined reference to `rcu_has_sys_membarrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_unlock_memb':
/usr/local/include/urcu/static/urcu.h:254: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `smp_mb_slave':
/usr/local/include/urcu/static/urcu.h:107: undefined reference to `rcu_has_sys_membarrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_unlock_update_and_wakeup':
/usr/local/include/urcu/static/urcu.h:238: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `wake_up_gp':
/usr/local/include/urcu/static/urcu.h:168: undefined reference to `rcu_gp_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_lock_memb':
/usr/local/include/urcu/static/urcu.h:222: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_lock_update':
/usr/local/include/urcu/static/urcu.h:201: undefined reference to `rcu_gp_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `smp_mb_slave':
/usr/local/include/urcu/static/urcu.h:107: undefined reference to `rcu_has_sys_membarrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_unlock_memb':
/usr/local/include/urcu/static/urcu.h:254: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `smp_mb_slave':
/usr/local/include/urcu/static/urcu.h:107: undefined reference to `rcu_has_sys_membarrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_unlock_update_and_wakeup':
/usr/local/include/urcu/static/urcu.h:238: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `wake_up_gp':
/usr/local/include/urcu/static/urcu.h:168: undefined reference to `rcu_gp_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_lock_memb':
/usr/local/include/urcu/static/urcu.h:222: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_lock_update':
/usr/local/include/urcu/static/urcu.h:201: undefined reference to `rcu_gp_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `smp_mb_slave':
/usr/local/include/urcu/static/urcu.h:107: undefined reference to `rcu_has_sys_membarrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `UA_NodeStore_replace':
/home/ironcaterpillar/src/open62541-master/src/server/ua_nodestore_concurrent.c:316: undefined reference to `call_rcu_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_unlock_memb':
/usr/local/include/urcu/static/urcu.h:254: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `smp_mb_slave':
/usr/local/include/urcu/static/urcu.h:107: undefined reference to `rcu_has_sys_membarrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_unlock_update_and_wakeup':
/usr/local/include/urcu/static/urcu.h:238: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `wake_up_gp':
[... 14 lines cut ...]
/usr/local/include/urcu/static/urcu.h:238: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `wake_up_gp':
/usr/local/include/urcu/static/urcu.h:168: undefined reference to `rcu_gp_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_lock_memb':
/usr/local/include/urcu/static/urcu.h:222: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_lock_update':
/usr/local/include/urcu/static/urcu.h:201: undefined reference to `rcu_gp_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `smp_mb_slave':
/usr/local/include/urcu/static/urcu.h:107: undefined reference to `rcu_has_sys_membarrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_unlock_memb':
/usr/local/include/urcu/static/urcu.h:254: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `smp_mb_slave':
/usr/local/include/urcu/static/urcu.h:107: undefined reference to `rcu_has_sys_membarrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_unlock_update_and_wakeup':
/usr/local/include/urcu/static/urcu.h:238: undefined reference to `rcu_reader_memb'
[... cut ...]
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `smp_mb_slave':
/usr/local/include/urcu/static/urcu.h:107: undefined reference to `rcu_has_sys_membarrier_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `_rcu_read_unlock_update_and_wakeup':
/usr/local/include/urcu/static/urcu.h:238: undefined reference to `rcu_reader_memb'
CMakeFiles/open62541-objects.dir/src/server/ua_nodestore_concurrent.c.o: In function `wake_up_gp':
/usr/local/include/urcu/static/urcu.h:168: undefined reference to `rcu_gp_memb'*

Stasik0, 2015-01-27:
hmm, 14.04 brings just a 0.7 urcu version, try to install a 0.8.5 package of liburcu and liburcu-dev manually (deb links on the bottom):
* i386: https://launchpad.net/ubuntu/+source/liburcu/0.8.5-1ubuntu1/+build/6513816
* amd64: https://launchpad.net/ubuntu/+source/liburcu/0.8.5-1ubuntu1/+build/6513813
then it compiles, however an error is thrown
```
/home/sten/Desktop/open62541/src/server/ua_nodestore_concurrent.c: In function ‘UA_NodeStore_release’:
/home/sten/Desktop/open62541/src/server/ua_nodestore_concurrent.c:97:5: error: comparison of unsigned expression < 0 is always false [-Werror=type-limits]
     if(uatomic_sub_return(&entry->refcount, 1) > 0)
     ^
```
without beeing to deep into multithreading, I guess, it is a macro problem, changing the line 
```
if(uatomic_sub_return(&entry->refcount, 1) > 0)
```
to
```
if(uatomic_add_return(&entry->refcount, -1) > 0)
```
lets the stuff compile and the server seems to work

@jpfr should i commit this workaround into the trunk? it is sematically okay
@jpfr we need to check for an appropriate version of liburcu

IronCaterpillar, 2015-01-27:
Now it compiles correctly. Thank you for your help!

jpfr, 2015-01-27:
@Stasik0 
Which compiler gives the warning? GCC 4.9 compiles fine.
And there should be no comparison with < 0 anywhere near that line.
Besides that, the change looks good.
+1 for just adding it.
+1 for checking the liburcu version.

Stasik0, 2015-01-27:
`gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1)`, i think it something with the macro where sub is converted to add, could you try with 4.8 (use `gcc-4.8`)?

jpfr, 2015-01-27:
Uhm. No error also with gcc-4.8.

```
jpfr@jpfr-VirtualBox:~/software/open62541/build$ gcc-4.8 --version
gcc-4.8 (Ubuntu 4.8.2-19ubuntu1) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

jpfr@jpfr-VirtualBox:~/software/open62541/build$ apt-cache show liburcu-dev
Package: liburcu-dev
Status: install ok installed
Priority: extra
Section: libdevel
Installed-Size: 1102
Maintainer: Jon Bernard <jbernard@debian.org>
Architecture: amd64
Source: liburcu
Version: 0.8.6-1
```

Stasik0, 2015-01-28:
regarding finiding liburcu, does this help? https://github.com/markusa/netsniff-ng_filter/blob/master/src/cmake/modules/FindLibURCU.cmake no idea how to get the version, though

jpfr, 2016-03-05:
This seems resolved.
Please reopen if this is still an active issue to you.

closed 2016-03-05 by jpfr