open62541/open62541 #162: Recurrent Clean-Up in the Server

jpfr, 2015-01-28:
We now have the infrastructure for *timed* and *repeated* work in the server.
When the timer runs out, the work (a function call with some pointer to data) is executed.

Currently we do kick out users whose securechannel or session has timed out.
How about we add one function that is repeated every n seconds that handles all internal cleanups of that sort?

@FlorianPalm You built the *managers* for securechannels and sessions. Is the clean-up as easy as iterating over the linked lists in the managers? What if we add more "state" to the sessions, such as subscriptions?

FlorianPalm, 2015-01-28:
This function could also update the session array in ns0.
In my oppinion yes it is. Subscritption would be cool. This would lead to another managing component within the server which allows clients to subcript variables/nodes, checks time to live of subscriptions etc.

Stasik0, 2015-04-10:
we need to cleanup stale securechannels and sessions at first

jpfr, 2015-04-26:
done.
There is a function UA_Server_cleanup that is called every 10 seconds.

closed 2015-04-26 by jpfr