open62541/open62541 #165: Raspberry Pi - Cross Compiling Environment/Tutorial

Stasik0, 2015-02-05:
I think, we have a consesnsus, that a RPI build with some simple I (cpu termperature) and O (built in leds) is comming in next weeks/months. For this, we need a stable cross-compilation platform.

http://stackoverflow.com/a/19269715/383834

jpfr, 2015-02-25:
What else is there besides adding the compiler to travis and having some ```/examples/server_raspi.c```?

Stasik0, 2015-02-25:
In my oppinion -- nothing.

Stasik0, 2015-02-25:
I will look into cross compilation asap

Stasik0, 2015-02-25:
hmm, all the  compiled files give me a segfault... however, I run them on my OpenELEC and not on raspbian
@FlorianPalm could you try to run every of them on raspbian? http://s000.tinyupload.com/?file_id=36940670328361458070

jpfr, 2015-02-25:
Can you run the debug version in gdb on this distri?

```bash
$ gdb -e exampleServer
# now produce a backtrace after running (and crashing) in gdb
r
bt
```

Stasik0, 2015-02-25:
not on openelec, on raspbian you have got everything...., however I am to lazy to destroy my media center at the moment (got no spare sd-card here)

Stasik0, 2015-03-09:
no tutorial, but compiled sources https://github.com/acplt/open62541/releases

closed 2015-03-09 by Stasik0