open62541/open62541 #166: Is message fragmentation possible?

Stasik0, 2015-02-05:
As @LEvertz pointed out, we may have a problem with the tcp networklayer.

As far as i see, the content of the buffer is directly handed over to the worker(s)
https://github.com/acplt/open62541/blob/master/examples/networklayer_tcp.c#L343
and then it gets into ```UA_Server_processBinaryMessage```.

Now, what happends if only a part of an OPC UA message is delivered by read? Then an error will be triggered here https://github.com/acplt/open62541/blob/e7c21d9a8a1b30a953885f9297f81627f4dabfd4/src/server/ua_server_binary.c#L443 and the message will be dropped.

So, should we intoduce some fragmentation analysis in the network layer? In OV it is solved as follows
*1 the message's header is dedcoded and if the buffer's content is smaller than the announced message length, a counter is increased and the buffer is not cleared (wait for more)
*2  step 1 is repeated for a certain number of steps, than we break up (to cope with faulty messages)

UDP layer should be fine, since no fragmentation is possible there

jpfr, 2015-02-05:
Yes. That's a very good point.
TCP guarantees that content arrives in the correct order.
But packages can be randomly split and merged when we retrieve the buffer from the OS.

If we are already dealing with that kind of buffering, we could also do chunk merging with the same logic.

Stasik0, 2015-02-05:
but no chunking for UDP, righto?

uleon, 2015-02-05:
right :-)

jpfr, 2015-04-10:
resolved by merging the hardening_client branch

closed 2015-04-10 by jpfr