open62541/open62541 #168: Change Node Structs

jpfr, 2015-02-12:
Hey all,

The node structures defined in the XML definition are never actually de/encoded for transportation.
The services all have special structures that reference specific attributes.

So, in the typedescription branch we no longer generate the node structs from xml, but have them defined in a seperate header: https://github.com/acplt/open62541/blob/typedescription/src/server/ua_nodes.h

There are a couple of things we should change (short, mid, longterm). Can we just start collecting them?

- Remove redundant attributes for the variable node. E.g. the valuerank and the array dimension is already stored in the value-variant. So we don't have "multiple sources of truth".
- Add a method-pointer to the methodnode. Otherwise we can't do method call, can we? ;-)
- Replace the writeMask and userWriteMask integers with an internal access management. The masks are then generated on the fly based on the user/session.
- Add the hierarchical reference to the parent as an extra member outside of the references array. (**check before**: is it possible for nodes to have multiple hierarchical parents?)

Stasik0, 2015-04-10:
Whats the status of this one?

jpfr, 2015-04-10:
Let's close this issue.
Every node type has a separate structure that can be changed when a concrete need comes up.

closed 2015-04-10 by jpfr