open62541/open62541 #170: Problem Building open62541

kadomcev, 2015-02-18:
Hi!
I cant compile the project.
sudo apt-get install git build-essential gcc cmake python python-lxml
git clone https://github.com/acplt/open62541.git
cd open62541
mkdir build
cmake -DEXAMPLESERVER=ON ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/open62541
user@iMX6-bld:~/open62541$ make
[  3%] Generating src_generated/ua_transport_generated.c, src_generated/ua_transport_generated.h
Traceback (most recent call last):
  File "/home/user/open62541/tools/generate_builtin.py", line 6, in <module>
    from collections import OrderedDict
ImportError: cannot import name OrderedDict
make[2]: *** [src_generated/ua_transport_generated.c] Error 1
make[1]: *** [CMakeFiles/open62541-objects.dir/all] Error 2
make: *** [all] Error 2

What's the matter?

Stasik0, 2015-02-18:
Try pip install ordereddict or make sure you use python 2.7

kadomcev, 2015-02-19:
I install python 2.7 and have the following problem:

stend@admin /c/open62541
$ mingw32-make.exe
[  4%] Generating src_generated/ua_transport_generated.c, src_generated/ua_trans
port_generated.h
['C:/open62541/tools/generate_builtin.py', '--additional-includes=ua_transport.h
', 'C:/open62541/tools/schema/Custom.Opc.Ua.Transport.bsd', 'C:/open62541/src_ge
nerated/ua_transport']
[  8%] Generating src_generated/ua_types_generated.c, src_generated/ua_types_gen
erated.h
['C:/open62541/tools/generate_builtin.py', '--export-prototypes', 'C:/open62541/
tools/schema/Opc.Ua.Types.bsd', 'C:/open62541/src_generated/ua_types']
[ 12%] Generating src_generated/ua_namespace_0.c, src_generated/ua_namespace_0.h

Scanning dependencies of target open62541-objects
[ 16%] Building C object CMakeFiles/open62541-objects.dir/src/ua_types.c.obj
C:\open62541\src\ua_types.c:271:45: error: 'struct timezone' declared inside par
ameter list [-Werror]
 int gettimeofday(struct timeval *tp, struct timezone *tzp);
                                             ^
C:\open62541\src\ua_types.c:271:45: error: its scope is only this definition or
declaration, which is probably not what you want [-Werror]
C:\open62541\src\ua_types.c:272:45: error: 'struct timezone' declared inside par
ameter list [-Werror]
 int gettimeofday(struct timeval *tp, struct timezone *tzp) {
                                             ^
C:\open62541\src\ua_types.c:272:5: error: conflicting types for 'gettimeofday'
 int gettimeofday(struct timeval *tp, struct timezone *tzp) {
     ^
C:\open62541\src\ua_types.c:271:5: note: previous declaration of 'gettimeofday'
was here
 int gettimeofday(struct timeval *tp, struct timezone *tzp);
     ^
cc1.exe: all warnings being treated as errors
mingw32-make[2]: *** [CMakeFiles/open62541-objects.dir/src/ua_types.c.obj] Error
 1
mingw32-make[1]: *** [CMakeFiles/open62541-objects.dir/all] Error 2
mingw32-make: *** [all] Error 2

stend@admin /c/open62541

Stasik0, 2015-02-19:
It seems to be mingw related... Can you make sure to get the latest mingw version? @jpfr any idea?

uleon, 2015-02-19:
Obviously struct timezone is not declared in any included header. Scanning the net I've found an article which states that this data structure actually is not used in Linux:

http://linux.about.com/od/commands/l/blcmdl2_gettime.htm

You might want to conditionally declare this structure for your particular version of mingw...

Stasik0, 2015-02-19:
@kadomcev okay, it seems to compile now, @uleon was right, it was the missing strcut timezone definition in conjunction with different build configurations - try now please

kadomcev, 2015-02-19:
ok, it works!)
thank you

Stasik0, 2015-02-19:
you are welcome

closed 2015-02-19 by Stasik0