open62541/open62541 #171: Problem server example in win7

kadomcev, 2015-02-19:
Hi all.
When the server is running in Windows XP is no problem.
When the server is running on Windows 7, the client can not connect.

C:\Users\admin\workspace\MyOPC-UA\Debug>MyOPC-UA.exe
Listening for TCP connections on 0.0.0.0:16664
UA_StatusCode retval = 0

UaExpert Client: I try connect to opc.tcp://localhost:16664
22:54:08.670 | Server Node | 1111 | Could not connect to server: BadInvalidArgument.

Stasik0, 2015-02-19:
I can not reproduce it :( windows 7 works for me
[image]

uleon, 2015-02-19:
The output "Listening on 0.0.0.0" seems a little strange. I've got no access to the code right now, but shouldn't this be the ip-address of the interface we are listening at?

I remember some pitfalls with multiple interfaces...

Furthermore I'd like to ask for details of your working / non working configuration such as
- OS: 32 vs. 64 Win 7,
- Toolchain: gnuc vs. Visual Studio
- Libraries: mingw, clib version

jpfr, 2015-02-19:
If the message was "listen on 127.0.0.1", then only a localhost client could connect.
Here, any address under which the server is reachable is valid.

Stasik0, 2015-02-19:
working one: windows 7 64 bit, compiled with mingw32, today's version "2013072200".
server binds on 0.0.0.0:16664, connect works for both opc.tcp://localhost:16664 and opc.tcp://127.0.0.1:16664

kadomcev, 2015-02-20:
Checked on another computer with Win 7, everything works fine.
Maybe the problem antivirus ESET NOD 32 v8.0 or security policies windows?

Stasik0, 2015-02-20:
I'd say it is a firewall blockig the port...

closed 2015-02-21 by Stasik0