open62541/open62541 #172: Problem with nodeId.identifierType = UA_NODEIDTYPE_STRING

kadomcev, 2015-02-20:
Hi all.
Trying to change the node type to a string.

    UA_VariableNode *tmpNode = UA_VariableNode_new();
    sprintf(str,"Byte");
    UA_QualifiedName_copycstring(str,&tmpNode->browseName);
    UA_LocalizedText_copycstring(str,&tmpNode->displayName);
    UA_LocalizedText_copycstring("", &tmpNode->description);

    tmpNode->nodeId.namespaceIndex = 2;
    char a[] = "Demo";
    tmpNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
    tmpNode->nodeId.identifier.string.data = (UA_Byte*)a;
    tmpNode->nodeId.identifier.string.length = strlen(a);;
    tmpNode->nodeClass = UA_NODECLASS_VARIABLE;

    tmpNode->value.vt = &UA_TYPES[UA_INT32];
    tmpNode->value.storage.data.dataPtr = &data;
    tmpNode->value.storageType = UA_VARIANT_DATA_NODELETE;
    tmpNode->value.storage.data.arrayLength = 1;

    UA_Server_addNode(server, (const UA_Node**)&tmpNode,
                      &UA_EXPANDEDNODEIDS[UA_OBJECTSFOLDER],
                      &UA_NODEIDS[UA_HASCOMPONENT]);

UaExpert Client:
Listening for TCP connections on 0.0.0.0:16664
SL_processMessage - unknown request, namespace=0, request=787
The message size was not as announced or the message could not be processed, ski
pping to the end of the message.

Node does not appear on the server, prompt please, what is the problem?

jpfr, 2015-02-20:
Hey @kadomcev ,

try the current master. I fixed a bug related to nodeid handling.
This works for me:
```c
    UA_VariableNode *tmpNode = UA_VariableNode_new();
    char str[15];
    sprintf(str,"Byte");
    UA_QualifiedName_copycstring(str,&tmpNode->browseName);
    UA_LocalizedText_copycstring(str,&tmpNode->displayName);

    tmpNode->nodeId.namespaceIndex = 2;
    UA_String_copycstring("Demo", &tmpNode->nodeId.identifier.string);
    tmpNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
    tmpNode->nodeClass = UA_NODECLASS_VARIABLE;

    tmpNode->value.vt = &UA_TYPES[UA_INT32];
    tmpNode->value.storage.data.dataPtr = myInteger;
    tmpNode->value.storageType = UA_VARIANT_DATA_NODELETE;
    tmpNode->value.storage.data.arrayLength = 1;

    UA_Server_addNode(server, (const UA_Node**)&tmpNode,
                      &UA_EXPANDEDNODEIDS[UA_OBJECTSFOLDER],
                      &UA_NODEIDS[UA_ORGANIZES]);
```

kadomcev, 2015-02-20:
Well, it works!)
Q: Why not set the DataType, and the Value is not displayed?

	tmpNode->dataType.identifier.numeric = 3;
	tmpNode->dataType.identifierType = UA_NODEIDTYPE_NUMERIC;
	tmpNode->dataType.namespaceIndex = 0;

jpfr, 2015-02-20:
Yes. The data type is set manually in the variant-node.
This will change in the future as we can infer the datatype from the variant already contained in the node...

We can close this issue?

kadomcev, 2015-02-20:
Yes, ok)

closed 2015-02-20 by jpfr