open62541/open62541 #173: Problem server example in win7

kadomcev, 2015-02-23:
Hi all.
I can't solve a problem with the connection to the server example in win 7 Enterprise.
Firewall disabled, port 16664 opened, UaExpert can't connect to server example:
15:34:50.188 | DiscoveryWidget    |                                | Discovery GetEndpoints on opc.tcp://localhost:4840 failed (BadTimeout)
15:34:49.678 | DiscoveryWidget    |                                | Discovery FindServers on opc.tcp://localhost:4840 failed (BadTimeout)
15:28:50.953 | DiscoveryWidget    |                                | Discovery GetEndpoints on opc.tcp://robertstar:16664 failed (BadInvalidArgument)
15:28:50.953 | DiscoveryWidget    |                                | Discovery FindServers on opc.tcp://robertstar:16664 failed (BadInvalidArgument)
15:28:49.300 | DiscoveryWidget    |                                | Discovery GetEndpoints on opc.tcp://localhost:4840 failed (BadTimeout)
15:28:48.798 | DiscoveryWidget    |                                | Discovery FindServers on opc.tcp://localhost:4840 failed (BadTimeout)

TCP client (hercules_3-2-8) can connect to server example.
discovery server can't return endpoint.
what endpoint server example?

However, the UaExpert can connect to the UaCPPServer:
15:34:54.341 | AddressSpaceModel  | UaServerCpp@robertstar - No... | Browse succeeded.
15:34:54.341 | AddressSpaceModel  | UaServerCpp@robertstar - No... | Browse succeeded.
15:34:54.325 | Server Node        | UaServerCpp@robertstar - No... | Revised values: SessionTimeout=1200000, SecureChannelLifetime=3600000
15:34:54.325 | Server Node        | UaServerCpp@robertstar - No... | Successfully connected UA server.
15:34:54.325 | Server Node        | UaServerCpp@robertstar - No... | Connection status of server 'UaServerCpp@robertstar - None - None (uatcp-uasc-uabinary)' changed to 'Connected'.
15:34:54.325 | AddressSpaceModel  | UaServerCpp@robertstar - No... | Registered for ModelChangeEvents
15:34:53.902 | Server Node        | UaServerCpp@robertstar - No... | ApplicationUri: 'urn:robertstar:UnifiedAutomation:UaServerCpp'
15:34:53.902 | Server Node        | UaServerCpp@robertstar - No... | Found security policy 'http://opcfoundation.org/UA/SecurityPolicy#None'
15:34:53.871 | Server Node        | UaServerCpp@robertstar - No... | Found endpoint 'opc.tcp://robertstar:48010'
15:34:52.669 | DiscoveryWidget    |                                | Adding Url opc.tcp://robertstar:48010
15:34:52.669 | DiscoveryWidget    |                                | Adding Url opc.tcp://robertstar:48010
15:34:52.669 | DiscoveryWidget    |                                | Adding Url opc.tcp://robertstar:48010
15:34:52.669 | DiscoveryWidget    |                                | Adding Url opc.tcp://robertstar:48010
15:34:52.669 | DiscoveryWidget    |                                | Adding Url opc.tcp://robertstar:48010
15:34:51.514 | DiscoveryWidget    |                                | Adding Server UaServerCpp@robertstar with URL opc.tcp://robertstar:48010

Stasik0, 2015-02-23:
* is there another win 7 machine that is working?
  * if yes, it is the same client/client version version you are using?
* get wireshark
  * Enable opc ua for port 16664 (Edit->Preferences->Protocols->OpcUa->Add 16664 to the lis)
* try to capture some working/not working combiations (Enter "OPC" in the filter while capturing)
* show the captures to us

HolgerJeromin, 2015-02-23:
capturing localhost on windows is not (easy possible) in wireshark. So perhaps you have to use another computer on your network to be able to capture.

kadomcev, 2015-02-23:
Wireshark capture:
http://s000.tinyupload.com/?file_id=49153932903732361294
http://s000.tinyupload.com/?file_id=66030192159390318316

Stasik0, 2015-02-23:
okay, from a quick look i found 2 differences in OpenSecureChannelResponse:
* we set timestamp to 0 (instead of *_now)
* we set array length to -1 (instead of 0)
fixed it in the master, does it help?

kadomcev, 2015-02-24:
Where this can be change?

kadomcev, 2015-02-24:
find

Stasik0, 2015-02-24:
Just checkout master again

Stasik0, 2015-02-24:
@kadomcev did it help? can i close the issue?

kadomcev, 2015-02-24:
the problem is not solved

Stasik0, 2015-02-24:
Can you make one more wireshark trace of non-working exchange, then I'd compre it once again. Or better can you write exactly what client an version are you using?

kadomcev, 2015-02-24:
I installed Win 7 x86 in Virtualbox, my server run with client UaExpert.
I will try install Win 7 x64 and check how it will work.

Stasik0, 2015-02-24:
Which uaexpert, version please. And a wirehshark dump if possible :)

kadomcev, 2015-02-24:
Version 1.3.1 206

kadomcev, 2015-02-24:
wirehshark dump later

Stasik0, 2015-02-24:
it seems to be a problem with the newer UaExpert version

Stasik0, 2015-02-24:
closing and moving to #176

closed 2015-02-24 by Stasik0