open62541/open62541 #174: Add nodes

kadomcev, 2015-02-24:
How I can add nodes in my folder?	
// My Folder

	UA_ObjectNode *root = UA_ObjectNode_new();
	root->nodeId = UA_NODEIDS[UA_ROOTFOLDER_NS0];
	root->nodeClass = UA_NODECLASS_OBJECT;
	UA_QUALIFIEDNAME_STATIC(root->browseName, "My Folder");
	UA_LOCALIZEDTEXT_STATIC(root->displayName, "My Folder");
	UA_LOCALIZEDTEXT_STATIC(root->description, "My Folder");

	UA_Server_addNode(server, (const UA_Node**)&root,
					  &UA_EXPANDEDNODEIDS[UA_OBJECTSFOLDER],
					  &UA_NODEIDS[UA_ORGANIZES]);

jpfr, 2015-02-24:
Have a look at the example server starting at line 74: https://github.com/acplt/open62541/blob/master/examples/opcuaServer.c

Do you want to add only variables are also more fancy things?

kadomcev, 2015-02-24:
ok)
	UA_VariableNode *tmpNode = UA_VariableNode_new();
	UA_QualifiedName_copycstring("12",&tmpNode->browseName);
	UA_LocalizedText_copycstring("12",&tmpNode->displayName);
	UA_LocalizedText_copycstring("Int", &tmpNode->description);
	tmpNode->nodeId.identifier.numeric = 1;
	tmpNode->nodeClass = UA_NODECLASS_VARIABLE;
	tmpNode->value.vt = &UA_TYPES[UA_INT32];
	tmpNode->value.storage.data.dataPtr = &data;
	tmpNode->value.storageType = UA_VARIANT_DATA_NODELETE;
	tmpNode->value.storage.data.arrayLength = 1;
	UA_Server_addNode(server, (const UA_Node**)&tmpNode,
					  &UA_EXPANDEDNODEIDS[UA_EXPANDEDNODEID],
					  &UA_NODEIDS[UA_HASCOMPONENT]);

Tell me, why DataType is NULL and value not visible?

kadomcev, 2015-02-24:
how can I do random value at node in server?

kadomcev, 2015-02-24:
/* Add Folder 1 */
	UA_ObjectNode *root = UA_ObjectNode_new();
	root->nodeId = UA_NODEIDS[UA_EXPANDEDNODEID];
	root->nodeClass = UA_NODECLASS_OBJECT;
	UA_QUALIFIEDNAME_STATIC(root->browseName, "1");
	UA_LOCALIZEDTEXT_STATIC(root->displayName, "1");
	UA_LOCALIZEDTEXT_STATIC(root->description, "1");
	UA_Server_addNode(server, (const UA_Node**)&root,
					  &UA_EXPANDEDNODEIDS[UA_ROOTFOLDER],
					  &UA_NODEIDS[UA_ORGANIZES]);

	/* Add node in Folder 1 */
	UA_VariableNode *tmpNode = UA_VariableNode_new();
	UA_QualifiedName_copycstring("12",&tmpNode->browseName);
	UA_LocalizedText_copycstring("12",&tmpNode->displayName);
	UA_LocalizedText_copycstring("Int", &tmpNode->description);
	tmpNode->nodeClass = UA_NODECLASS_VARIABLE;
	tmpNode->value.vt = &UA_TYPES[UA_INT32];
	tmpNode->value.storage.data.dataPtr = &data;
	tmpNode->value.storageType = UA_VARIANT_DATA_NODELETE;
	tmpNode->value.storage.data.arrayLength = 4;
	UA_Server_addNode(server, (const UA_Node**)&tmpNode,
					  &UA_EXPANDEDNODEIDS[UA_EXPANDEDNODEID],
					  &UA_NODEIDS[UA_HASCOMPONENT]);

how can I do add another folder and node in root folder?

jpfr, 2015-02-24:
Hello kadomcev,

so you want to have a node that returns a different value for every read?

That's currently difficult to do on the master branch.
And easy to do in the ```typedescription``` branch that will be merge to master (hopefully) soon.

See here: https://github.com/acplt/open62541/blob/typedescription/examples/server_datasource.c#L22

kadomcev, 2015-02-24:
how can I do add another folder and node in root folder?

jpfr, 2015-02-24:
According to convention, you should only add custom nodes to the /objects node. Not the root itself.

Take a look at server.h and how the internal node hierarchy is is built in /src/server/ua_server.c

Especially, there is a function
```c
UA_AddNodesResult UA_Server_addNode(UA_Server *server, const UA_Node **node,
                                    const UA_ExpandedNodeId *parentNodeId,
                                    const UA_NodeId *referenceTypeId);
```

With this function, you can add an object node (UA_ObjectNode_new) to the hierarchy. If you know the nodeid of this new object node (by specifying it before adding it to the server, or by reading the UA_AddNodesResult), you can add whatever variables you want.
That's what parentNodeId of the addNode function is for.

Does that help you?

kadomcev, 2015-02-24:
UA_AddNodesResult returns id of the added node?

kadomcev, 2015-02-24:
How can i do, view node in Data Access View (request 787 ?)

kadomcev, 2015-02-25:
UA_AddNodesResult returns id of the added node?
How can i do, view node in Data Access View (request 787 ?)

Stasik0, 2015-03-01:
relates #194

closed 2015-03-01 by Stasik0