open62541/open62541 #175: deliver generated src, too

HolgerJeromin, 2015-02-24:
Right now, to build the source you have to have cmake and python. 

On a well known Operating System python is not available on every machine. 
I have an idea, to deliver the generated_src in the repository. 
So if someone only want to use this code, he does not have to install the python setup. 

All others could do the generation on request. 
Travis can check the src with the python output (i know a repository with this logic). 
If someone changes the python scripts 
he has to commit the updated generated src, too.

Stasik0, 2015-02-24:
Let me spin the wheel,

* At first we eliminate python and deliver generated src
* As the second step we eliminate cmake and deliver dozens of generated makefiles/visual studio configs
* ...

Delivering everything (including binarys) is a good idea, however, imho, we should not be store the stuff in our main repo in order not to have double-maintenance.

@HolgerJeromin why not pushing generated stuff and binaries for linux/windows (that are already there) to open62541-www repo?

uleon, 2015-02-24:
@Stasik0: I do not understand why storing the generated stuff in our main repo would be double maintenance?

Stasik0, 2015-02-24:
@uleon hmm, I just thought about someone changing the script... or manually chaing some generated files? how will this work be? Will we have to remove the dependency between the py scripts and the .c files? or are the changes only adding stuff and changing travis script?

HolgerJeromin, 2015-02-24:
If someone changes the generated file he will be "stopped" by travis :-)
Right now he is stopped by cmake who is overwriting his changes on build.

If he (or she :-) change the script he has to commit the generated files, too (and travis would be happy). So no real change for people with python installed. 

We will have to make the python calls optional (similar to ENABLE_SELFSIGNED), add stuff and change travis.
As a extra magic we perhaps can activate the UPDATE_GENERATED_SRC setting with cmake if python is available

Stasik0, 2015-02-24:
somehow i do not like the increasing amount of magic... :( i do not also see why pyhton should be elminiated, since users will have to install cmake anyways....

however, it the majority likes the idea, I'd go for it... if we are going to do it, I'd vote for checking for pyhon in cmake and doing optional calls... who wants to be assigned with that?

jpfr, 2015-02-24:
Have you seen #150?

It should be straighforward to merge the source files (and the internal headers) to a single ```open62541.c``` and the header files in /include to ```open62541.h```. That removes **all** the compilation/cmake/python frictions for users.

There's a bit of fuzzing around with #includes, but nothing that a regex and a few lines of python can't handle.
If we add a compile target that creates open62541.c/.h, we can generate a current version for downloading similar to the doxygen generation.

jpfr, 2015-04-06:
solved

closed 2015-04-06 by jpfr