open62541/open62541 #176: Server broken for latest version of UaExpert (v 1.3.1 206)

Stasik0, 2015-02-24:
Hey there,

it seems that problems reported in #173 are not related to windows.

I have tested our server with the latest v 1.3.1 206 client and it cannot discover endpoints or connect.
here a list

* Windows 7 64bit - UaExpert 1.3.1 206 - broken
* Ubuntu 32bit - UaExpert 1.3.1 206 - broken

201 version was working flawlessly.

WireShark shows that the conversation ends after OpenSecureChannelResponse, I have started to work on it yesterday 11e0c632aa6fd16d18257ab0f8dada708dd21716, however it does not work yet.

@jpfr could you maybe just look into a conformance test? There is probably some trivially wrong field.

kadomcev, 2015-02-24:
where you can download UaExpert 1.3.1 201 ?

Stasik0, 2015-02-24:
@kadomcev I just increased the token lifetime: af0e90b20575056b1a7d6ea7a07826dcc86d1591, it seems to work

checkout latest master and try out

kadomcev, 2015-02-24:
-#define TOKENLIFETIME 10000
+#define TOKENLIFETIME 600000

Confirm that the problem is solved in Win 7 x64, everything works)

Stasik0, 2015-02-24:
@jpfr @FlorianPalm is this client behaviour standard-conformant?

FlorianPalm, 2015-02-24:
I do not know anything about a lower boundary for the TokenLifetime. Perhaps the client does not connect because the revisedLifetime is smaller than the requestedLifetime. 
Try #define TOKENLIFETIME 599999 and check if the client still does not connect.

Stasik0, 2015-02-24:
yes, exactly, you need to have at least the requested lifetime, shoule i just add some zeros to make sure we cover more clients?

FlorianPalm, 2015-02-24:
yes, I am okay with that.

Stasik0, 2015-02-24:
or mayb sometime later..

closed 2015-02-24 by Stasik0