open62541/open62541 #181: Problem with attributes node

kadomcev, 2015-02-25:
Why is in attributes node not set DataType?

jpfr, 2015-02-25:
Sorry, I didn't get that.
Do you have some more context?

kadomcev, 2015-02-25:
UaExpert -> node -> Attributes -> DataType not set	

       /* Add Folder 1 */
	UA_ObjectNode *root = UA_ObjectNode_new();
	root->nodeId = UA_NODEIDS[UA_EXPANDEDNODEID];
	root->nodeClass = UA_NODECLASS_OBJECT;
	UA_QUALIFIEDNAME_STATIC(root->browseName, "ADE7878");
	UA_LOCALIZEDTEXT_STATIC(root->displayName, "ADE7878");
	UA_LOCALIZEDTEXT_STATIC(root->description, "ADE7878");
	UA_Server_addNode(server, (const UA_Node**)&root,
					  	  	&UA_EXPANDEDNODEIDS[UA_OBJECTSFOLDER],
							&UA_NODEIDS[UA_ORGANIZES]);

	/* Add node in Folder 1 */
	UA_VariableNode *tmpNode = UA_VariableNode_new();
	UA_QualifiedName_copycstring("Ua",&tmpNode->browseName);
	UA_LocalizedText_copycstring("Ua",&tmpNode->displayName);
	UA_LocalizedText_copycstring("Int", &tmpNode->description);

	tmpNode->nodeClass = UA_NODECLASS_VARIABLE;

	tmpNode->dataType = UA_NODEIDS[UA_UINT32];  //is that correct?

	tmpNode->value.vt = &UA_TYPES[UA_INT32];
	tmpNode->value.storage.data.dataPtr = &data;
	tmpNode->value.storageType = UA_VARIANT_DATA_NODELETE;
	tmpNode->value.storage.data.arrayLength = 1;
	UA_Server_addNode(server, (const UA_Node**)&tmpNode,
					  	  	&UA_EXPANDEDNODEIDS[UA_EXPANDEDNODEID],
							&UA_NODEIDS[UA_HASCOMPONENT]);

kadomcev, 2015-02-25:
Tell me how to change the icon of the node? (change on to folder)

jpfr, 2015-02-25:
Yes. You found a bug/regression.
The objects node does not have an "organizedby" reference as it should have.
Looking into it...

kadomcev, 2015-02-25:
so what can i do?

jpfr, 2015-02-25:
@kadomcev 
Hmm. I resolved the missing references. But the folder icon is still missing.
I know that old versions of the UA Expert browse for an "Icon" node that we don't have. Don't know if they still do that.

Besides UA Expert showing the generic "object" icon, there should be no difference in functionality.
So this is WONTFIX for now.

About your original questions about attributes. Could you resolve that in the meantime?

Stasik0, 2015-02-25:
@jpfr have you got a server with a proper icon? Should we look into wireshark?

Stasik0, 2015-02-25:
@jpfr it turns out that the icon is based on "HasTypeDefinition" reference with value/or to node "FolderType", it is a shame that I do not know how to add this from userspace atm :/

jpfr, 2015-02-25:
Have a look at the current master.
We do have that reference, e.g. for the folders object. But still no icon...

jpfr, 2015-02-25:
It could be that ua expert expects a more complete objecttype hierarchy.

If you want to add your own references, use ```UA_Server_addReference``` that is visible from userspace.

Stasik0, 2015-02-25:
Something like HasSupertypeDefinition?

Stasik0, 2015-02-25:
Does not help, should we work on covering more on NS0? or just leave it?

Stasik0, 2015-02-26:
I opened a ticket http://forum.unified-automation.com/topic1473.html

Stasik0, 2015-02-27:
dang: http://forum.unified-automation.com/topic1473.html#p2635 @jpfr maybe you can run the conformance tester?

Stasik0, 2015-02-27:
Hey guys, it turns out that the browse service was the problem... ;) can you confirm @kadomcev, @jpfr?

jpfr, 2015-02-27:
Confirmed on Win/MSVC.
Great job!

closed 2015-02-27 by jpfr

kadomcev, 2015-02-27:
Confirmed on Win)

Stasik0, 2015-02-28:
@cacamille3 thanks for your help, it was the browse service :(