open62541/open62541 #182: Cannot read ServerStatus

Stasik0, 2015-02-25:
Seems to be an another regression: while trying to click on Objects->Server->ServerStatus I get ´´´Read attributes of node 'NS0|Numeric|2256' failed [ret = BadNotFound].´´´

Stasik0, 2015-03-01:
@jpfr i have found the problem, while encoding the extension object in the variant, we take nodeid
```
ServerStatusDataType,862,DataType
```
however, UaExpert expects
```
ServerStatusDataType_Encoding_DefaultBinary,864,Object
```
i dirty workarounded an exception for 862 in source, however this is *not a fix*, unfortunately i fail to see, in what cases we need to add +2 and in which we do not have to :( (probably i need to RTFS, but have no standard at hand)

jpfr, 2015-03-01:
That's a good question.

I was under the impression that the nodeid in an extensionobject would always be the "normal" nodeid since binary encoding is triggered by the encoding byte.
[image]

Is it correct that we encode ServerStatusDataType_Encoding_DefaultBinary = 864 **and** set the encoding byte to binary?

Stasik0, 2015-03-01:
this is the current *working* version

[image]

Stasik0, 2015-03-01:
it seems that ```StructuredType``` structs are affected, probably we need to store this info in ```UA_DataType``` and evaluate in ```*_encodeBinary``` accordingly, or do you have it already?

closed 2015-03-15 by Stasik0