open62541/open62541 #184: The problem in new master branch

kadomcev, 2015-02-26:
Function UA_Server_addNode not add node in server.

UA_Int32 *myInteger = UA_Int32_new();
*myInteger = 42;

char str[15];
sprintf(str,"Byte");
UA_VariableNode *tmpNode = UA_VariableNode_new();
UA_QualifiedName_copycstring(str,&tmpNode->browseName);
UA_LocalizedText_copycstring(str,&tmpNode->displayName);

tmpNode->nodeId.namespaceIndex = 2;
UA_String_copycstring("Demo", &tmpNode->nodeId.identifier.string);
tmpNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
tmpNode->nodeClass = UA_NODECLASS_VARIABLE;
tmpNode->value.type = &UA_TYPES[UA_TYPES_INT32];
tmpNode->value.storage.data.dataPtr = myInteger;
tmpNode->value.storageType = UA_VARIANT_DATA_NODELETE;
tmpNode->value.storage.data.arrayLength = 1;

UA_NodeId objects, organizes;
UA_NODEID_ASSIGN(objects, UA_NS0ID_OBJECTSFOLDER, 0);
UA_NODEID_ASSIGN(organizes, UA_NS0ID_ORGANIZES, 0);
UA_Server_addNode(server, (UA_Node*)&tmpNode, (const UA_ExpandedNodeId *)&objects, &organizes);

What is the problem?

kadomcev, 2015-02-26:
statusCode = 805B0000   // The parent node id does not to refer to a valid node.
Listening for TCP connections on 0.0.0.0:16664
testcallback

what did I do wrong?

kadomcev, 2015-02-26:
UA_Int32 *myInteger = UA_Int32_new();
*myInteger = 42;

char str[15];
sprintf(str,"Byte");
UA_VariableNode *tmpNode = UA_VariableNode_new();
UA_QualifiedName_copycstring(str,&tmpNode->browseName);
UA_LocalizedText_copycstring(str,&tmpNode->displayName);

tmpNode->nodeId.namespaceIndex = 2;
UA_String_copycstring("Demo", &tmpNode->nodeId.identifier.string);
tmpNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
tmpNode->nodeClass = UA_NODECLASS_VARIABLE;
tmpNode->value.type = &UA_TYPES[UA_TYPES_INT32];
tmpNode->value.storage.data.dataPtr = myInteger;
tmpNode->value.storageType = UA_VARIANT_DATA_NODELETE;
tmpNode->value.storage.data.arrayLength = 1;

UA_NodeId organizes;
UA_ExpandedNodeId obj;

obj.nodeId.namespaceIndex = 0;
obj.nodeId.identifierType = UA_NODEIDTYPE_NUMERIC;
obj.nodeId.identifier.numeric = UA_NS0ID_OBJECTSFOLDER;
obj.namespaceUri.data = UA_NULL;
obj.namespaceUri.length = -1;
obj.serverIndex = 0;

UA_NODEID_ASSIGN(organizes, UA_NS0ID_ORGANIZES, 0);
UA_AddNodesResult res = UA_Server_addNode(server, (UA_Node*)&tmpNode, (const UA_ExpandedNodeId *)&obj, &organizes);
printf("statusCode = %08X\n",res.statusCode);

statusCode = 804c0000  // The operation could not be processed because all continuation 
                                            points have been allocated.
Listening for TCP connections on 0.0.0.0:16664
testcallback

kadomcev, 2015-02-26:
UA_Int32 *myInteger = UA_Int32_new();
*myInteger = 42;

char str[15];
sprintf(str,"Byte");
UA_VariableNode *tmpNode = UA_VariableNode_new();
UA_QualifiedName_copycstring(str,&tmpNode->browseName);
UA_LocalizedText_copycstring(str,&tmpNode->displayName);

tmpNode->nodeId.namespaceIndex = 2;
UA_String_copycstring("Demo", &tmpNode->nodeId.identifier.string);
tmpNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
tmpNode->nodeClass = UA_NODECLASS_VARIABLE;
tmpNode->value.type = &UA_TYPES[UA_TYPES_INT32];
tmpNode->value.storage.data.dataPtr = myInteger;
tmpNode->value.storageType = UA_VARIANT_DATA_NODELETE;
tmpNode->value.storage.data.arrayLength = 1;

UA_NodeId reftype;
UA_ExpandedNodeId parentnode;

parentnode.nodeId.namespaceIndex = 0;
parentnode.nodeId.identifierType = UA_NODEIDTYPE_NUMERIC;
parentnode.nodeId.identifier.numeric = UA_NS0ID_OBJECTSFOLDER;
parentnode.namespaceUri.data = UA_NULL;
parentnode.namespaceUri.length = -1;
parentnode.serverIndex = 0;

reftype.namespaceIndex = 0;
reftype.identifierType = UA_NODEIDTYPE_NUMERIC;
reftype.identifier.numeric = UA_NS0ID_ORGANIZES;

UA_AddNodesResult res = UA_Server_addNode(server, (UA_Node*)&tmpNode, &parentnode, &reftype);
printf("statusCode = %08X\n",res.statusCode);

statusCode = 80030000 // Not enough memory to complete the operation.
Listening for TCP connections on 0.0.0.0:16664
testcallback

kadomcev, 2015-02-26:
Problem solved)
UA_AddNodesResult res = UA_Server_addNode(server, (UA_Node*)tmpNode,
	                      	  	  	  &UA_EXPANDEDNODEID_STATIC(0, UA_NS0ID_OBJECTSFOLDER),
						  &UA_NODEID_STATIC(0, UA_NS0ID_ORGANIZES));

jpfr, 2015-02-26:
Closed, @kadomcev found a solution.

closed 2015-02-26 by jpfr