open62541/open62541 #187: Crash exampleClient in Windows

kadomcev, 2015-02-27:
Hi all.
When exampleClient run, appears crash.

Settings CMakeLists.txt :
# build example client
option(CLIENT "Build a test client" ON)
if(CLIENT)
	message(STATUS "Extensions: enabling client")
	add_definitions( -DBENCHMARK=1 )
    # the client is built directly with the .o files as it currently uses
    # internal functions that are not exported to the shared lib.
	add_executable(exampleClient $<TARGET_OBJECTS:open62541-objects> examples/client.c)
    if(WIN32)
        target_link_libraries(exampleClient ws2_32 pthread)
    else()
        target_link_libraries(exampleClient rt)
	endif()	
	if(MULTITHREADING)
        target_link_libraries(exampleClient urcu-cds urcu urcu-common pthread)
    endif()
    if(EXTENSION_STATELESS)
        add_executable(statelessClient $<TARGET_OBJECTS:open62541-objects> examples/client_stateless.c)
        if(MULTITHREADING)
            target_link_libraries(statelessClient urcu-cds urcu urcu-common pthread)
        endif()
    endif()
endif()

Code of client.c:
/*
 C ECHO client example using sockets
 This is an example client for internal benchmarks. It works, but is not ready
 for serious use. We do not really check any of the returns from the server.
 */
 
#include <time.h>
#include <stdio.h>
#include <string.h> //strlen
#include <signal.h>
#include <errno.h> // errno, EINTR
#include <fcntl.h> // fcntl

#include "ua_transport_generated.h"
#include "ua_types_encoding_binary.h"
#include "ua_util.h"
#include "ua_types.h"

#include <stdlib.h> // malloc, free
#ifdef _WIN32
#include <malloc.h>
#include <winsock2.h>
#include <sys/types.h>
#include <windows.h>
#include <ws2tcpip.h>
//#define CLOSESOCKET(S) closesocket(S)
#else
#include <sys/select.h> 
#include <netinet/in.h>
#include <netinet/tcp.h>
#include <sys/socketvar.h>
#include <sys/ioctl.h>
#include <unistd.h> // read, write, close
#include <arpa/inet.h>
//#define CLOSESOCKET(S) close(S)
#endif

typedef struct ConnectionInfo {
	UA_Int32 socket;
	UA_UInt32 channelId;
	UA_SequenceHeader sequenceHdr;
[... 554 lines cut ...]
	FILE* fHandle = UA_NULL;
	if (defaultParams) {
		fHandle =  fopen("client.log", &flag);
	}else{
		fHandle =  fopen(argv[3], &flag);
	}
	//header

	UA_Int32 bytesToWrite = sprintf(data, "measurement %s in ms, nodesToRead %d \n", argv[3], nodesToReadSize);
	fwrite(data,1,bytesToWrite,fHandle);
	for(UA_UInt32 i=0;i<tries;i++) {
		bytesToWrite = sprintf(data,"%16.10f \n",timeDiffs[i]);
		fwrite(data,1,bytesToWrite,fHandle);
	}
	fclose(fHandle);

	UA_String_deleteMembers(&reply);
	UA_Array_delete(nodesToRead,&UA_TYPES[UA_TYPES_NODEID], nodesToReadSize);
    UA_free(timeDiffs);

	return 0;
	
}

What is the problem?

kadomcev, 2015-02-27:
$ exampleClient.exe
1st parameter: number of nodes to read
2nd parameter: number of read-tries
3rd parameter: name of the file to save measurement data
4th parameter: 1 = read same node, 0 = read different nodes
5th parameter: ip adress
6th parameter: port
7th parameter: 0=stateful, 1=stateless
8th parameter: 0=tcp, 1=udp (only with stateless calls)

Using default parameters.
sock = 1884
sock = 1872
mean time for handling request:     7.5000000000 ms
received: 66

kadomcev, 2015-02-27:
crash occurs when the code:
	//save to file
	char data[100];
	const char flag = 'a';
	FILE* fHandle = UA_NULL;
	if (defaultParams) {
		fHandle =  fopen("client.log", &flag);
	}else{
		fHandle =  fopen(argv[3], &flag);
	}
	//header

	UA_Int32 bytesToWrite = sprintf(data, "measurement %s in ms, nodesToRead %d \n", argv[3], nodesToReadSize);
	fwrite(data,1,bytesToWrite,fHandle);
	for(UA_UInt32 i=0;i<tries;i++) {
		bytesToWrite = sprintf(data,"%16.10f \n",timeDiffs[i]);
		fwrite(data,1,bytesToWrite,fHandle);
	}
	fclose(fHandle);

kadomcev, 2015-02-27:
problem solved

FlorianPalm, 2015-02-27:
What was the problem?

kadomcev, 2015-02-27:
no problem)

closed 2015-02-27 by Stasik0