open62541/open62541 #188: exampleClient

kadomcev, 2015-02-27:
How get value node from server?
I have nodes on the server with an ID starting from 1900, how i can get node value from server?

Stasik0, 2015-02-27:
yea, you hit the nail again...

the problem is that you have to decode the message manually at the moment, since it is not fully implemented yet. We currently use exampleClient to benchmark the speed fo read requests without actually looking into it.

well you have to cast and decode the ReadResponce accordingly... (starting from here: https://github.com/acplt/open62541/blob/master/examples/client.c#L536). So you need to call UA_ReadResponse_decodeBinary for the buffer and so on.

So you need to contribute.. There was some work on making a better client in the "client" branch here: https://github.com/acplt/open62541/tree/client and in #179... first you need to "restore" the functionality of the current exampleServer and then to extend it...

kadomcev, 2015-02-27:
sendReadRequest(&connectionInfo,nodesToReadSize,nodesToRead);
received = recv(connectionInfo.socket, (char *)reply.data, 2000, 0);
UA_ReadResponse_decodeBinary(reply.data, 2000, MyBuff)   is that correct???

kadomcev, 2015-02-27:
or 
UA_Node node;
UA_ReadResponse_decodeBinary(reply.data, 2000, node)

kadomcev, 2015-02-27:
UA_ReadResponse resp;
UA_ReadResponse_decodeBinary(reply.data, 2000, resp)

Stasik0, 2015-02-27:
similar to what starts from line https://github.com/acplt/open62541/blob/master/examples/client.c#L396

something like

```size_t recvOffset = 0;
UA_ReadResponse resp;
UA_ReadResponse_decodeBinary(&reply, &recvOffset, &resp);```

and then to dig into the respons, it is always youseful to look into wireshark dump aswell to check which value to extract etc

kadomcev, 2015-02-27:
$ exampleClient.exe 10 10 log.txt 0 127.0.0.1 16664 0 0
1st parameter: number of nodes to read = 10
2nd parameter: number of read-tries = 10
3rd parameter: name of the file to save measurement data = 0
recvOffset = 21, resp.resultsSize =-1, resp.diagnosticInfosSize = -1
recvOffset = 21, resp.resultsSize =-1, resp.diagnosticInfosSize = -1
recvOffset = 21, resp.resultsSize =-1, resp.diagnosticInfosSize = -1
recvOffset = 21, resp.resultsSize =-1, resp.diagnosticInfosSize = -1
recvOffset = 21, resp.resultsSize =-1, resp.diagnosticInfosSize = -1
recvOffset = 21, resp.resultsSize =-1, resp.diagnosticInfosSize = -1
recvOffset = 21, resp.resultsSize =-1, resp.diagnosticInfosSize = -1
recvOffset = 21, resp.resultsSize =-1, resp.diagnosticInfosSize = -1
recvOffset = 21, resp.resultsSize =-1, resp.diagnosticInfosSize = -1
recvOffset = 21, resp.resultsSize =-1, resp.diagnosticInfosSize = -1
mean time for handling request:    12.5000000000 ms
received: 120
bytesToWrite = 16, data: nodesToRead 10

What is the problem?

kadomcev, 2015-02-27:
server option udp off
if(ua_client_connectUA(argv[5],atoi(argv[6]),&endpoint,&connectionInfo,stateless,udp) != 0)
Connect to server tcp possibly?

Stasik0, 2015-02-27:
param 7 and 8 have ti be 0, then it is tcp and stateful... not really understanding whats your problem :(

kadomcev, 2015-02-27:
sendReadRequest(&connectionInfo,nodesToReadSize,nodesToRead);
received = recv(connectionInfo.socket, (char *)reply.data, 2000, 0);

2000 - is an arbitrary value?	

size_t recvOffset = 0; 
UA_ReadResponse resp; 
UA_ReadResponse_decodeBinary(&reply, &recvOffset, &resp);
printf("recvOffset = %d, resp.resultsSize =%d, resp.diagnosticInfosSize = %d\n",recvOffset, resp.resultsSize, resp.diagnosticInfosSize);

recvOffset = 21    - The value of received bytes?
resp.resultsSize =-1    data not received?
resp.diagnosticInfosSize = -1 data not received?

Stasik0, 2015-02-27:
2000 is arbitrary

the problem is that there is more stuff like sequence id etc. before the actual ReadResponse, so you have either to decode them as well or to adjust recvOffset accordingly... (have a look in wireshark's view of ReadResponse)... the actual read response starts with offset of about 43 (have a look into wireshark)

jpfr, 2015-02-27:
@kadomcev,
of course you can build a client solution "manually" with the type handling from open62541. However, this will be a lot of work as you need to build securechannel+session+services handling yourself,
The exampleClient we provide is useful for speed tests but will be hard to scale beyond a very limited set of functionality.

As @Stasik0 mentioned, we will eventually support client functionality in open62541.
If you can wait 2,3 weeks things will surely have improved.

Stasik0, 2015-03-31:
the exampleClient is live in the master

closed 2015-03-31 by Stasik0