open62541/open62541 #191: UaExpert > Attribute of node > DataType > not set

kadomcev, 2015-02-27:
/* Add Folder ADE7878 - TYPE:String */
UA_ObjectNode *root = UA_ObjectNode_new();
UA_QUALIFIEDNAME_STATIC(root->browseName, "ADE7878");
UA_LOCALIZEDTEXT_STATIC(root->displayName, "ADE7878");
UA_LOCALIZEDTEXT_STATIC(root->description, "ADE7878");

root->nodeClass = UA_NODECLASS_OBJECT;
root->nodeId.namespaceIndex = 0;
root->nodeId.identifierType = UA_NODEIDTYPE_STRING;
root->eventNotifier = UA_FALSE;
UA_String_copycstring("ADE7878", &root->nodeId.identifier.string);
res = UA_Server_addNode(server, (UA_Node*)root,
							&UA_EXPANDEDNODEID_STATIC(0, UA_NS0ID_OBJECTSFOLDER),
							&UA_NODEID_STATIC(0, UA_NS0ID_ORGANIZES));
							//&UA_NODEID_STATIC(0, UA_NS0ID_FOLDERTYPE));
ADE7878 = res.addedNodeId;
printf("statusCode = %08X\n",res.statusCode);
ADDREFERENCE(ADE7878, UA_NODEID_STATIC(0, UA_NS0ID_HASTYPEDEFINITION),
				 UA_EXPANDEDNODEID_STATIC(0, UA_NS0ID_FOLDERTYPE));

/********************************************************************************************/
/* Add node 'Ua' in folder ADE7878 - TYPE:String */
UA_ExpandedNodeId parentNode;
parentNode.nodeId = ADE7878;
UA_VariableNode *rootNode = UA_VariableNode_new();
UA_QUALIFIEDNAME_STATIC(rootNode->browseName, "Ua");
UA_LOCALIZEDTEXT_STATIC(rootNode->displayName, "Ua");
UA_LOCALIZEDTEXT_STATIC(rootNode->description, "Ua");

rootNode->nodeClass = UA_NODECLASS_VARIABLE;
rootNode->nodeId.namespaceIndex = 0;
rootNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
UA_String_copycstring("ADE7878.Ua", &rootNode->nodeId.identifier.string);

rootNode->value.type = &UA_TYPES[UA_TYPES_INT32];
rootNode->value.storage.data.dataPtr = myInteger;
rootNode->value.storageType = UA_VARIANT_DATA_NODELETE;
rootNode->value.storage.data.arrayLength = 1;
res = UA_Server_addNode(server, (UA_Node*)rootNode,
							&parentNode,
							&UA_NODEID_STATIC(0, UA_NS0ID_HASCOMPONENT));
printf("statusCode = %08X\n",res.statusCode);

kadomcev, 2015-02-27:
[image]

jpfr, 2015-02-27:
Look at how the variant type is defined.
https://github.com/acplt/open62541/blob/master/include/ua_types.h#L220

You do not set the typeId in your current code.
Just use this function to set the value of the variant: ```UA_Variant_setValue(UA_Variant *v, void *p, UA_UInt16 typeIndex);```

kadomcev, 2015-02-27:
/* Add node 'Ua' in folder ADE7878 - TYPE:String */
UA_ExpandedNodeId parentNode;
parentNode.nodeId = ADE7878;
UA_VariableNode *rootNode = UA_VariableNode_new();
UA_QUALIFIEDNAME_STATIC(rootNode->browseName, "Ua");
UA_LOCALIZEDTEXT_STATIC(rootNode->displayName, "Ua");
UA_LOCALIZEDTEXT_STATIC(rootNode->description, "Ua");

rootNode->nodeClass = UA_NODECLASS_VARIABLE;
rootNode->nodeId.namespaceIndex = 0;
rootNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
UA_String_copycstring("ADE7878.Ua", &rootNode->nodeId.identifier.string);

UA_Variant_setValue(&rootNode->value, myInteger, UA_TYPES_INT32);
//rootNode->value.type = &UA_TYPES[UA_TYPES_INT32];
//rootNode->value.storage.data.dataPtr = myInteger;
rootNode->value.storageType = UA_VARIANT_DATA_NODELETE;
rootNode->value.storage.data.arrayLength = 1;
res = UA_Server_addNode(server, (UA_Node*)rootNode,
							&parentNode,
							&UA_NODEID_STATIC(0, UA_NS0ID_HASCOMPONENT));
printf("statusCode = %08X\n",res.statusCode);

changed only identifier

kadomcev, 2015-02-27:
[image]

jpfr, 2015-02-27:
Looks god. NodeId(ns=0, i=6) stands for UA_Int32.
Can we close this?

kadomcev, 2015-02-28:
[image]

kadomcev, 2015-02-28:
UaCPPServer > Attribute of node > DataType > Byte is set

Stasik0, 2015-02-28:
I stared into wireshark just for an hour, and i can not find any difference :( on the wire it looks the same (at least for me), anyone else any idea?

Stasik0, 2015-03-01:
please confirm the fix @kadomcev

closed 2015-03-01 by Stasik0

kadomcev, 2015-03-01:
I confirm, all fine
[image]
However, in the Data Access View - >DataType -> NULL
[image]

Stasik0, 2015-03-01:
And also no value, correct? :(

kadomcev, 2015-03-01:
yes:(

Stasik0, 2015-03-01:
Could you make a new bug and close this one, please? With the screenshot?

kadomcev, 2015-03-01:
ok

commit 9b03e6bd (2015-03-01): extending NS0 with DataTypes, seems to fix #191