open62541/open62541 #195: statusCode = 805E0000: The requested node id is already used by another node

kadomcev, 2015-03-02:
Hi all.
Problem with added node "ADE7878.Uc".

UA_Int32 *myInteger = UA_Int32_new();
*myInteger = 42;
UA_NodeId ADE7878;
UA_AddNodesResult res;

/********************************************************************************************/
/* Add Folder ADE7878 - TYPE:String */
UA_ObjectNode *root = UA_ObjectNode_new();
UA_QUALIFIEDNAME_STATIC(root->browseName, "ADE7878");
UA_LOCALIZEDTEXT_STATIC(root->displayName, "ADE7878");
UA_LOCALIZEDTEXT_STATIC(root->description, "ADE7878");

root->nodeClass = UA_NODECLASS_OBJECT;
root->nodeId.namespaceIndex = 0;
root->nodeId.identifierType = UA_NODEIDTYPE_STRING;
root->eventNotifier = UA_FALSE;
UA_String_copycstring("ADE7878", &root->nodeId.identifier.string);
res = UA_Server_addNode(server, (UA_Node*)root,
							&UA_EXPANDEDNODEID_STATIC(0, UA_NS0ID_OBJECTSFOLDER),
							&UA_NODEID_STATIC(0, UA_NS0ID_ORGANIZES));
ADE7878 = res.addedNodeId;
printf("statusCode = %08X\n",res.statusCode);
ADDREFERENCE(ADE7878, UA_NODEID_STATIC(0, UA_NS0ID_HASTYPEDEFINITION),
				 UA_EXPANDEDNODEID_STATIC(0, UA_NS0ID_FOLDERTYPE));

UA_ExpandedNodeId parentNode;
parentNode.nodeId = ADE7878;
UA_VariableNode *rootNode;

/********************************************************************************************/
/* Add node 'Uc' in folder ADE7878 - TYPE:String */
rootNode = UA_VariableNode_new();
UA_QUALIFIEDNAME_STATIC(rootNode->browseName, "Напряжение фазы C");
UA_LOCALIZEDTEXT_STATIC(rootNode->displayName, "Напряжение фазы C");
UA_LOCALIZEDTEXT_STATIC(rootNode->description, "Напряжение фазы C");
UA_String_copycstring("ADE7878.Uc", &rootNode->nodeId.identifier.string);
UA_Variant_setValue(&rootNode->value, myInteger, UA_TYPES_INT32);

rootNode->nodeClass = UA_NODECLASS_VARIABLE;
rootNode->nodeId.namespaceIndex = 0;
rootNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
rootNode->value.storageType = UA_VARIANT_DATASOURCE;
rootNode->value.storage.data.arrayLength = 1;
rootNode->value.storage.datasource = (UA_VariantDataSource)
{.handle = UA_NULL,
 .read = readTimeData,
 .release = releaseTimeData,
 .write = writeTimeData};
res = UA_Server_addNode(server, (UA_Node*)rootNode,
							&parentNode,
							&UA_NODEID_STATIC(0, UA_NS0ID_HASCOMPONENT));
printf("statusCode = %08X\n",res.statusCode);
[image]
[image]

kadomcev, 2015-03-02:
How get all added nodes in server and print all?

Stasik0, 2015-03-02:
> UA_QUALIFIEDNAME_STATIC(rootNode->browseName, "Íàïðÿæåíèå ôàçû C");
> UA_LOCALIZEDTEXT_STATIC(rootNode->displayName, "Íàïðÿæåíèå ôàçû C");
> UA_LOCALIZEDTEXT_STATIC(rootNode->description, "Íàïðÿæåíèå ôàçû C");

Do you need identifier string? I think for string identifiers you need to der one explicitly, now you have two empty strings as idetentifiers and hence a collision. Id better go for numeric identifiers.

kadomcev, 2015-03-02:
No, the problem arises precisely when adding a node name "ADE7878.Uc"
[image]
in other cases it's OK

Stasik0, 2015-03-02:
hmm, could you commit your code to a fork, and let us look precisely into it? it is difficult to work like this

kadomcev, 2015-03-02:
Ia, Ib, Ic rename to  Ua, Ub, Uc

static UA_NodeId addObjectNode(UA_Server *server, UA_String identifier, UA_String browseName, UA_String displayName, UA_String description, UA_UInt16 parent)
{
	UA_AddNodesResult res;
	UA_NodeId MyNodeId;
	uint8_t nameNode[255];

	memset(&nameNode,0x00,255);
	memcpy(&nameNode, identifier.data, identifier.length);

	UA_ObjectNode *root 			= UA_ObjectNode_new();
	root->browseName.name 			= browseName;
	root->displayName.text 			= displayName;
	root->description.text			= description;
	root->nodeClass 				= UA_NODECLASS_OBJECT;
	root->nodeId.namespaceIndex 	= 0;
	root->nodeId.identifierType 	= UA_NODEIDTYPE_STRING;
	root->nodeId.identifier.string 	= identifier;
	root->eventNotifier 			= UA_FALSE;
	res = UA_Server_addNode(server, (UA_Node*)root,
								&UA_EXPANDEDNODEID_STATIC(0, UA_NS0ID_OBJECTSFOLDER),
								&UA_NODEID_STATIC(0, UA_NS0ID_ORGANIZES));
	MyNodeId = res.addedNodeId;
	printf("addObjectNode: %s, statusCode = %08X\n",nameNode, res.statusCode);
	ADDREFERENCE(MyNodeId, UA_NODEID_STATIC(0, UA_NS0ID_HASTYPEDEFINITION),
	                 UA_EXPANDEDNODEID_STATIC(0, UA_NS0ID_FOLDERTYPE));
	return MyNodeId;
}

int main()
{
	signal(SIGINT, signalHandler);

	UA_Server *server = UA_Server_new();
    UA_Server_addNetworkLayer(server, ServerNetworkLayerTCP_new(UA_ConnectionConfig_standard, 16664));
    UA_WorkItem work = {.type = UA_WORKITEMTYPE_METHODCALL, .work.methodCall = {.method = testCallback, .data = UA_NULL} };
    UA_Server_addRepeatedWorkItem(server, &work, 20000000, UA_NULL); // call every 2 sec

    UA_Int32 *myInteger = UA_Int32_new();
    *myInteger = 42;
    UA_AddNodesResult res;
    UA_NodeId ADE7878, DI, DO, MAX1271, AD5422;

	/********************************************************************************************/
	/* Add Folder ADE7878 - TYPE:String */
    UA_String identifier, browseName, displayName, description;
    UA_String_copycstring("ADE7878",&identifier);
    UA_String_copycstring("ADE7878",&browseName);
    UA_String_copycstring("ADE7878",&displayName);
    UA_String_copycstring("ADE7878",&description);
    ADE7878 = addObjectNode(server,identifier,browseName,displayName,description,UA_NS0ID_OBJECTSFOLDER);

    /********************************************************************************************/
	/* Add Folder DO - TYPE:String */
	UA_String_copycstring("DO",&identifier);
	UA_String_copycstring("DO",&browseName);
	UA_String_copycstring("DO",&displayName);
	UA_String_copycstring("DO",&description);
	DO = addObjectNode(server,identifier,browseName,displayName,description,UA_NS0ID_OBJECTSFOLDER);

    /********************************************************************************************/
	/* Add Folder DI - TYPE:String */
	UA_String_copycstring("DI",&identifier);
	UA_String_copycstring("DI",&browseName);
	UA_String_copycstring("DI",&displayName);
	UA_String_copycstring("DI",&description);
	DI = addObjectNode(server,identifier,browseName,displayName,description,UA_NS0ID_OBJECTSFOLDER);

	/********************************************************************************************/
[... 91 lines cut ...]
	UA_String_copycstring("DO.1", &rootNode->nodeId.identifier.string);

	UA_Variant_setValue(&rootNode->value, myInteger, UA_TYPES_INT32);

	rootNode->nodeClass = UA_NODECLASS_VARIABLE;
	rootNode->nodeId.namespaceIndex = 0;
	rootNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
	rootNode->value.storageType = UA_VARIANT_DATASOURCE;
	rootNode->value.storage.data.arrayLength = 1;
	rootNode->value.storage.datasource = (UA_VariantDataSource)
	{.handle = UA_NULL,
	 .read = readTimeData,
	 .release = releaseTimeData,
	 .write = writeTimeData};
	res = UA_Server_addNode(server, (UA_Node*)rootNode,
								&parentNode,
								&UA_NODEID_STATIC(0, UA_NS0ID_HASCOMPONENT));
	printf("statusCode = %08X\n",res.statusCode);

	/********************************************************************************************/
    UA_StatusCode retval = UA_Server_run(server, 1, &running);
	UA_Server_delete(server);

	return retval;
}

Stasik0, 2015-03-02:
could you some how upload a whole file, i can not compile your code at the moment, sorry

the dataSource decalration change today, would you please adopt it and send the file to me, so I can debug it...

kadomcev, 2015-03-02:
https://github.com/kadomcev/open62541

Stasik0, 2015-03-02:
okay, it seems to fix your problem, we were sometimes comparing strings with integers which is like Germans say "comparing apples to pears"

p.s. i'd suggest to use
'''
COPYNAMES(rootNode, "Uc");
''''
instead of
'''
UA_QUALIFIEDNAME_STATIC(rootNode->browseName, "Uc");
UA_LOCALIZEDTEXT_STATIC(rootNode->displayName, "Uc");
UA_LOCALIZEDTEXT_STATIC(rootNode->description, "Uc");
'''
otherwise you get memory crashes (which are 'maked out' by windows), since stack tries to free a static variable

please confirm wheter it is fixed

kadomcev, 2015-03-03:
Crash server in Windows with my code:

#include <signal.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#define __USE_XOPEN2K
#include <pthread.h>
#include <time.h>

#include "ua_server.h"
#include "logger_stdout.h"
#include "networklayer_tcp.h"
#include "ua_connection.h"
#include "ua_types_generated.h"
#include "ua_util.h"
#include "ua_nodes.h"

#include "ua_types_encoding_binary.h"

UA_Boolean running = UA_TRUE;

#define UA_STRING_STATIC(VARIABLE, STRING) do { \
        VARIABLE.length = sizeof(STRING)-1;     \
        VARIABLE.data   = (UA_Byte *)STRING; } while(0)

/* LocalizedText */
#define UA_LOCALIZEDTEXT_STATIC(VARIABLE, STRING) do { \
        UA_STRING_STATIC(VARIABLE.locale, "en");       \
        UA_STRING_STATIC(VARIABLE.text, STRING); } while(0)

/* QualifiedName */
#define UA_QUALIFIEDNAME_STATIC(VARIABLE, STRING) do { \
        VARIABLE.namespaceIndex = 0;                   \
        UA_STRING_STATIC(VARIABLE.name, STRING); } while(0)

#define ADDREFERENCE(NODEID, REFTYPE_NODEID, TARGET_EXPNODEID) do {     \
        UA_AddReferencesItem item;                                      \
        UA_AddReferencesItem_init(&item);                               \
        item.sourceNodeId = NODEID;                                     \
        item.referenceTypeId = REFTYPE_NODEID;                          \
        item.isForward = UA_TRUE;                                       \
        item.targetNodeId = TARGET_EXPNODEID;                           \
        UA_Server_addReference(server, &item);                          \
    } while(0)

#define COPYNAMES(TARGET, NAME) do {									\
        UA_QualifiedName_copycstring(NAME, &TARGET->browseName);		\
        UA_LocalizedText_copycstring(NAME, &TARGET->displayName);		\
        UA_LocalizedText_copycstring(NAME, &TARGET->description);		\
    } while(0)

static void signalHandler(int sign) {
    printf("Received Ctrl-C\n");
	running = UA_FALSE;
}

UA_Int32 deviceStatus = 0;
pthread_rwlock_t deviceStatusLock;

static void printDeviceStatus(UA_Server *server, void *data) {
    printf("Device Status: %i\n", deviceStatus);
}

static UA_StatusCode readDeviceStatus(const void *handle, UA_DataValue *value) {
    /* In order to reduce blocking time, we could alloc memory for every read
       and return a copy of the data. */
    pthread_rwlock_rdlock(&deviceStatusLock);
    value->value.type = &UA_TYPES[UA_TYPES_INT32];
    value->value.arrayLength = 1;
[... 95 lines cut ...]
	/* Add node 'Uc' in folder ADE7878 - TYPE:String */
	rootNode = UA_VariableNode_new();
	COPYNAMES(rootNode, "Uc");
	UA_String_copycstring("ADE7878.Uc", &rootNode->nodeId.identifier.string);
	UA_Variant_setValue(&rootNode->variable.variant, myInteger, &UA_TYPES[UA_TYPES_INT32]);

	rootNode->nodeId.namespaceIndex = 0;
	rootNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
	rootNode->nodeClass = UA_NODECLASS_VARIABLE;

	rootNode->variable.dataSource = (UA_DataSource)
	{.handle = NULL,
	 .read = readDeviceStatus,
	 .release = releaseDeviceStatus,
	 .write = writeDeviceStatus};
	res = UA_Server_addNode(server, (UA_Node*)rootNode,
								&parentNode,
								&UA_NODEID_STATIC(0, UA_NS0ID_HASCOMPONENT));
	printf("statusCode = %08X\n",res.statusCode);

    UA_StatusCode retval = UA_Server_run(server, 1, &running);
	UA_Server_delete(server);
	pthread_rwlock_destroy(&deviceStatusLock);
	return retval;
}

Stasik0, 2015-03-03:
Hmm, than switch back to the working one... Just to see if the name conflict ist there

kadomcev, 2015-03-03:
No problem, if i added node with type:
/*Add node 'Uc' in folder ADE7878 - TYPE:String */
 rootNode = UA_VariableNode_new();
 COPYNAMES(rootNode, "Uc");
 UA_String_copycstring("ADE7878.Uc", &rootNode->nodeId.identifier.string);
 UA_Variant_setValue(&rootNode->variable.variant, myInteger, &UA_TYPES[UA_TYPES_INT32]);

 rootNode->nodeClass = UA_NODECLASS_VARIABLE;
 rootNode->nodeId.namespaceIndex = 0;
 rootNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
 rootNode->variable.variant.storageType = UA_VARIANT_DATA_NODELETE;
 rootNode->variable.variant.arrayLength = 1;
 res = UA_Server_addNode(server, (UA_Node*)rootNode,
                             &parentNode,
                             &UA_NODEID_STATIC(0, UA_NS0ID_HASCOMPONENT));
 printf("statusCode = %08X\n",res.statusCode);

Problem there is, if I added node with type:
/* Add node 'Uc' in folder ADE7878 - TYPE:String */
rootNode = UA_VariableNode_new();
COPYNAMES(rootNode, "Uc");
UA_String_copycstring("ADE7878.Uc", &rootNode->nodeId.identifier.string);
UA_Variant_setValue(&rootNode->variable.variant, myInteger, &UA_TYPES[UA_TYPES_INT32]);

rootNode->nodeId.namespaceIndex = 0;
rootNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
rootNode->nodeClass = UA_NODECLASS_VARIABLE;

rootNode->variable.dataSource = (UA_DataSource)
{.handle = NULL,
 .read = readDeviceStatus,
 .release = releaseDeviceStatus,
 .write = writeDeviceStatus};
res = UA_Server_addNode(server, (UA_Node*)rootNode,
							&parentNode,
							&UA_NODEID_STATIC(0, UA_NS0ID_HASCOMPONENT));
printf("statusCode = %08X\n",res.statusCode);

Stasik0, 2015-03-03:
So is it now fixed? :)

Stasik0, 2015-03-03:
Coluld you commit the code again, i will look into it in the evening...

kadomcev, 2015-03-03:
I need add node with type:
/* Add node 'Uc' in folder ADE7878 - TYPE:String */
rootNode = UA_VariableNode_new();
COPYNAMES(rootNode, "Uc");
UA_String_copycstring("ADE7878.Uc", &rootNode->nodeId.identifier.string);
UA_Variant_setValue(&rootNode->variable.variant, myInteger, &UA_TYPES[UA_TYPES_INT32]);

rootNode->nodeId.namespaceIndex = 0;
rootNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
rootNode->nodeClass = UA_NODECLASS_VARIABLE;

rootNode->variable.dataSource = (UA_DataSource)
{.handle = NULL,
.read = readDeviceStatus,
.release = releaseDeviceStatus,
.write = writeDeviceStatus};
res = UA_Server_addNode(server, (UA_Node*)rootNode,
&parentNode,
&UA_NODEID_STATIC(0, UA_NS0ID_HASCOMPONENT));
printf("statusCode = %08X\n",res.statusCode);

Problem there is:(

kadomcev, 2015-03-03:
Сan be a problem with the type of node: UA_VariableNode and >variable.dataSource = (UA_DataSource)
{.handle = NULL,
.read = readDeviceStatus,
.release = releaseDeviceStatus,
.write = writeDeviceStatus};

may be lost references between UA_VariableNode and  >variable.dataSource?

kadomcev, 2015-03-03:
server run, when :
{.handle = NULL,
 .read = NULL,
 .release = NULL,
 .write = NULL};   but when reading attributes node "ADE7878.Uc" server crash

Stasik0, 2015-03-03:
Did you see updated datasource code from master? Can you compile ur server against current master?

jpfr, 2015-03-03:
@Stasik0 You had inserted checking for nulled callbacks in the datasource.
Is it okay if we require the presence of .read and .release? There seems to be little use cases that don't have them. And if reading shall be really forbidden, the .read function can always return an error code and do nothing.

@kadomcev 
I think you are calling into a function that doesn't exist (.read = NULL), which would be obvious in a debugger. There are vey good IDEs for windows that support MinGW+Debugging. For example Eclipse with the C/C++ developer extensions.

Otherwise, just use the gdb debugger in the terminal:
- Compile with the debug profile (in cmake)
- Open executable in gdb (gdb <filename>)
- run with the "r" command
- use the server until the crash
- inspect the backtrace with the "bt" command

Stasik0, 2015-03-03:
@jpfr empty write callback will return a standard error, empty should IMHO be possible for read callbacks not allocating memory e.g. to return a builtin value from stack

jpfr, 2015-03-03:
We are talking about the arguments being null or the callback-function being null?

Stasik0, 2015-03-03:
Cb being null like readCallback=UA_NULL

Stasik0, 2015-03-04:
is it still there?

kadomcev, 2015-03-04:
I made by another (UA_Server_addDataSourceVariableNode)

Stasik0, 2015-03-04:
so we can close?

kadomcev, 2015-03-04:
perhaps

Stasik0, 2015-03-04:
kk

closed 2015-03-04 by Stasik0