open62541/open62541 #196: Handler read

kadomcev, 2015-03-02:
How to make one handler to read sources (UA_VariantDataSource) ?

Stasik0, 2015-03-02:
you can pass node as a handle like

```
{.handle = node,
...
```

in the callback you cast `handle` to `(UA_Node*)` and work with it

kadomcev, 2015-03-02:
please, more details)

Stasik0, 2015-03-02:
oh, i notice that you need to know the node in advance, before you add it to server, @jpfr got an idea how to pass node's id or node itself to the callback?

jpfr, 2015-03-02:
You can point to whatever data you like in the handle
If you need the nodeid in your implementation, put it there.

There's a new API for datasources. We handle it at the node level instead of at the variant level.
```c
UA_StatusCode UA_EXPORT UA_Server_addDataSourceVariableNode(UA_Server *server, UA_DataSource dataSource,
                                                            UA_NodeId *nodeId, UA_QualifiedName *browseName,
                                                            const UA_NodeId *parentNodeId,
                                                            const UA_NodeId *referenceTypeId);
```

@Stasik0 U ok with that?

kadomcev, 2015-03-02:
I solved problem:

#define UA_DO1 1
#define UA_DO2 2

UA_Boolean do1, do2, do3, do4, do5, do6, do7, do8;

static UA_StatusCode readNode(const void *handle, UA_VariantData *data) {
	UA_UInt32 nodeId = (UA_UInt32)handle; //Get node ID value
	printf("read nodeId = %d\n",nodeId);

	switch(nodeId){
		case UA_DO1:{
		    pthread_rwlock_rdlock(&mutex);
		    data->arrayLength = 1;
		    data->dataPtr = &do1;
		    data->arrayDimensionsSize = -1;
		    data->arrayDimensions = UA_NULL;
		}
		break;

		case UA_DO2:{
		    pthread_rwlock_rdlock(&mutex);
		    data->arrayLength = 1;
		    data->dataPtr = &do2;
		    data->arrayDimensionsSize = -1;
		    data->arrayDimensions = UA_NULL;
		}
		break;

		default:{
			data->arrayLength = 1;
			data->dataPtr = UA_NULL;
			data->arrayDimensionsSize = -1;
			data->arrayDimensions = UA_NULL;
		}
	}
    return UA_STATUSCODE_GOOD;
}

static void releaseNode(const void *handle, UA_VariantData *data) {
    data->dataPtr = UA_NULL;
    data->arrayLength = -1;
    pthread_rwlock_unlock(&mutex);
}

static UA_StatusCode writeNode(const void *handle, const UA_VariantData *data) {
	UA_UInt32 nodeId = (UA_UInt32)handle; //Get node ID value
	printf("write nodeId = %d\n",nodeId);

	switch(nodeId){
		case UA_DO1:{
		    pthread_rwlock_wrlock(&mutex);
		    if(data->dataPtr != UA_NULL)
		    	do1 = *(UA_Boolean*)data->dataPtr;
		    pthread_rwlock_unlock(&mutex);
		    return UA_STATUSCODE_GOOD;
		}
		break;

		case UA_DO2:{
		    pthread_rwlock_wrlock(&mutex);
		    if(data->dataPtr != UA_NULL)
		    	do2 = *(UA_Boolean*)data->dataPtr;
		    pthread_rwlock_unlock(&mutex);
		    return UA_STATUSCODE_GOOD;
		}
		break;

		default:{
			return UA_STATUSCODE_BADINTERNALERROR;
		}
	}
    return UA_STATUSCODE_GOOD;
}

	/********************************************************************************************/
	/* Add node 'DO.1' in folder DO - TYPE:String */
	parentNode.nodeId = DO;
	rootNode = UA_VariableNode_new();
	UA_QUALIFIEDNAME_STATIC(rootNode->browseName, "1");
	UA_LOCALIZEDTEXT_STATIC(rootNode->displayName, "1");
	UA_LOCALIZEDTEXT_STATIC(rootNode->description, "1");
	UA_String_copycstring("Discrete Output.1", &rootNode->nodeId.identifier.string);
	UA_Variant_setValue(&rootNode->value, myBool, UA_TYPES_BOOLEAN);

	rootNode->nodeClass = UA_NODECLASS_VARIABLE;
	rootNode->nodeId.namespaceIndex = 0;
	rootNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
	rootNode->value.storageType = UA_VARIANT_DATASOURCE;
	rootNode->value.storage.data.arrayLength = 1;
	rootNode->value.storage.datasource = (UA_VariantDataSource)
	{
	 .handle = (void *)UA_DO1,
	 .read = readNode,
	 .release = releaseNode,
	 .write = writeNode};
	res = UA_Server_addNode(server, (UA_Node*)rootNode,
								&parentNode,
								&UA_NODEID_STATIC(0, UA_NS0ID_HASCOMPONENT));
	printf("statusCode = %08X\n",res.statusCode);

	/********************************************************************************************/
	/* Add node 'DO.2' in folder DO - TYPE:String */
	parentNode.nodeId = DO;
	rootNode = UA_VariableNode_new();
	UA_QUALIFIEDNAME_STATIC(rootNode->browseName, "2");
	UA_LOCALIZEDTEXT_STATIC(rootNode->displayName, "2");
	UA_LOCALIZEDTEXT_STATIC(rootNode->description, "2");
	UA_String_copycstring("Discrete Output.2", &rootNode->nodeId.identifier.string);
	UA_Variant_setValue(&rootNode->value, myBool, UA_TYPES_BOOLEAN);

	rootNode->nodeClass = UA_NODECLASS_VARIABLE;
	rootNode->nodeId.namespaceIndex = 0;
	rootNode->nodeId.identifierType = UA_NODEIDTYPE_STRING;
	rootNode->value.storageType = UA_VARIANT_DATASOURCE;
	rootNode->value.storage.data.arrayLength = 1;
	rootNode->value.storage.datasource = (UA_VariantDataSource)
	{
	 .handle = (void *)UA_DO2,
	 .read = readNode,
	 .release = releaseNode,
	 .write = writeNode};
	res = UA_Server_addNode(server, (UA_Node*)rootNode,
								&parentNode,
								&UA_NODEID_STATIC(0, UA_NS0ID_HASCOMPONENT));
	printf("statusCode = %08X\n",res.statusCode);

Stasik0, 2015-03-02:
@jpfr yea, so now you can read the node in the callback, old code seems to work (CurrentTime in the value of node Root.Sever.ServerStatus

closed 2015-03-02 by jpfr