open62541/open62541 #197: Problem: Create libopen62541

kadomcev, 2015-03-03:
Problem: How create libopen62541 for QNX ?

jpfr, 2015-03-03:
Set the right compiler in cmake.

A random google search brought this file:
https://www.mail-archive.com/cmake@cmake.org/msg24398/Toolchain-qnx-640-armle.cmake

And this command to run cmake with it.
cmake -DCMAKE_TOOLCHAIN_FILE=c:\Toolchain-qnx-640-armle.cmake

But since there are a lot of differences between embedded systems, better consult your own documenation.

kadomcev, 2015-03-04:
Problem solved for QNX.
I compile the project for QNX.
I confirm open62541 works in QNX !))

Stasik0, 2015-03-04:
nice! Some special tweaks or just normally compiled?

kadomcev, 2015-03-04:
all the sources compiled in QNX Momentics IDE 4.7 (-std=c99 conflict with libraries IDE, so i correct code to standart c)

Stasik0, 2015-03-04:
I added QNX to open62541.org

closed 2015-03-04 by Stasik0