open62541/open62541 #198: Subscriptions

kadomcev, 2015-03-04:
How to use (check out) functions, are defined in ua_services_subscription.с and ua_services_monitoreditems.c ? : 
Service_CreateSubscription, Service_Publish, Service_SetPublishingMode, Service_CreateMonitoredItems

Stasik0, 2015-03-04:
yea, you need to implement them... ua_server_binary.c has a dispatcher function processMSG where messages are checked in a case-statement. There you need to add the ids for abovementioned servicecalls and implement them ;)

kadomcev, 2015-03-04:
My problem: UaExpert  - > CreateMonitoredItems failed [ret = BadUnknownResponse]

void Service_CreateMonitoredItems(UA_Server *server, UA_Session *session,
                                  const UA_CreateMonitoredItemsRequest *request,
                                  UA_CreateMonitoredItemsResponse *response) {
    if(request->itemsToCreateSize <= 0)
        return;

    //mock up
    response->resultsSize = request->itemsToCreateSize;
	
    //UA_Array_new((void**)&response->results, response->resultsSize, &UA_[UA_MONITOREDITEMCREATERESULT]);
	response->results = UA_Array_new(&UA_TYPES[UA_TYPES_MONITOREDITEMCREATERESULT], response->resultsSize);
    for(int i = 0;request->itemsToCreateSize > 0 && i < request->itemsToCreateSize;i++) {
        //FIXME: search the object in the namespace
        if(request->itemsToCreate[i].itemToMonitor.nodeId.identifier.numeric == 2253) {  // server
            response->results[i].statusCode       = UA_STATUSCODE_GOOD;
            response->results[i].monitoredItemId  = 1;
            response->results[i].revisedSamplingInterval = 4294967295;
            response->results[i].revisedQueueSize = 0;
            continue;
        } else {
            // response->results[i]->statusCode = UA_STATUSCODE_BAD_NODEIDUNKNOWN;
            response->results[i].statusCode = -1;
        }
    }
}
this is correct?

Stasik0, 2015-03-04:
not sure, you have to try. Hang this service in, somewhere here:
https://github.com/acplt/open62541/blob/master/src/server/ua_server_binary.c#L326
and test (Wireshark-time).

kadomcev, 2015-03-04:
case UA_NS0ID_TRANSLATEBROWSEPATHSTONODEIDSREQUEST:
    		INVOKE_SERVICE(TranslateBrowsePathsToNodeIds);
    		break;
					
		case UA_NS0ID_CREATESUBSCRIPTIONREQUEST:{
			printf("CREATESUBSCRIPTIONREQUEST: %d \n",requestType.identifier.numeric - UA_ENCODINGOFFSET_BINARY);
			INVOKE_SERVICE(CreateSubscription);
    		break;
		}
			
		case UA_NS0ID_MONITOREDITEMCREATEREQUEST:{
			printf("MONITOREDITEMCREATEREQUEST: %d \n",requestType.identifier.numeric - UA_ENCODINGOFFSET_BINARY);
			INVOKE_SERVICE(CreateMonitoredItems);
    		break;
		}
			
		case UA_NS0ID_PUBLISHREQUEST:{
			printf("PUBLISHREQUEST: %d \n",requestType.identifier.numeric - UA_ENCODINGOFFSET_BINARY);
			INVOKE_SERVICE(Publish);
    		break;
		}
			
    	default: {
    		printf("SL_processMessage - unknown request, namespace=%d, request=%d\n",
    				requestType.namespaceIndex, requestType.identifier.numeric);
    		UA_RequestHeader  p;
    		UA_ResponseHeader r;
    		if(UA_RequestHeader_decodeBinary(msg, pos, &p))
                return;
    		UA_ResponseHeader_init(&r);
    		init_response_header(&p, &r);
    		r.serviceResult = UA_STATUSCODE_BADSERVICEUNSUPPORTED;
    		ALLOC_MESSAGE(message, UA_ResponseHeader_calcSizeBinary(&r));
    		UA_ResponseHeader_encodeBinary(&r, message, &sendOffset);
    		UA_RequestHeader_deleteMembers(&p);
    		UA_ResponseHeader_deleteMembers(&r);
    		responseType = UA_NS0ID_RESPONSEHEADER + UA_ENCODINGOFFSET_BINARY;
    	}
    	break;
    	}

kadomcev, 2015-03-04:
[image]
what is the problem?

jpfr, 2015-03-04:
Hey @kadomcev 
looking good! It seems you can already negotiate new monitoreditems with the client.

But subscriptions are a tricky business. Now you need to send the actual data.
The client will send you PublishRequest messages.
But you don't answer them right away. They are answered only when 
- there was a change
- a timer ran out

depending on the configuration of the subscription.

You can use the "TimedWork" approach (see ua_server.h and ua_server_worker.c) to delay a function call. And you somehow need a queue of current publishrequests so you can delete them when a subscription is cancelled. And you need to link existing subscriptions to the node. And you need to link the current subscriptions to the user-session....

That's quite a bit of work.
If you are willing to invest the time, we can gladly help with getting the conceptual stuff right first.
By the way. Do you have access to the standard?

kadomcev, 2015-03-04:
You can use the "TimedWork"   - >  UA_Server_addTimedWorkItem ??
you have a working at least one subscription? (example) 
what do you mean? (Do you have access to the standard?)

kadomcev, 2015-03-04:
How create notificationData and add to Service_Publish ??

kadomcev, 2015-03-04:
how i can create link existing subscriptions to the node ??

jpfr, 2015-03-04:
I'll try to answer your questions in order.

1. No we don't have a working subscription. The feature is not implemented.
3. Do you have access to the IEC62541 specification or the standard documents from the OPC Foundation? That is where this feature is defined.
2. All the datatypes are defined in ```/include/ua_types.h``` and ```<build>/src_generated/ua_types_generated.h```. You can do ```UA_NotificationMessage_new()```. The notificationData member is an ExtensionObject and may contain data of type
    - UA_DataChangeNotification
    - UA_EventNotificationList
    - UA_StatusChangeNotification

4. You need to add an appropriate array to the node definition. Its all in ```/src/server/ua_nodes.h```.

As I said. There is quite some work left to do for support subscriptions.

Stasik0, 2015-03-04:
4) or you look into some manager e.g. (https://github.com/acplt/open62541/blob/master/src/server/ua_securechannel_manager.c) and organize subscriptions in a double linked list @jpfr what do you think?

kadomcev, 2015-03-04:
How i get (UA_ByteString body) from my Node?

kadomcev, 2015-03-04:
how can i paste data from my Node into UA_ExtensionObject ?

kadomcev, 2015-03-05:
how i can create delay for function PublishResponse? (example)

kadomcev, 2015-03-06:
I implemented one fixed subscription

Stasik0, 2015-03-06:
is it in your fork? for delays you may look into 'UA_Server_addTimedWorkItem'

kadomcev, 2015-03-06:
not yet

closed 2015-03-11 by Stasik0