open62541/open62541 #199: Problem with ExtensionObject

kadomcev, 2015-03-06:
Type ExtensionObject has field TypeId: UA_NodeId however, it should be UA_ExpandedNodeId

kadomcev, 2015-03-06:
[image]
[image]
[image]
[image]

kadomcev, 2015-03-06:
shift two bytes

jpfr, 2015-03-06:
The standard says it needs to be a normal nodeid.
[image]

That said, I actually would prefer the expandednodeid.
After all, the binary encoding is exactly the same if there is no serverindex and namespaceuri.

kadomcev, 2015-03-06:
i have to change NodeId(4 byte) in order to client read data

kadomcev, 2015-03-06:
otherwise NodeId(2 byte), client not read data(subscription)

jpfr, 2015-03-06:
That's already done internally.
We only store Int32s in the nodeid to simplify the interface. But if the identifier is small enough (and the namespace is zero), we encode it as a 2byte-nodeid or 4byte-nodeid. Decoding does the same.

kadomcev, 2015-03-06:
UaExpert -> not read PublishResponse, if NodeId has 2byte.
If NodeId has 4byte -> all fine.

Stasik0, 2015-03-06:
hmmm, what do we do?

kadomcev, 2015-03-06:
I think, that should be used NodeId with 4byte

jpfr, 2015-03-06:
What is the numeric identifier and namespace in question?

kadomcev, 2015-03-06:
You can see PublishResponse in Wireshark

Stasik0, 2015-03-06:
And reqeust?

jpfr, 2015-03-06:
@kadomcev 
In the screenshots, the nodeid identifier is also different.

The original server sends the identiier 811 (DataChangeNotification_Encoding_DefaultBinary).
You send the identifier 66 (ObjectTypeNode_Encoding_DefaultBinary).

So the difference seems to be not the encoding but the content.
NodeId(ns=0,i=66) can be encoded in 2bytes, but NodeId(ns=0,i=811) cannot.

kadomcev, 2015-03-06:
PublishResponse -> OpcUa Binary Protocol -> OpcUa Service : Encodeable Object -> TypeId : ExpandedNodeId (4bytes)

kadomcev, 2015-03-06:
when NodeId (2bytes)  NodeId =66   -> not work
when NodeId (4bytes)  NodeId =811 -> it work

kadomcev, 2015-03-06:
problem occurs when the two-byte shift

jpfr, 2015-03-06:
Ok.
Go to file ```ua_types_encoding_binary.c```.
Starting at line 290, replace the next two functions with the code below. That replaces the 2 bytes encoding with 4 bytes (or full encoding).

My guess is: The problem are not the 2 bytes. The problem is the 66 identifier.

Try and find out.

```c
size_t UA_NodeId_calcSizeBinary(UA_NodeId const *p) {
    size_t length = 0;
    switch(p->identifierType) {
    case UA_NODEIDTYPE_NUMERIC:
        if(p->identifier.numeric > UA_UINT16_MAX || p->namespaceIndex > UA_BYTE_MAX)
            length = sizeof(UA_Byte) + sizeof(UA_UInt16) + sizeof(UA_UInt32);
        else
            length = 4;  /* UA_NODEIDTYPE_FOURBYTE */
        break;
    case UA_NODEIDTYPE_STRING:
        length = sizeof(UA_Byte) + sizeof(UA_UInt16) + UA_String_calcSizeBinary(&p->identifier.string);
        break;
    case UA_NODEIDTYPE_GUID:
        length = sizeof(UA_Byte) + sizeof(UA_UInt16) + UA_Guid_calcSizeBinary(&p->identifier.guid);
        break;
    case UA_NODEIDTYPE_BYTESTRING:
        length = sizeof(UA_Byte) + sizeof(UA_UInt16) + UA_ByteString_calcSizeBinary(&p->identifier.byteString);
        break;
    default:
        UA_assert(UA_FALSE); // this must never happen
        break;
    }
    return length;
}

UA_StatusCode UA_NodeId_encodeBinary(UA_NodeId const *src, UA_ByteString * dst, size_t *offset) {
    UA_StatusCode retval = UA_STATUSCODE_GOOD;
    // temporary variables for endian-save code
    UA_Byte srcByte;
    UA_UInt16 srcUInt16;
    switch(src->identifierType) {
    case UA_NODEIDTYPE_NUMERIC:
        if(src->identifier.numeric > UA_UINT16_MAX || src->namespaceIndex > UA_BYTE_MAX) {
            srcByte = UA_NODEIDTYPE_NUMERIC;
            retval |= UA_Byte_encodeBinary(&srcByte, dst, offset);
            retval |= UA_UInt16_encodeBinary(&src->namespaceIndex, dst, offset);
            retval |= UA_UInt32_encodeBinary(&src->identifier.numeric, dst, offset);
        } else {
            srcByte = UA_NODEIDTYPE_FOURBYTE;
            retval |= UA_Byte_encodeBinary(&srcByte, dst, offset);
            srcByte = src->namespaceIndex;
            srcUInt16 = src->identifier.numeric;
            retval |= UA_Byte_encodeBinary(&srcByte, dst, offset);
            retval |= UA_UInt16_encodeBinary(&srcUInt16, dst, offset);
        }
        break;
    case UA_NODEIDTYPE_STRING:
        srcByte = UA_NODEIDTYPE_STRING;
        retval |= UA_Byte_encodeBinary(&srcByte, dst, offset);
        retval |= UA_UInt16_encodeBinary(&src->namespaceIndex, dst, offset);
        retval |= UA_String_encodeBinary(&src->identifier.string, dst, offset);
        break;
    case UA_NODEIDTYPE_GUID:
        srcByte = UA_NODEIDTYPE_GUID;
        retval |= UA_Byte_encodeBinary(&srcByte, dst, offset);
        retval |= UA_UInt16_encodeBinary(&src->namespaceIndex, dst, offset);
        retval |= UA_Guid_encodeBinary(&src->identifier.guid, dst, offset);
        break;
    case UA_NODEIDTYPE_BYTESTRING:
        srcByte = UA_NODEIDTYPE_BYTESTRING;
        retval |= UA_Byte_encodeBinary(&srcByte, dst, offset);
        retval |= UA_UInt16_encodeBinary(&src->namespaceIndex, dst, offset);
        retval |= UA_ByteString_encodeBinary(&src->identifier.byteString, dst, offset);
        break;
    default:
        UA_assert(UA_FALSE);
    }
    return retval;
}
```

kadomcev, 2015-03-06:
How then to be if NodeId is less than two bytes ?

jpfr, 2015-03-06:
Decoding is unaffected.
A NodeId never **is** less than two bytes. That's just one **possible encoding** if the identifier is small enough to fit into one byte.

kadomcev, 2015-03-06:
UA_ExtensionObject *exobj = UA_ExtensionObject_new();

        exobj->typeId.nodeId.identifier.numeric = 66;
        exobj->typeId.nodeId.identifierType = UA_NODEIDTYPE_NUMERIC;
        exobj->typeId.nodeId.namespaceIndex = 0;

must be: exobj->typeId.nodeId = MyNode.NodeId       from my server   ????

jpfr, 2015-03-06:
Since the server you tested sends an UA_DataChangeNotification structure.
That seems plausible, doesn't it?

kadomcev, 2015-03-06:
NotificationData: Array of ExtensionObject

kadomcev, 2015-03-06:
UA_ExtensionObject *exobj = UA_ExtensionObject_new();

    exobj->typeId.nodeId.identifier.numeric = 66;
    exobj->typeId.nodeId.identifierType = UA_NODEIDTYPE_NUMERIC;
    exobj->typeId.nodeId.namespaceIndex = 0;
    exobj->encoding = UA_EXTENSIONOBJECT_ENCODINGMASK_BODYISBYTESTRING;
    exobj->body = *message;

    response->notificationMessage.notificationData = exobj;
    response->notificationMessage.notificationDataSize = 1;
    response->results = status;
    response->resultsSize = 1;

jpfr, 2015-03-06:
Yes. But you can't encode whatever you like.
 > The NotificationMessage structure used in the Subscription Service set allows specifying different
types of NotificationData. The NotificationData parameter is an extensible parameter whose
structure depends on the type of Notification being sent.

Allowed are DataChangeNotification, EventNotificationList, StatusChangeNotification.

kadomcev, 2015-03-06:
as should have been done?

kadomcev, 2015-03-06:
I've done each byte as in Wireshark

jpfr, 2015-03-06:
@kadomcev 
You need to encode one of the three types I mentioned above in the extensionobject.
Where does the 66 identifier come from anyway?

```c
UA_DataChangeNotification_encodeBinary(notification, message.data, 0);
UA_ExtensionObject *exobj = UA_ExtensionObject_new();
exobj->typeId.nodeId = UA_TYPES[UA_TYPES_DATACHANGENOTIFICATION].typeId;
exobj->encoding = UA_EXTENSIONOBJECT_ENCODINGMASK_BODYISBYTESTRING;
exobj->body = *message;
```

kadomcev, 2015-03-06:
UA_DataChangeNotification_encodeBinary(notification, message.data, 0);
what's this notification?
what's this message.data?

jpfr, 2015-03-06:
We aim to provide an easy to use solution to our users.
But since subscriptions are currently not supported in open62541t, you will need *a lot* of additional information. You should get access to the standard, as we are unable to reiterate its entire content here.

I am closing this issue, since your question is no longer about extensionobject encoding.
Feel free to open a new issue under an appropriate title.

closed 2015-03-06 by jpfr