open62541/open62541 #201: get NodeId

kadomcev, 2015-03-06:
How can i get NodeId from:  
UA_DataSource varNode ???

jpfr, 2015-03-06:
```c
// assume UA_VariableNode *varNode with
// varNode->variableType == UA_VARIABLENODETYPE_DATASOURCE
UA_DataValue myDataValue;
varNode->variable.dataSource.read(varNode->variable.dataSource.handle, &myDataValue);
UA_NodeId myNode;
UA_NodeId_copy(&myDataValue.value.type->nodeId, myNodeId);
varNode->variable.dataSource.release(varNode->variable.dataSource.handle, &myDataValue);
```

kadomcev, 2015-03-10:
UA_VariableNode *varNode_t = UA_VariableNode_new();
varNode_t->variable.dataSource = (UA_DataSource)
{.handle = (void *)2,
 .read = readDeviceStatus,
 .release = releaseDeviceStatus,
 .write = writeDeviceStatus};
UA_QUALIFIEDNAME_ASSIGN(statusName, "3");
UA_Server_addDataSourceVariableNode(server, varNode, &UA_NODEID_NULL, &statusName,
									&DO,
									&UA_NODEID_STATIC(0, UA_NS0ID_HASCOMPONENT));
UA_DataValue myDataValue;
varNode_t->variable.dataSource.read(varNode_t->variable.dataSource.handle, &myDataValue);
printf("\n Add nodeId: %d \n", myDataValue.value.type->typeId.identifier.numeric);

return: 1  though NodeId.identifier.numeric = 70

closed 2015-04-06 by Stasik0