open62541/open62541 #204: CreateSubscription

kadomcev, 2015-03-10:
Hi all.
How i get DataType of node, when i create subscription?

Stasik0, 2015-03-10:
first you get the node ```UA_Node* node = get_node(..);```, than you look into `node->nodeClass` this is an enum
```
typedef enum { 
    UA_NODECLASS_UNSPECIFIED = 0,
    UA_NODECLASS_OBJECT = 1,
    UA_NODECLASS_VARIABLE = 2,
    UA_NODECLASS_METHOD = 4,
    UA_NODECLASS_OBJECTTYPE = 8,
    UA_NODECLASS_VARIABLETYPE = 16,
    UA_NODECLASS_REFERENCETYPE = 32,
    UA_NODECLASS_DATATYPE = 64,
    UA_NODECLASS_VIEW = 128
} UA_NodeClass;
```
depending on the value you can cast the pointer in an appropriate way e.g.
```
if(node->nodeClass == UA_NODECLASS_VARIABLE){
    UA_VarialbeNode* variable_node = (UA_VarialbeNode*) node;
}
```

kadomcev, 2015-03-10:
get_node(..) where this function?

Stasik0, 2015-03-10:
something like this?
`UA_Node const *node = UA_NodeStore_get(server->nodestore, &nodeId);`

kadomcev, 2015-03-10:
In function Service_CreateMonitoredItems, i can get NodeId, how can i get DataType this of node?

kadomcev, 2015-03-10:
i'll check)

Stasik0, 2015-03-10:
look into the implementation of another services like read: https://github.com/acplt/open62541/blob/master/src/server/ua_services_attribute.c#L18

kadomcev, 2015-03-10:
Why Memory fault?? when : 
UA_Int32 Service_Publish(UA_Server *server, UA_Session *session, 
                                                           const UA_PublishRequest *request,
                                                           UA_PublishResponse *response) {

	UA_SubscriptionAcknowledgement *ttt;
	ttt = request->subscriptionAcknowledgements;
	printf("request->subscriptionAcknowledgements.subscriptionId: %d \n", ttt->subscriptionId);

Stasik0, 2015-03-10:
is subscriptionAcknowledgementsSize zero? is 'ttt==UA_NULL'? get gdb :)

kadomcev, 2015-03-10:
nodes can have the same subscriptionId?

kadomcev, 2015-03-10:
how to generate subscriptionId? function rand or something else?

Stasik0, 2015-03-10:
look into some manager, e.g. https://github.com/acplt/open62541/blob/master/src/server/ua_session_manager.c. You will probably need a list that stores subscriptions. And then increment a running id like in https://github.com/acplt/open62541/blob/master/src/server/ua_session_manager.c#L108

kadomcev, 2015-03-11:
How works subscription when changes the value of the node?

Stasik0, 2015-03-11:
does this help? http://documentation.unified-automation.com/uasdkc/1.4.0/html/L2UaSubscription.html in doubt, you will need to get IEC 62541

Stasik0, 2015-03-17:
hey @kadomcev, how is it goin?

kadomcev, 2015-03-21:
i'm on vacation)

ichrispa, 2015-04-25:
What is the status of the subscription and monitoring services? @kadomcev are you still on this? We have a rather pressing need to get this functionality running. If noone is handling this issue, I might be inclined to get working on it.

jpfr, 2015-04-25:
@ichrispa Cool. Can you add a branch for this functionality?

Imho we should "wet our feet" by adding subscriptions to the client first.
That enables us to make the right design choices for the server..

1. Generate the datatypes (https://github.com/acplt/open62541/blob/master/tools/generate_datatypes.py#L60)
2. Add a couple of service wrappers to the client (https://github.com/acplt/open62541/blob/master/src/client/ua_client.c#L573)
4. Demux responses from the server. Every message could either relate to a recent service call or be a notification.
3. Then, there are loads of rules on how to add subscriptions, how to keep them alive, how to feed the server with publish requests (that the server queues and uses to send responses at the right time), etc. For that, the client library needs to either
- have a second thread to act on incoming events, do keepalives, etc.
- expose the "work loop" as a function that is called regularly by the user.

The server is a lot more difficult, as there is per-session and per-node information that is tightly linked.

Stasik0, 2015-04-25:
Since subscriptions are beyond nano profile, please declare subscriptions to be a dedicated feature that can be turned on and off in cmake. This also related to generated data types.

ichrispa, 2015-04-30:
Greetings. I have been given the all-clear to take two weeks in order to get the subscriptions/monitoring integrated into open62541. So I am basically taking this issue over from @kadomcev.

I will start working at the server first, because (1) I have a client that I can test my mess with and (2) I am more familiar with the code. The plan is to register a linear list of subscriptions and monitoreditems with the server. MonitoredItems will be regularly processed to fullfills sampling and trigger requirements. This also implies at least a simple event subsystem for emitting events.

Since I am about to build linlists into the server "class"/struct, this might also touch #236.

jpfr, 2015-05-03:
top!

ichrispa, 2015-05-05:
The Development Branch now supports rudimentary CreateSubscriptions, DeleteSubscriptions and CreateMonitoredItems Services. They are rudimentary in the sense that all necessary data from these requests is stored with the server and later freed, but no actual subsystem for monitoring exists yet. Subscriptions are registered in the servers `struct SubscriptionManager`, Monitored Items are registered in each subscriptions `struct MonitoredItems`. Both lists are handled as linked lists using either posix queue.h or the queue.h included with the server.

I will check and improve the error-handling, particularly in regard to field initializations. There is currently no support for PublishRequests other then returning `UA_STATUSCODE_BADSERVICEUNSUPPORTED`.

I consider the monitoring/event subsystem and publishing/notification to be a discussion in its own right. I will for now take the liberty to close this issue and will open #245 for followups.

closed 2015-05-05 by ichrispa