open62541/open62541 #206: Question on 'maturity'

PhotoKevin, 2015-03-12:
I'm looking at converting some software from being an OPC DA client to using OPC UA, and am considering using open62541.  Do the developers think this is up to being used in a production environment, or is it still too young? I see the current version is v0.00, but that doesn't mean much in the open source world. 

OK, I realize that's kind of open ended. I'd be looking at pulling a couple hundred data items from a KepWare OPC server, at 20Hz.  I'd also need a call back for a data changed event.

uleon, 2015-03-12:
open62541 is designed to be a research stack. While we address productivity related issues such as low memory footprint, fast reaction time, portability, maintainability and extendability imho an off-the-shelf utilization in production environments is far beyond the scope of open62541.

jpfr, 2015-03-13:
Hey @PhotoKevin 

- We are working towards a first numbered "release". There are still some issues in the conformance testing tools, but they are vanishing fast.
- Concerning code quality, we are in pretty good shape. Unit tests, continuous integration and static code checkers help a lot.

That said, we will not and can not provide the same level of support as a professional outlet.

And: We currently don't provide a client API.
It should be in the "few man days" order of required work, since a lot of the required functionality is already in place. But priorities are currently aligned towards making a release of the server part.

closed 2015-03-17 by jpfr