open62541/open62541 #212: _BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE

l29ah, 2015-04-03:
/usr/include/features.h:148:3: error: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Werror=cpp]

-- glibc-2.20

Stasik0, 2015-04-03:
which system do you use?

l29ah, 2015-04-03:
On Fri, Apr 03, 2015 at 09:17:19AM -0700, Sten wrote:
> which system do you use?

Gentoo Linux.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

Stasik0, 2015-04-03:
does it work now?

l29ah, 2015-04-03:
On Fri, Apr 03, 2015 at 09:44:05AM -0700, Sten wrote:
> does it work now?

open62541/examples/networklayer_tcp.c:424:5: error: implicit declaration of function 'gethostname' [-Werror=implicit-function-declaration]

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

Stasik0, 2015-04-03:
but i do use ```#define __USE_BSD```! Can you look in your `man gethostname` or in the header `unistd.h` itself?

l29ah, 2015-04-03:
On Fri, Apr 03, 2015 at 11:30:27AM -0700, Sten wrote:
> but i do use ```#define __USE_BSD```! Can you look in your `man gethostname` or in the header `unistd.h` itself?

       gethostname():
           Since glibc 2.12: _BSD_SOURCE || _XOPEN_SOURCE >= 500
           || /* Since glibc 2.12: */ _POSIX_C_SOURCE >= 200112L

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

Stasik0, 2015-04-03:
does it help?

l29ah, 2015-04-03:
On Fri, Apr 03, 2015 at 11:49:18AM -0700, Sten wrote:
> does it help?

I've added
#define _XOPEN_SOURCE 500
to the file and it compiled fine. Not sure what's the proper place for it.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

Stasik0, 2015-04-03:
and my changes?

l29ah, 2015-04-03:
On Fri, Apr 03, 2015 at 12:04:44PM -0700, Sten wrote:
> and my changes? 

Nope.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

Stasik0, 2015-04-03:
it contradicts the manpage somehow, okay i will rework your fix

Stasik0, 2015-04-03:
ok, please confirm if it works

l29ah, 2015-04-03:
On Fri, Apr 03, 2015 at 12:10:01PM -0700, Sten wrote:
> ok, please confirm if it works

It works only if i put the declaration above all the includes.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

Stasik0, 2015-04-03:
like this?

l29ah, 2015-04-04:
On Fri, Apr 03, 2015 at 12:15:15PM -0700, Sten wrote:
> like this?

No, above stdlib even.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

Stasik0, 2015-04-05:
@l29ah okay now?

l29ah, 2015-04-06:
On Sun, Apr 05, 2015 at 12:19:02PM -0700, Sten wrote:
> @l29ah okay now?

Yep.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

closed 2015-04-06 by Stasik0