open62541/open62541 #215: definition of UA_QUALIFIEDNAME_ASSIGN is missing

tgeulig, 2015-04-07:
The demo nodeset does not compile:
```
/home/tequila/geulig/exclude/opc_ua/tgeulig/open62541/src/server/ua_server.c: In function ‘UA_Server_new’:
/home/tequila/geulig/exclude/opc_ua/tgeulig/open62541/src/server/ua_server.c:759:6: error: implicit declaration of function ‘UA_QUALIFIEDNAME_ASSIGN’ [-Werror=implicit-function-declaration]
      UA_QUALIFIEDNAME_ASSIGN(myIntegerName, name);
      ^
cc1: all warnings being treated as errors
```

jpfr, 2015-04-07:
Hey @tgeulig, should be fine now.
Thanks for pointing this out.

closed 2015-04-07 by jpfr

Stasik0, 2015-04-07:
I will enable demo nodeset in one of our testbuilds...

tgeulig, 2015-04-07:
FYI There is another occurrence of UA_QUALIFIEDNAME_ASSIGN in examples/server_udp.c.

commit 970ebc00 (2015-04-07): fix #215