open62541/open62541 #219: awaitResponse timeouts are hardcoded in src/client/ua_client.c

l29ah, 2015-04-10:
So it's impossible to use the library over high latency connections like GPRS at the moment.

Stasik0, 2015-04-10:
Good point, so timeouts need to go into the client config!

Stasik0, 2015-04-11:
@l29ah any other points that we might to lift up in the userspace?

Stasik0, 2015-04-13:
@l29ah now it looks like that
```
    UA_Client *client = UA_Client_new();
    client->config.timeout=500;
```
everyone ok with that?

FlorianPalm, 2015-04-13:
Okay, seems to me unsual, I would have expected something like that:
```c
client_config_init(&config); //default values
config.timeout = 100;
config.password = 123;
...
UA_Client_new(&config);
```
At the moment I cannot imagine a case where there should be a problem with your solution.

Stasik0, 2015-04-13:
@FlorianPalm and `UA_Cient_new(NULL)` would be the default constructor, hmm... we need some external referee... @jpfr, @uleon are called ^^

Stasik0, 2015-04-13:
@FlorianPalm i made it like you proposed: https://github.com/acplt/open62541/commit/099c69763488d0f81ca83c61aca5ca40ee01f05a

closed 2015-04-14 by Stasik0