open62541/open62541 #221: Independent memory management for the network layer

jpfr, 2015-04-16:
Hey all,

there's a proposal for independent network layer memory mgmt in the new_tcp branch.

https://github.com/acplt/open62541/commit/1420f81106f486608a8055405027f62b61694b48

Basically, UA_Connection has two new function-pointers to get and release buffers that are managed by the network layer.

That is useful for embedded devices (e.g. based on lwip) and lets us a cyclic buffer (no malloc) for networking. Also we get closer to zero-copy networking for high-performance solutions.

U all okay with this?

Stasik0, 2015-04-16:
yes, we can still point it to malloc and have the old behaviour, yes?

jpfr, 2015-04-16:
that's the idea. :-)

FlorianPalm, 2015-04-16:
yesss

Stasik0, 2015-04-16:
what will be missing for in-situ decode of received binary stuff? I.e. just returning a struct with properly set pointers.

jpfr, 2015-04-16:
Real in-situ decoding won't happen anytime soon.

For example, a UA_ReadRequest looks like this:
```c
typedef struct {
    UA_RequestHeader requestHeader;
    UA_Double maxAge;
    UA_TimestampsToReturn timestampsToReturn;
    UA_Int32 nodesToReadSize;
    UA_ReadValueId *nodesToRead;
} UA_ReadRequest;
```
Each UA_ReadValueId in the array contains a string. They look like this

on the stream: [something, [length, "a long string"]], [something, [length, "another long string"]]
in memory: [something, [length, ptr]], [something, [length, ptr]]

So we cannot jump to the next element with sizeof(UA_ReadValueId), as we would expect in an array.
Hence, we need to create the array in memory and decode into it.

**However**
We can point to the strings inside the received stream without copying.
That holds for all dataTypes where type->zeroCopyable == True. We already have that information for all types.

Besides strings, very few arrays are created during each interaction. Just a few hundred bytes per received message. We could have a second malloc for the *transient* memory used during a single interaction. Basically just a buffer and an offset that we reset at the end of the interaction.

So yeah, we can have a complete request/response cycle with zero mallocs.
But that's still a bit of work.

closed 2015-05-02 by Stasik0