open62541/open62541 #222: Offlay Namespace node creation into an isolated header/c source

ichrispa, 2015-04-16:
Currently the servers namespace is being initialized based on hardcoded node creation in src/server/ua_server.c:UA_Server_new(void), mixing namespace creation up with the actual server init.

I would like to request that the code after line 303 in ua_server.c, beginning at the bootstrapping of the namespace, is removed from Server_new() and isolated in a standalone header/c file. If this namespace is to be automatically generated, this header and c source will be overwritten. The Server_new() function would only call something akin to server_ns0_nodestore_initialize(server) (the function being in a standalone c file described by its own header).

This is only code relocation and should not influence any existing functions or concepts.

jpfr, 2015-04-16:
Very good!

What do we do with the "special" nodes in ns0 that have external datasources? For example, the server-state nodes use datasources with callback functions. Can we give you a list of nodes to be added manually afterwards?

uleon, 2015-04-16:
I'd suggest to not add nodes manually but to have an extendable template or plugin concept that makes it possible to add some target platform specific structures, methods and method calls for mapping the actual data of the nodes to some memory space, to initialize data, to handle read/write-access, a.s.o.. From a tool chain perspective I'd like to sketch the following architekture
```
                                               
 +----------+     .---------.     +-----------+
 |namespace |     |namespace|     |platform   |
 |source    +---->+source file--->+independent|
 |file      |     |parser   |     |ns-model   |
 +----------+     '---------'     +-----------+
                                        |      
  +-------------------------------------+      
  |                                            
  |                                            
  |               .---------.     +-----------+
  +--------------->namespace|     |includable |
 +----------+     |code     +----->namespace  |
 |platform  +----->generator|     |code       |
 |specific  |     '---------'     +-----------+
 |extensions|                                  
 +----------+                                  
                                               
Symbols:                                               
 +------------------+     .-------------------.
 |file or model     |     |process/executable |
 +------------------+     '-------------------'

```

Stasik0, 2015-04-16:
Well the manual addition is a shim at the moment, since the affort of writing a parser was too high.

I thought there is an XML format of node description? Like the format in which ns0 is described?

I propose to write a parser in c, than we have two options:
- parse at runtime or
- parse and create a c file at compile-time and include it.
With a parser in c there is no big difference between executing functions or writing code executing same functions to a c file.

This relates to "namespace code generator" in your diagram.

@uleon: ist there an ascii diagram tool out there? Or did you draw it manually?

uleon, 2015-04-16:
@Stasik0: I once started to write a parser in C which turned out to be an almost unmaintainable piece of software :-1:
Concerning ascii art: have a look at http://asciiflow.com/

Stasik0, 2015-04-16:
Will taking c++ for a parser help? I would really like to have an option to parse at runtimr...

ichrispa, 2015-04-16:
@ jpfr: You can give me an XML description for your nodes.

I have an additional question that goes along with this issue: the parent node is not something opc-ua uses. A node may be referenced by one or more nodes, multiple times if necessary. My compiler currently declares the first node with a forward reference to the node in question as parent. Will that cause an issue?

Writing a namespace parser/compiler/linker for OPC-UA Nodes in python was a pain. I really discourage you from attempting this in C. Generating an initial namespace is a compile time issue from my point of view.

Tools like UA Modeller should be able to export models to XML; I suppose an application could use the same tool to create its nodeset as a header. 

Because my application requires it the script only creates linked models - meaning all referenced nodes must exist in one (or more) XML files read. Invalid references are removed. That is because I need hard address links to nodes in memory, not NodeIDs. This represents a problem if a user only wants to compile his/her nodeset, but not recreate the NS0 nodes. What exactly is the behavior of the open62541 server if a NodeID is assigned multiple times? UA_Server_addNodeWithSession() does not catch this case, so duplicate nodes may exist. If adding a Node with a non-unique ID returns an error, a user could use NS0 along with his own NS in a header without problems and my Namespace-compiler would be happy. Is that viable?

As a signing off issue for today, here is the code dynamically generated for HasFalseSubState (picked at random):

// Node: opcua_node_referenceType_t(i=9005), HasFalseSubState, 
UA_ReferenceTypeNode opcua_node_referencetype_t_i_9005 = UA_ReferenceTypeNode_new();
copyNames((UA_Node*) opcua_node_referencetype_t_i_9005, "HasFalseSubState");
opcua_node_referencetype_t_i_9005->inverseName = UA_LOCALIZEDTEXT_ALLOC("", "IsFalseSubStateOf");
opcua_node_referencetype_t_i_9005->nodeId.identifier.numeric = 9005;
opcua_node_referencetype_t_i_9005->isAbstract = UA_FALSE;
opcua_node_referencetype_t_i_9005->symmetric  = UA_FALSE;
// First referencing node found and declared as parent: i=32/NonHierarchicalReferences using i=45/HasSubtype
UA_Server_addNode(server, (UA_Node*) opcua_node_referencetype_t_i_9005, &UA_EXPANDEDNODEID_NUMERIC(0, 32), &UA_NODEID_NUMERIC(0, 45));

jpfr, 2015-04-16:
@uleon this approach would generate code at compile time?
That's good for embedded targets then.

@Stasik0 Only more powerful boxes (with an operating system) would parse the XML off a file at runtime. Instead of going for C++, why not add a nice'n'small scripting language (lua??) and expose just enough of the C-API to add the nodes...
Having a (functional + garbage-collected) scripting language with good libs makes non-speed-critical runtime work a lot less painful.

Stasik0, 2015-04-16:
@jpfr: right, there is no difference between c++ and lua in this matter

I just suggest to reuse code for compile and runtime parsing. Where runtime parsing implies a powerfull hardware

Stasik0, 2015-04-16:
Just a question  that was raised earliaer... How do we describe nodes with c callbacks? Like the one with the current server time?

jpfr, 2015-04-16:
Yes, that's it. Just give the variablenode a datasource-struct with function pointers instead of a variant.

ichrispa, 2015-04-16:
@jpfr: devices being run without an operating system will not be able to run an opc-ua server for the coming decade due to memory limitations and insufficient processing power. I don't think devices that highly embedded are a viable target for considerations.

jpfr, 2015-04-16:
Our server currently needs 68kb of ROM in the MinSizeRel setting.
RAM can be as little as you want to fit a minimal set of nodes and what the messages themselves need.

So yeah, some 8bit arduino won't run it. Everything 32bit with enough space should work.
Yes, you can run an RTOS or Contiki [1] on these chips. But you don't need an OS per se, nor for open62541 in specific.

[1] http://www.contiki-os.org/

Stasik0, 2015-04-16:
@ichrispa we are currently trying to port it to stm32-discovery...

ichrispa, 2015-04-16:
The STM32 was in "the grey zone" of devices I analyzed in my thesis. It's ambitious but in theory a server with reduced functionality can be implemented if you reuse the TCP Buffer as a message chunk. Otherwise you need 10kB to be conformant with the spec. I'd go with FreeRTOS (or similar), because you can reuse the tcp/ip stack and memory management. But I agree with you in that point, big microcontrollers can run OPC-UA.

When thinking "embedded", I was trending towards PIC18F, AtMega or Big MSP430s.

FlorianPalm, 2015-04-16:
Actually, it is a TI TM4C1294XL, basically the same (Cortex M4, 120 MHz, 1 MB Flash, 256 KByte RAM).
ATMEGA and Co. are way too small, I think. The TCP/IP stack would already almost use the whole memory. As far as I know ARM (M3,M4) processors getting more and more interesting for embedded systems, even for small applications.

ichrispa, 2015-04-17:
I'm happy to report that the generated node model can be loaded by open62541. UAExpert at least can browse the entire Namespace 0. Currently only nodetype and references are created (not data values). References get all attributes set though. 

Registering Datavalues and setting additional node attributes (descriptions and displayname) is still not implemented in the generator.

I would like to request this ticket to be expedided if possible; I currently have my own hack that implements loading the header, but it would be better if someone familiar with the code could restructure ua_server.c:ua_Server_New(). I would suggest a switch as in -DLOADGENERATEDNS as oposed to using the smaller version currently implemented in the function. If the switch is set, a header file will be loaded in ua_server.c:ua_Server_New() instead of the hardcoded init. I'll adapt the script to spit out any functionname you define.

Stasik0, 2015-04-17:
What about just something like this for the beginning? Just as an easy hack?
```
#ifdef GENERATEDNS
#include "your generated code"
#else
[our current code]
#endif
```
is it okay for you?

ichrispa, 2015-04-17:
I'll work it into my fork on monday and request a review/merge.

ichrispa, 2015-04-21:
The hack has been committed along with the script. Both will be activated when `cmake -DLOADGENERATEDNS` is called. Please note that generating the namespace is quite slow and the compilation will hang at ua_server.c for some time.

Several new issues are introduced by using namespace 0 in this manner, most noticably the script will create nodes that are inserted or manipulated by the current server implementation. This will result in an error, since the nodes already exist. So there might be the need to rearrange some of the static nodes created along with the server, such as ServerEndPoints.

closed 2015-04-21 by ichrispa