open62541/open62541 #224: Adding/Removing references reallocates nodes

ichrispa, 2015-04-17:
Nodes are handled structures with a ReferenceCount field and a pointer to the first element to an array of reference structs. 

When calling the  function <code>static UA_StatusCode addOneWayReferenceWithSession</code> in /src/server/ua_server_addressspace.c creates a new node, it:
- first creates a new node that is one reference bigger than the old one using malloc 
- then copies all data from the old node into the new one
- adds the new reference at the end of the reference array
- removes the old node from the nodestore
- inserts the new one into the nodestore

Stasik explained the reason behind this strategy is to allow for atomics. Here the problem I see: Image a namespace where you create a lot of nodes, for example during startup. One Node in Namespace 0 has an average of 3.4 references. Registering a node requires adding a reference to the parrent. It is quite possible the other 2.4 references will be created after their target nodes are created. This results in a scenario where for each new node in a namespace, 3.4 potential other nodes must be deallocated, remalloced and copied.

The "stress" on memory management facilities is huge when doing this. Especially when considering the "gaps" left in the linear address space when deallocating a small node, the memory management facilities will pretty soon be forced to compact memory or "garbage collect" to provide enough free space for the new node instances.

Imagine what this will do to small devices running at ~100-200Mhz with 10-100kB RAM. Could reallocation the strategy be changed to using linear lists for references and other dynamic attributes and still keep multi-threaded implementations in sync?

jpfr, 2015-04-19:
Hey @ichrispa,

you are right. We malloc too often during startup.

You describe the functions in `ua_server_addressspace.c`. Please also have a look at the headers `ua_nodestore.h` and `ua_nodes.h`. Maybe we need to "work around" `ua_server_addressspace.c` and add dedicated mechanisms for startup.

As @Stasik0 mentioned off-issue, nodes are immutable. You can only replace them entirely (with references and everything). That way, we can exchange nodes with a single pointer compare-and-swap [1]. Multiple versions of a node can exist at the same time, if it is replaced but other threads still work on an old version. The old node is garbage collected when the last reader releases it with `UA_Nodestore_release(...)`. The nodestore uses a reference counting mechanism for that.

The node-references are just an array and not a linked list. We could also compare-and-swap the references-array. But since another thread might still be reading it, we can never know when to actually free it. So if you want to add or remove a reference, the entire node needs to be replaced.

That means, we need to copy quite some overhead. But there are no locks involved for concurrent reading and writing. Making copies is a lot cheaper than locking. And since Moores law has ended, going many-core is inevitable.

For constrained devices, we could try to make nodes "perfect" before adding them during startup. But then we need to know beforehand what (and how many) references they will contain. If your script has this information, we can be a lot smarter about this.

[1] https://www.kernel.org/pub/linux/kernel/people/paulmck/perfbook/perfbook.2015.01.31a.pdf

jpfr, 2015-04-19:
That last commit should remove most of the burden for adding references.

jpfr, 2015-04-19:
A running example_server currently needs 35kb RAM. It becomes much more when messages are exchanged. But that's since message buffers are set to the max size of 65kb per default.
Constrained devices with adapted settings should run fine on less than 100kb.

ichrispa, 2015-04-19:
Calling realloc will certainly improve performance. Since this issue is only a question (and it was answered), I think it can be closed.

I attempted to compress the namespace for my thesis using what I consider near optimally coded bytefields. A full namespace 0 will require ~150kB; you can achieve up to 80kB if you use the Burrow-Wheeler compression algorithm on anything stringy or the data payload. So you can expect that you will need some more memory when the actual namespace 0 is generated and added to the server. Plus of course any memory allocated for TCP/IP, Chunk assembly and crypto. 

In essence, you will need 10kB of runtime memory and ~200kB of additional memory for the namespace.

closed 2015-04-19 by ichrispa