open62541/open62541 #226: Howto systematically create variable Nodes with data contents

ichrispa, 2015-04-20:
I need a bit of help with the adaption of my python script that creates Namespace 0 for the open62541 server. All nodes are being created and the script automatically creates a header that is included in UA_Server_New() using the cmake-switch -DLOADGENERATEDNS. I have now arrived at the final piece, which is initializing values in variablenodes.

I need a systematic way to set the Variant field of the UA_VariableNode struct. The example server covers the creation of a new variable node containing a single builtin value. Is there an example for Expansionobjects or Arrays included somewhere?

jpfr, 2015-04-20:
See the function definitions here: https://github.com/acplt/open62541/blob/master/include/ua_types.h#L457

You probably want to use `UA_Variant_setArray` or `UA_Variant_setArray`. The variant struct is easy enough to just set it directly.
The data-pointer points to the array, the datatype points to the correct datatype and the arrayLength is the length... :-)

Normally, you don't want to set ExtensionObjects in the variant. If the datatype is not a builtin-type, the data will be automatically wrapped in an extensionobject. But only during encoding.

ichrispa, 2015-04-20:
So for an array I would create something like UA_Int32[14] and initialize the elements, then use `UA_Variant_setArray`, which just sets the arraysize, datatype and a pointer to the beginning of the array. I then pack the value into the variant. Can I then just point `datatypenode.variant->yourVariant`? That sound easy enough.

How do I initialize an ExtensionObject? Do I create a struct describing it and the set the individual fields, then point `datatypenode.variant->&myStruct`? Since that can't set the DataType field, I know the approach is wrong, but I have no clue as to how to do it otherwise.

jpfr, 2015-04-20:
1) Yes.
2) Why do you need extensionobjects in a variant? They should only be used during encoding and when you get a variant **whose type is unknown**.

ichrispa, 2015-04-21:
1) Works rather well ^^ Numeric and Stringbased types are now encoded for variableNodes.
2) Namespace 0 defines several data structures that are variants. Their type is not unknown - they can be reduced to builtin types using the DataType Nodes. Their structure can be reduces to a series of builtins, meaning that they are encodable. 

VariableNodes have two relevant fields: dataSource and variant. I am declaring static data as defined in the XML descriptions, hence I would propably use the variant as a container.

Here's an example of a ExtensionObject Array encoded in VariableNode i=12169. My Script has identified a list of types (array) as the value and determined the encoding of each member of the ExtensionObjects struct. The first string in each entry is the alias for the field (i.e. what the XML calls it):

```
DBG: opcua_value_t.printOpen62541CCode(): Encountered array of Extensionobject in node i=12169:
['EnumValueType':ExtensionObject([
     'Value':Int64(1), 
     'DisplayName':LocalizedText(['', u'Mandatory']), 
     'Description':LocalizedText(['', u'The BrowseName must appear in all instances of the type.'])
]), 
'EnumValueType':ExtensionObject([
    'Value':Int64(2), 
    'DisplayName':LocalizedText(['', u'Optional']), 
    'Description':LocalizedText(['', u'The BrowseName may appear in an instance of the type.'])
]), 
'EnumValueType':ExtensionObject([
    'Value':Int64(3), 
    'DisplayName':LocalizedText(['', u'Constraint']), 
    'Description':LocalizedText(['', u'The modelling rule defines a constraint and the BrowseName is not used in an instance of the type.'])]
)]
```
While the strategy for printing simpler types is clear, I can't find an example for creating ExtensionObjects (let alone arrays of them).

jpfr, 2015-04-21:
Hey @ichrispa,

Please don't create extensionobjects if the variant content is a "known" datatype.
Just set the pointer, the datatype and the array size. The rest is done automatically during encoding.

If the extensionobject business interests you, read the definition of the variant encoding in part 6 of the standard.

The script is looking good!

jpfr, 2015-04-21:
If you absolutely need an array of extensionobjects, then use the UA_calcSize and UA_encodeBinary functions to write the content into a bytestring (in the extensionobject).
Then set the type id (UA_TYPES[UA_TYPES_INT32].typeId) and the encoding (binary) in the extensionobject

Stasik0, 2015-04-21:
typeId+BINARY_ENCODING_OFFSET i.e. typeId+2 ;)

ichrispa, 2015-04-21:
I'll fiddle with the script a bit at the end of the week and try to get at least scalar versions of extensionobjects encoded. As far as I am concerned there is currently no "easy" way to group variables and then condense them into an expansionobject as with setArrayCopy() or setScalarCopy().

closed 2015-04-21 by ichrispa