open62541/open62541 #227: implicit declaration of htoleXX

LEvertz, 2015-04-23:
On older and some smaller systems (glib < 2.9 or uclibc) the endian functions are not defined in endian.h. This leads to implicit declaration warnigns and undefined references.
As each use of the htoleXX functions is surrounded by an #ifdef clause to determine endianness I suggest to alter this if clause to

#if (! defined (_WIN32)) && ( __BYTE_ORDER != __LITTLE_ENDIAN )

this is merely a work-aournd to get older systems with native little-endian running but it works as __BYTE_ORDER seems to be always defined in endian.h

Stasik0, 2015-04-23:
@all: is there another *defenitive* fix for older glibc that is tolerable with a tolerable work-load? Or does this one fix everything?

LEvertz, 2015-04-23:
A further idea would be to check if these functions (as they are actually macros) are defined and if not define it with the byteswaps (as done in newer endian.h versions) but I don’t know whether the byteswaps may be as well “new” functions

jpfr, 2015-04-23:
Glib 2.9 was released 10 years ago.
@LEvertz Have you run into this problem or are we talking hypothetical?

Other architectures, such as some RTOS, don't have endian.h at all. We could add the possibility for users to define the relevant htoleXX themselves (e.g. in ua_configure.h.in) and include endian.h otherwise.

LEvertz, 2015-04-23:
I ran into it. Actually on a uclibc machine. But the undefined references came up on a glibc-machine with a 2011 OSELAS toolchain as well. The latter is confusing as endian.h has the proper definitions in it. I played around with the _BSD_SOURCE and_GNU_SOURCE defines, but it made no difference. So I ended up with the define workaround here as well.

 

Concerning the other architectures: you are including endian.h anyways. So the workaround makes no difference here right now.

LEvertz, 2015-04-23:
The user defines in configure.h works for me :-)

Stasik0, 2015-04-23:
Id propose than to check the defined in configure.h and run into #error to ask user to define stuff. U all okay with that? Who is implementing?

LEvertz, 2015-04-23:
ok

jpfr, 2015-04-23:
- Let user define htoXX in ua_configure.h.in (just say "define here" in 
a comment)
- Add a check in ua_util.h if we still need endian.h

Btw, should we move ua_configure.h.in to /include?

Stasik0, 2015-04-30:
any updates? @LEvertz could you pull-request your changes?

Stasik0, 2015-05-05:
@LEvertz ping?

LEvertz, 2015-05-05:
Sorry, overlooked the last one.

My changes do not reflect the proposed “define here”. They are just workarounds when the functions are used.

I will try such changes today, if I get to it. Then I will do the pull-request.

 

Von: Sten [mailto:notifications@github.com] 
Gesendet: Dienstag, 5. Mai 2015 12:27
An: acplt/open62541
Cc: Evertz, Lars
Betreff: Re: [open62541] implicit declaration of htoleXX (#227)

 

@LEvertz <https://github.com/LEvertz>  ping?

Stasik0, 2015-05-05:
@LEvertz thank you

closed 2015-05-07 by Stasik0