open62541/open62541 #229: remove useless typedefs

nwmcsween, 2015-04-24:
Looking though the code it seems there are a lot of useless typedefs for transparent structs and such, these make reasoning about the code a pain and are useless.

Stasik0, 2015-04-24:
@nwmcsween could you please show us a particular example? Are you talking about generated types or built-in ones?

nwmcsween, 2015-04-24:
Almost everything is typedefed, even simple things like a const char *
which is a bad code smell.

jpfr, 2015-04-25:
Typedefing is mostly a matter of taste.
The kernel people are against it. But many (modern and non-smelly) projects typedef their structs.
https://github.com/GNOME/gtk/blob/master/gtk/gtkwindow.h
https://github.com/neovim/neovim/blob/master/src/nvim/api/private/defs.h

We actually could write struct everywhere. But I don't think that will impact readability a lot.
`UA_`-prefixed types that are neither integers nor floats are structs. Simple as that.

nwmcsween, 2015-04-25:
It still hides information which makes the reader have to take a level 
of indirection to figure out what it is for no benefit.

ichrispa, 2015-04-25:
On the contrary. The reader will want to create data structures that comply with OPC UA and he won't want to find out how those types got implemented. He will know beforehand that there is a UA_String (which may be encoded in different ways and these encodings are not c99 compliant). He will not want to know how to make this specific library treat a char[] as a OPC UA string.

What I would like to see is reference material that documents how to create and handle the OPC UA types. Methods like stringCopy or variantScalarCopy should be documented alongside the typedefs. But given that the project is in an early phase, this will have to wait a bit.

jpfr, 2015-04-25:
Word

closed 2015-04-25 by jpfr