open62541/open62541 #230: ua_types.h - wtf?

nwmcsween, 2015-04-24:
why in gods name do you typedef already typedefed types by the c library?

Stasik0, 2015-04-24:
do you mean something like this? ```typedef int64_t UA_Int64;```

nwmcsween, 2015-04-24:
Yes it is uneeded as the libc types the primitive types

Stasik0, 2015-04-24:
the reason is that we want to be close to the IEC 62541 standard. There a "UInt32" type is defined and not a "uint32_t".

ichrispa, 2015-04-25:
I perfectly agree that using the standard compliant naming scheme is essential. Lots of the code gets generated by using the standard as a reference. Node values in the namespace generator script for example only need to `print "UA_"+self.getStringName` to create a new UA_Int32, UA_String, UA_Byte, etc.. The script was written completely independantly of open62541 an thanks to the standard compliant naming scheme, this sort of stuff works ^^

closed 2015-04-29 by jpfr