open62541/open62541 #232: More clients to test

Stasik0, 2015-04-27:
Hi,

as a work towards 0.1 i'd like to know which clients are worth testing except UaExpert. At the moment we test our server with UaExpert and our own client.

I'd appreciate a list of (freeware/open) clients that are worth testing or maybe a test report. If you need an actual binary to test - i can provide it on request.

pspeybro, 2015-04-28:
For the other opc server projects I tested (FreeOpcUa and OpenOpcUa) I also tested with Inductive automation Ignition.
It can be downloaded for free to run in demo mode (need to reset every 2 hours) but for the OPC server you can request a free license to run the OPC server permanently without further addons (like the vizalisation suite, alarms, ...).

This OPC server can connect to other OPC servers as a client to link data from other servers and make it available in ignition. 
This is an interesting use case that I tested with the idea of running Ignition as a centralized solution grouping other small remote opc UA servers of other vendors or open source solutions.

However, I guess this requires support for subscriptions.
I think you can also manually refresh the data in the client interface without actually subscribing, so it is still a useful test I think.

HolgerJeromin, 2015-04-28:
Is this helpful? 
http://m.heise.de/developer/meldung/Industrie-4-0-OPC-Unified-Architecture-wird-offener-2617787.html

Stasik0, 2015-04-30:
@pspeybro thank you for the hint

are there some volunteers to test? ;)

pspeybro, 2015-05-07:
I had not tested Ignition yet, I was just mentioning it as a possible "client" to test with.
However, I tried this yesterday and it indeed needs subscriptions to be working. 
A client that is not working is also a good client for testing :-)

Does this (subscription) happen to be on the roadmap somewhere?

pspeybro, 2015-05-07:
https://groups.google.com/forum/#!topic/open62541/aE6eYrtZCog

Stasik0, 2015-05-07:
@pspeybro there is already ongoing work on subscriptions in the dev branch, and e.g. in #245 
@pspeybro softing [x], ua c++ sdk [x]

closed 2015-05-21 by Stasik0