open62541/open62541 #234: Example code using a library, errors

davy7125, 2015-04-28:
Hi,

I tried to use the dll provided in a C# project. I was stuck on an AccessViolationError in the method UA_Client_connect but it may be related to problems with managed / unmanaged code. So finally I'll make a C++ project using the library.

Anyway, here are some changes to do in the example code and in the .h, if someone else try to use the library instead of the .h / .c:

UA_Client *client = UA_Client_new(UA_ClientConfig_standard);
 ==> UA_Client_new has a logger as a second argument
 ==> UA_ClientConfig_standard is not exported with UA_EXPORT and cannot be used (Logger_Stdout_new is also not exported)

UA_StatusCode retval = UA_Client_connect(client, ClientNetworkLayerTCP_connect,
                                             "opc.tcp://localhost:16664");
 ==> ClientNetworkLayerTCP_connect is not exported either

req.nodesToRead = UA_ReadValueId_new();
 ==> a cast "(UA_ReadValueId*)" is missing (some compiler won't accept this)

req.nodesToRead[0].nodeId = UA_NODEID_STRING(1, "the.answer");
 ==> please use macro only inside the library. It triggers errors with C++ compilers and it's not exportable. Wrap the different versions in a function (which has to be exported with UA_EXPORT as well) for a better compatibility.

Regards,
Davy

Stasik0, 2015-04-28:
hi Davy,

the problems you have are with the c++ compiler right?

Best Regards,
Sten

davy7125, 2015-04-28:
There is no problem building the library using a C++ compiler (I tried Visual C++ 2010 & 2013).

What I pointed out is the fact that the dll doesn't comprise some functions present in the example code. To fix this, some UA_EXPORT must be added and some macros should be wrapped within exported functions (UA_NODEID_STRING for instance). By doing so, the dll could be easily used with dll imports in other projects.

Davy

Stasik0, 2015-04-28:
okay, now i understand we do not notice the problem since we do not link the client with the dll in our test builds...

Stasik0, 2015-04-28:
@davy7125 okay, the structs and functions are exposed now, I have added a dynamic linking to the standard test build - we will not miss such an error later
Wrapping macros is still open

davy7125, 2015-04-28:
Thanks a lot!

Regarding the macros, I quickly drafted something like this:

(in .h, important)
UA_NodeId UA_EXPORT GetNodeId(UA_UInt16 namespaceIndex, const char * identifier);

(in .c, you are free)
UA_NodeId GetNodeId(UA_UInt16 namespaceIndex, const char * identifier)
{
	UA_NodeId nodeIdRet;
	nodeIdRet.namespaceIndex = namespaceIndex;
	nodeIdRet.identifierType = UA_NODEIDTYPE_STRING;
	nodeIdRet.identifier.string = UA_STRING(identifier);
	return nodeIdRet;
}

jpfr, 2015-04-28:
Hello Davy,
the changes you proposed have landed in cf33d08b7df06b9a6b57e33bdb03c85e2503ea8d.

Stasik0, 2015-05-02:
@davy7125 is it okay to close?

davy7125, 2015-05-03:
I don't have my computer right now but from what I saw in the commits that sounds good!

Stasik0, 2015-05-03:
@davy7125 okay :) are you interesting in publishing your C# biding together with open62541 when you are done?

davy7125, 2015-05-04:
It didn't work in binding the library directly in a C# project. I had an access memory violation when the library called a method from WinSock.
What I can provide is the class I implemented with C++ / CLI (C++ with a mix of managed / unmanaged for the .NET framework). Here is a link that can describe the process:
http://www.codeproject.com/Articles/651516/Exposing-native-to-managed-Cplusplus-CLI-vs-P-Invo
But I'm still with the RC3 version of the library. What is the tool providing a single .h and .c from the sources?

Regards,
Davy

Stasik0, 2015-05-04:
to obtain single source files you basically need to checkout and to run
`cmake -DENABLE_AMALGAMATION=ON .`

Stasik0, 2015-05-05:
@davy7125 i have just tagged an RC4, you can use the files from it

davy7125, 2015-05-06:
Thank you for your work. I used the .h and .c provided and everything is fine.

Here are the steps to use open62541 with a C# .NET project (I may open a new thread in the forum, or I let you copy / paste these information somewhere else).

# Building open62541 with visual studio

First, I compiled open62541.h and open62541.cpp using Visual C++ 2013, which is compatible with C99. The project I created is a C++ project, excluding the Common Language Runtime Support (no managed code). I linked the project against winsock, which is w2_32.lib on Windows 7.
When the build is done, open62541.dll and open62541.lib are created.

# Creating a C++ / CLI project

Then I created a C++/CLI dll project. I added the .h you provided and linked against open62541.lib previously built. open62541.dll is in the project directory.

Note: for compatibility reason with my other projects I had to use Visual Studio 2010 and C99 is not enabled. I had to change these lines in open62541.h
#include <stdint.h>     =>   #include "stdint.h"
#include <stdbool.h>    =>   #include "stdbool.h"
I manually added the two files "stdint.h" and "stdbool.h" in the project.

I then created classes to interface with the library. Please note that it's in an early stage of the development and that only basic features are provided. I'm using log4net as logger.

## LibOpen62541.h

```C++
#pragma once

#include "Logger.h"
#include <string>
using namespace System;
using namespace log4net;

// Stands as a forward declaration for the opaque struct UA_Client,
// and a hack to get rid of the LNK4248 warnings
struct UA_Client {};

// Managed class (keyword 'ref')
public ref class LibOpen62541
{
public:
  LibOpen62541();
  ~LibOpen62541() { this->!LibOpen62541(); } // The destructor calls the finalizer
  !LibOpen62541(); // Disconnect and free unmanaged resources
  
  // Try to connect to an address (if the connection is not already done)
  // Return true if connected
  bool Connect(String^ hostAddress);

  // Retrieve a value
  // Works with bool, byte, char, (U)Int16/32/64, float, double, String^
  template<typename T> T Get(String^ parameter);
  template<> String^ Get<String^>(String^ parameter);

private:
  template<typename T> int GetType();
  template<> int GetType<bool>()    { return UA_TYPES_BOOLEAN; }
  template<> int GetType<char>()    { return UA_TYPES_SBYTE; }
  template<> int GetType<byte>()    { return UA_TYPES_BYTE; }
  template<> int GetType<Int16>()   { return UA_TYPES_INT16; }
  template<> int GetType<UInt16>()  { return UA_TYPES_UINT16; }
  template<> int GetType<Int32>()   { return UA_TYPES_INT32; }
  template<> int GetType<UInt32>()  { return UA_TYPES_UINT32; }
  template<> int GetType<Int64>()   { return UA_TYPES_INT64; }
  template<> int GetType<UInt64>()  { return UA_TYPES_UINT64; }
  template<> int GetType<float>()   { return UA_TYPES_FLOAT; }
  template<> int GetType<double>()  { return UA_TYPES_DOUBLE; }
  template<> int GetType<String^>() { return UA_TYPES_STRING; }
  static std::string ConvertToStandardString(String^ managedStr);

  ILog^ m_log; // My log4net logger
  Logger * m_logger;
  UA_Client * m_client;
[... 218 lines cut ...]
  IntPtr pointer = GCHandle::ToIntPtr(m_logHandle);
  LOG_HANDLE = pointer.ToPointer();
}

Logger::~Logger()
{
  // Free the handle on the log
  // The object targeted can now be garbage collected if there is no more references to it
  m_logHandle.Free();
}

UA_Logger Logger::GetLogger()
{
  return LOG_LEVEL_1;
}
```

# Usage

Using this library (assembly) in a C# project, you will be able to:
* create a new LibOpen62541(),
* call the method Connect with "opc.tcp://localhost:16664" as argument,
* display the value returned by Get < Int32 > ("the.answer"), which should be 42 if the server provided as example is used.

The dll "open62541.dll" has to be in the project directory, or found via the path.

Stasik0, 2015-05-06:
@davy7125 woooow! that a huge amount of work you have done! Two questions:
* did the supplied libopen62541.dll not work?
* what do you think would be a best practice to spread these files? Via repository, and a project file? Or just copy-pasting it somewhere?

davy7125, 2015-05-06:
Implicit linking on Windows requires a .lib file along with the .dll file (contrary to linux which only needs a .so file). This is explained [here](https://msdn.microsoft.com/en-us/library/253b8k2c.aspx).
Maybe you built the library with mingw? I'm not sure but I think mingw don't need .lib and this might be the cause of the missing .lib.

I didn't try explicit linking with libopen62541.dll since it's a longer method and also because I had a memory violation error in a C# project using it.

I don't know either what is the best way to spread and improve this unaccomplished work. Your role is to provide the library, and every user will then have to integrate it. Several ways for this, several languages... You can't cope with all that stuff. The code I wrote is already specialized by the use of the interface ILog, which is specific to my project.

For sure it has to be kept, but more like a source of inspiration or even a template if it becomes more generic. I'll work on the Get method as you asked.

A little question: is it the same way to retrieve a UA_String than to retrieve a UA_Int32 or any other type?

Stasik0, 2015-05-06:
@davy7125: do yo mean by a readValue? Yes same way. The resulting string is a bit different from cstring its is a struct with a pointer to byte and a length (https://github.com/acplt/open62541/blob/a7482ed0a66ac2f95f7f00b3f4eb1b26a134faed/include/ua_types.h#L113).

Regarding dlls: we do not even use mingw, instead windows binaries are cross-compiled on linux. There should be this lib file included in the releases (libopen62541.dll.a).

Regarding your custom ILog interface, so our way would be to create an own logger interface and probably a reference implementation?

davy7125, 2015-05-06:
Using libopen62541.dll.a resulted in link errors. I'm thus still using the .dll and .lib I built myself.

The code should now be able to query with integers and guids. The namespace index can be changed.
I introduced the interface ILogger, this is now generic and it looks better.

I just saw a possible bug by converting UA_String into String^, I will change the code soon.

Stasik0, 2015-05-21:
I'd like to have an example project file on that under "bindings"

jpfr, 2015-05-21:
parts of the project are nice and compliant.
new bindings will not be. at least initially.
we need to make clear what parts of the project are in what shape.

Stasik0, 2015-05-21:
yea but we need at least some "entry points", maybe spawning an open62541-bindings repository?

ichrispa, 2015-07-24:
We are having some problems getting the c-header def's from open62541 into C# code under MSWin/VS2015. 
@davy7125: Can you give me an example of how you managed that?

jpfr, 2015-10-16:
We now have continuous integration testing for C++.

closed 2015-10-16 by jpfr

klasik16, 2019-03-05:
Hello it is available somewhere Creating a C++ / CLI project for download?