open62541/open62541 #238: Log function callback

davy7125, 2015-04-30:
Dear developers,

I'm in the case where I interface your library with managed .Net code (using C++/cli).
The log callback is currently
"void logCallback(UA_LoggerCategory category, const char *msg, ...)"

The problem I'm facing is the following one: unmanaged functions with variable arguments (which is typically the function I will have to use as a callback for open62512), cannot deal with managed types, such as my class Logger which logs the messages.
To make it simple, a callback with variable arguments is not easy to provide with .Net framework (but not impossible though).

My request would be: is it possible to format the messages within the library before sending them to the log functions?

I know that I'm in a very specific context to use your library, but other users could benefit this improvement in that they only have to care about a char * to display or log.

Regards,
Davy

Stasik0, 2015-04-30:
@davy7125 okay, so we need to add an additional "layer"
@jpfr u okay with this? is it for 0.1 or for later?

jpfr, 2015-05-01:
Hey @davy7125 
you should get something working with the builtin mechanisms.

- Variadic arguments (argument lists) in C++: https://msdn.microsoft.com/en-us/library/9dt1588w.aspx
- The `params` keyword in C#: https://msdn.microsoft.com/en-us/library/w5zay9db.aspx

Let's keep this issue open.
If you have any troubles, we can add some internal printfs.

jpfr, 2015-05-01:
On older MSVC
```c++
String ^FormatAString(String ^format, ...array<Object^> ^args) {
  return String::Format(format, args);
}
```

With C++11 features: http://en.cppreference.com/w/cpp/language/variadic_arguments

uleon, 2015-05-01:
@davy7125 I'm not quite sure how to cope with the UA_Category, however there seems to be an undocumented keyword for the variadic argumentlist: **__arglist**, see e.g. http://www.codeproject.com/Articles/19274/A-printf-implementation-in-C?msg=2840231#xx2840231xx. 
```
[DllImport("open62541.dll", CharSet = CharSet.Ansi, CallingConvention = CallingConvention.Cdecl)]
    private static extern void logCallback(???, String format, __arglist);
...
logCallback(???,"%d,%s", __arglist(i,s));
```

davy7125, 2015-05-02:
Thank you for the links and code you provided. I found a solution so this issue is really minor.

I just thought that if the callback was more "classic", the library would be easier to integrate. Maybe other language such as Python will be impacted too, but not sure.

Regards

closed 2015-05-06 by Stasik0