open62541/open62541 #241: Missing UA_DeleteSubscriptions/MonitoredItemsRequest/Response structs in generated datatypes

ichrispa, 2015-05-04:
Hello all,

the generated datastructures are apparently missing a couple of definitions. Though DeleteSubscription and DeleteMonitoredItems is definetely in the csv schema under tools, the datatypes are not generated by the python script.

If this is a config issue, some advice would be much appreciated. Otherwise I would be very thankfull to the scriptauthor if these datatypes could be generated.

Best regards,

ichrispa

Stasik0, 2015-05-04:
Hi,

you need to generate the types

https://github.com/acplt/open62541/blob/master/tools/generate_datatypes.py

You need to extend the list minimal_types depending on your cmake config.

ichrispa, 2015-05-04:
Perfect, thats it :)

closed 2015-05-04 by ichrispa