open62541/open62541 #245: Event Subsystem and Monitoring

ichrispa, 2015-05-05:
I currently only strife to create a simple Publishing Service, able to summarize changes for the client. OPC UA however has a quite elaborate mechanism for handling sampling/queuing/filtering/triggering in the monitoring service set. I do not see this mechanism being covered by the current implementation.

The easy way to go would be to use multithreading to continuously loop over all monitoredItems, updating values and queues as necessary. We could also create a new queue-based workitem for this task, thus also allowing singlethreaded implementations to process monitoreditems whenever necessary.

The current server is network-event driven and creating a monitoring subsystem will require some fundamental fiddeling with the way work items are currently processed. I would like to set this issue up for discussion to find a principal consencous on how this should be implemented instead of just hacking away at a solution I prefer.

jpfr, 2015-05-05:
We have an elaborate mechanism for *timed function execution*. See `ua_server_worker.c`.
- Callbacks that are executed at a certain `UA_DateTime`
- Callbacks that are executed with a fixed repetition interval
- *Delayed* callbacks that are executed once, when all existing work (queued before) has finished [1]

Do you have subscriptions with pointers (nodeids) to nodes, or do the nodes also point to subscriptions?

[1] To make sure that no worker thread still uses some shared memory location before we free it. This mechanism is used to free UA_Connections in the multithreaded case.

ichrispa, 2015-05-05:
a struct UA_Subscription has a list of UA_MonitoredItems, which in turn point to the nodestore item that they monitor (node *). In essence, one could periodically scan all subscriptions of a server for monitoreditems that generate a notification, then queue them.

timed functions will be ok for things that define a sampleinterval. There is also Sample+Publish, which is perfectly in accordance with the network driven system currently in use. What about event driven trigger, such as OnChange for objects? Would those simply be scanned every n milliseconds? Or should there be an event queue so that listener objects can handle events of a certain type for a monitored item?

jpfr, 2015-05-05:
Guess I will have to read the spec. There's quite a bit of specific wording for subscriptions that I haven't internalized...

Yes, for the interval case, we could just loop over the subscribed values whenever the timer has run out. Is it ok if we just return the current value?

Assume we also have the backwards direction from nodes to subscriptionitems. Then a node change could trigger two types of reaction.
- interval: flag the subscription (and maybe queue the current value). The flag will be picked up when the subscription sample interval times out.
- onevent: send out the notification right away. That means, there is a session->securechannel->connection attached to the subscription. And we need to have a queue of publish requests of which we take the oldest one to send a matching response.

ichrispa, 2015-05-06:
My current understanding of subscriptions is something like this:

[image]

We basically have the asynchronous part that updates the monitoreditem queues. That happens every time the filter applies to an event (for now, let's just consider "Changed"). That causes the new value to be pushed onto the queue.

A subscriptions samples all its monitoredItems at a specific interval. No interval to my understanding means that the queues are read when the request is processed. Each time the subscription reads the queue of its items, it generates a notificationmessage. The client is basically polling the server for these messages. Several of them might accumulate between to polls, so they each have a sequence id. The server then sens the latest sample and notifies the client that there are more notifications available. The client then attaches a list of sequence IDs to its response, indicating which messages he has read.

I will try the following for now: I will create an explicit `monitoreditem_update` function that will force a new value into the queue. Ultimatively, an event should check if this is necessary. The subscriptions can use timedWork to read the queues on a specific sampling interval.

Am I on the right track here or do I have it completely wrong?

jpfr, 2015-05-06:
Yes. That looks like a sensible architecture.

- In my understanding, interval == 0 means that changes are immediately pushed to the client. Otherwise, a simple read request would have the same effect.
- Subscriptions are asynchronous, even if they are piggy-backed onto request/response pairs. The server puts all the publish requests in a queue and sends a matching response later on, when an event has happened. If the pool of publish-requests for a session is empty, then the subscription events are cached and sent out as soon as new publish requests have come in.

Stasik0, 2015-05-06:
How does the subscription system gets notified about a change (I talk about the asynchronous part)? VariableNodes usually have just a void pointer to a memory field that may change anytime. Here only a polling approach will work. 
I'd propose:
* to add an EVENT_ENABLED flag to a VariableNode/Variant  to indicate that the latter is updated *only* using setter functions, if this flag is disabled, polling needs to be used (i believe we need both types)
* to add a function UA_xxx_updateValue(UA_Type type,void* variableNode/variant, void* value) that will write the value and trigger an event flow

ichrispa, 2015-05-07:
We should definetely discuss the mechanism in the conference call. In short, I would disadvise polling and prefer an interrupt/event driven procedure. In essence all access to the nodestore will need to be proxied by access functions that insert an event into a queue. An event handler function should be able to subscribe to an event (or the event calls a handler?) so that it is only executed when it is necessary.

Polling for changes means that at any time, to copies of the nodestore must be present for comparison. And imagine the exampleServer running at 100% cpu load because it is polling for changes...

Stasik0, 2015-05-07:
Imho we need both, since sometimes using setter to trigger events in not possible e.g. when node points to some application memory of a legacy app that uses open62541.

Polling does not imply busy-loops and 100% cpu, I'd rather call it periodic sampling.

ichrispa, 2015-05-15:
What exactly is considered a "changed" value? Does it mean that it is not not hold the same value as before (as in being 0 and then being 23), or that it has been written/altered (as in being 0 and after a write still being 0)? Do we need to detect write access to nodes or is it enough to just parse the values into a bytestring and compare the last sample with the current value?

I would obviously prefer the later ^^

jpfr, 2015-05-18:
Yeah. Updating a value without changing it should be technically notify-able.

If somebody wants to monitor some memory (on changes from outside), he can then manually trigger the per-node "I-have-changed" mechanism to trigger relevant subscriptions in the background.

Stasik0, 2015-05-18:
Cant we use source time stamp for hashing ie to force its update?

mustafacamurli, 2016-06-19:
Hello,

I have just wanted to make couple of comments based on existing monitoring item sampling mechanism. I have seen that a different sample callback is registered for every active monitoring item. As server runs, callbacks are issued one by one from ua_server_worker.c. I supposed, sdk could lose stability if client wants to create big number of monitoring items for one or more subscriptions. 

Sampling interval and publish interval parameters belong to subscription. So in theory if the server can update monitoring item values, status codes and timestamps in one timer which runs in short intervals.

@ichrispa for your question on May 15,

there is different data change filter for monitoring items. So a monitoring item can be considered as changed if any of these files get changed -> value, status, timestamp.

Or you could have a deadband filter for the value. You can consider the value change if it's changed a certain amount. Eg. Deadband absolute.

But first, i suggest to change sampling mechanism to make it more performant.

Mustafa.

jpfr, 2016-06-20:
> sdk could lose stability if client wants to create big number of monitoring items for one or more subscriptions

That's possible. You can define a minimum sampling interval to prevent that. Adding a maximum of monitoreditems per session should be really easy. But let's do that post 0.2.

> Sampling interval and publish interval parameters belong to subscription. So in theory if the server can update monitoring item values, status codes and timestamps in one timer which runs in short intervals.

MonitoredItems can have a different sampling interval than the subscription. So that you get possibly many sample-notifications with every subscription message.

> Or you could have a deadband filter for the value. You can consider the value change if it's changed a certain amount. Eg. Deadband absolute.

That's a feature we definitely want to support.

> But first, i suggest to change sampling mechanism to make it more performant.

Notice the event-driven architecture. We **do not open a separate thread** for each callback.
I don't think we can make the sampling/publishing mechanism fundamentally more performant.

mustafacamurli, 2016-06-20:
Of course, limiting those values always sounds.

For deadband filter support;
Actually, i can add this support. But first I need to change node compare function which now compares encoded bytestrings, if i'm not mistaken.

It's not about opening a separate thread for each callback. My elaborations for the mechanism stood a little poor. I'll write down reasonable points after understanding sdk fully. 
My intention is not make the conflict but for a better arch :)

ichrispa, 2016-06-22:
I would not recommend removing the bytestring comparison, as the majority of types are either structs or non-numerical. I am thinking of composite types in particular.

One could however consider adding deadband as a surplus: if (changed && numerical) {check abs(oldvalue-value) > deadband}.

Pro, 2017-11-10:
As discussed in our last telco, I currently have a student working on the Event Notifications for Objects, especially with the focus on OPC UA Programs (Part 10) and the corresponding StateMachine StateChange Events.
When the feature is at a "good" stage a corresponding PR will be created

AlexSTI, 2018-01-15:
Can I give help? Maybe to speedup the process. In fact, I really need this functionality as soon as possible.

Pro, 2018-01-17:
@AlexSTI yes, definitely. Help is always appreciated. Since I was on holidays the last weeks and could not sync with the student, I'm not up to date with the current status. I will post more info in the following days and we can try to connect you both and distribute the work.

Our general approach is to somehow add a similar concept of ZeroMQ to open62541 which allows to have a pub/sub like structure internally so that objects can publish events within the stack which are cached, and if a client is connected to an event, the corresponding subscribe should read out the object's publish queue and deliver the events to the client.
A corresponding Use-case should then be shown using OPC UA Programs (Spec Part 10) which rely on event monitoring.

Pro, 2018-02-02:
@AlexSTI currently the best option to support us (myself and @aDogCalledSpot) in the development of events is by preparing some unit tests, i.e. you can make yourself familiar with the Events specification and already prepare some unit tests in a separate branch. As soon as the first prototypical implementation for events is available, we can then enable the unit tests step by step and see if both sides understood the specification in the same way. Please let us know if you would be interested in doing that.

AlexSTI, 2018-02-05:
Here I am. I already have a ported library in my project so I will (hopefully) be very quick inserting the new version and test it. If I can help shortening the developement I will run everywere... Best regards

Pro, 2018-02-05:
Great, than you could start with the unit tests. It will help if you separate the unit tests in to small portions in the source code (i.e. different tests in one file) so that we are able to enable them step by step.

AlexSTI, 2018-02-05:
Tell me what I can do and it will be done!

Pro, 2018-02-05:
As a first step you can create unit tests of any functionality you can think of for the events, which are using the client highlevel functions (e.g. as used here https://github.com/open62541/open62541/blob/master/examples/tutorial_client_events.c). That's then independent of our internal implementation/api.

Then just create a new PR which includes the files and reference this PR

AlexSTI, 2018-02-12:
Can I propose a Skype or VSee call in order to sync up?
Best regards

Pro, 2018-02-14:
@AlexSTI i just sent you a corresponding meeting invitation via Mail

Pro, 2018-04-13:
The PR for events implementation is now in #1739

jpfr, 2018-05-26:
#1739 was merged.

closed 2018-05-26 by jpfr