open62541/open62541 #246: Freezing C API

Stasik0, 2015-05-06:
As we observe in #234, there are already people out there relying to our C api.

So the question is hard: what is missing for a 0.1 api freeze, so we can implement it later.
Maybe we should look into existing client APIs?
Client:
* security parameter
* login type: anonymous, password + encryption config, certificate
* some securechannel config?
what else? (this post will be extended)

Server:
* security parameter
what else? (this post will be extended)

jpfr, 2015-05-06:
Additional functions/definitions can be easily added later on without breaking code.

Is there a function that would need to change if we add more functionality? Or are the leaky abstractions that won't hold for a full-featured implementation?

Stasik0, 2015-05-06:
hmm i am not sure, okay, we can probably push all the client stuff into the UA_ClientConfig and UA_ServerConfig

LEvertz, 2015-05-06:
Probably more a point for the 0.2 API freeze but I’ll bring it up anyway:

There are some functions in the “internal” API that need to be made visible when  external nodestores are used.

So these parts of the internal API should be frozen as well sometime (Imho rather sooner than later).

Stasik0, 2015-05-06:
@LEvertz can you point them out if these are a handful?

LEvertz, 2015-05-06:
I’m just going along my compiler-errors here. The listed functions, datatypes and constants are needed if you 

1.       Want to use an external nodestore and

2.       Use an alternative communication-layer.

 

Datatypes:

                UA_Server                                        (for external nodestore)

UA_ExternalNamespace             (for external nodestore)

                UA_ByteStringArray                      (for alternative communication layer)

                UA_TcpMessageHeader             (for alternative communication layer)

UA_TcpHelloMessage                  (for alternative communication layer)

 

Functions / Macros:

                LIST_INIT                                           (for external nodestore) 

 (used to initialize a server structure)

UA_malloc                                         (for external nodestore) (and consequently UA_free) 

 

UA_SecureChannelManager_init           (for alternative communication layer)

UA_SessionManager_init                           (for alternative communication layer)

 

Constants:

                UA_NULL            (OK, rather academic)

Stasik0, 2015-05-06:
Is a forward definition ok, or do you need the whole structure e.g. for UA_Server?

LEvertz, 2015-05-06:
Need the whole structure.

And forget about the byteStringArray. It seems to be completely gone from 
open62541 :-)

Stasik0, 2015-05-06:
@LEvertz damn, it will be hard do freeze it, what particular members do you need?

LEvertz, 2015-05-06:
The UA_Server structure becomes part of a bigger structure that is statically 
allocated. So we need the whole thing. This however could be rebuild on our 
side to use a pointer to a dynamically allocated UA_Server structure.

Currently we initialize the structure ourselves, so we actually set the 
following parts:

    UA_ApplicationDescription description;

    UA_Int32 endpointDescriptionsSize;

    UA_EndpointDescription *endpointDescriptions;

    UA_ByteString serverCertificate;

    UA_SecureChannelManager secureChannelManager;

    UA_SessionManager sessionManager;

    UA_Logger logger;

    UA_NodeStore *nodestore;

    UA_Int32 externalNamespacesSize;

    UA_ExternalNamespace *externalNamespaces;

    UA_Int32 nlsSize;

    UA_ServerNetworkLayer *nls;

    UA_UInt32 random_seed;

Actually we need most of them only to initialize the structure and hand over 
default values. The really interesting thing is the externalNamespace array 
and the network-layer pointer. We do set them to our needs. So a createServer 
function and some accessors would work either.

jpfr, 2015-05-06:
1. The current tcp networklayer uses **only the user-facing API in `/include`**. You don't need access to the internals to build a custom networklayer.
2. We can internally split out initialization from `UA_Server_new` into `UA_Server_init`, that acts on a previously allocated `UA_Server`.
3. Imho, It is unrealistic to freeze the internal API in its current state. That will seriously limit progress and the addition of features. That is also why we can't make the internal headers user-facing, as this implies a freeze. We can however freeze a subset of the internal APIs. What comes to mind is
   - binary encoding
   - the nodestore.

LEvertz, 2015-05-08:
I mixed up some functins in the initial post:

To @jpfr s comments

1.       That’s where I mixed some things up the init-functions for securechannel manager and session manager hve nothing to do with the comm-layer. I needed them to set up the server structure. Hence, I’m complete fine with 2. Maybe then a forward-declaration suffices. (Here it wold be cool, to init the server with user-given parameters, to reconfigure and restart it at runtime. But that’s for the future.)

The UA_TcpMessageHeader is needed as we call UA_TcpMessageHeader_decodeBinary before calling UA_Server_processBinaryMessage (which both are visible).

We need UA_TcpHelloMessage as we call UA_TcpHelloMessage_decodeBinary to find out if the first message in a connection is a valid hello. Again it seems the function is visible.

In both cases a fowrad declaration and _new-function suffice.

2.       As I said. Fine, but parameters for the init would be cool.

3.       That’s why I said it’s more something for the next freeze, in the first place. :-) I highly welcome your suggestion to publish the two mentioned parts.

billran, 2015-05-09:
I am not sure if this is the correct thread or not, but I will post my question anyway.  I want to create a java client using the jnaerator project tooling to create the JNI stub classes.  The generated java classes from the amalgamated header file open62541.h fail to compile since the UA_Server struct that is referenced in several functions (e.g. UA_Server_new) is just declared, not defined.  It is defined in ua_server_internal.h.  

My question is this:   Is this single header intended to be used for both client and server development or were you intending to have separate headers (and libraries) for the client and server sides?

Stasik0, 2015-05-09:
@billran well, we want to keep UA_Server opaque, since it is going to be extended later. So the problem is with the opaque structs, and you probably wont remedy it by a separation of the client and the server code (which is possible, however I'd try not to do it due to administration overhead). So once you get UA_Server removed you will probably hit another opaque struct in the client code... 

The definition of UA_Server is still there, it is just in the amalgamated open62541.c, not in the .h file (so we do not expose it to the user). Not knowing much of jnaerator, maybe you can give it both files? Or maybe there is some way of getting opaque types processed somehow? It is a pretty common technique in the C domain.

Stasik0, 2015-05-21:
@billran hey, i just gave jnaerator a try (on linux)
```$java -jar jnaerator-0.11-shaded.jar -mode StandaloneJar libopen62541.so open62541.h```

gives me 
```#
# SUCCESS: JNAeration completed !
# Output mode is 'StandaloneJar(JAR with bindings and runtime dependencies)
#
# => '/home/user/open62541/./out.jar'
#
```
do you still have the problems?

jpfr, 2015-06-13:
Please open a new issue if necessary. This is all pre-0.1.

closed 2015-06-13 by jpfr