open62541/open62541 #248: Channel Parameter Negotiation in Hello/Acknowledge

kevinherron, 2015-05-06:
It appears that open62541 is not respecting the receiveBufferSize and sendBufferSize in the client's Hello message. Part 6, Section 7.1.2, Table 39 outlines the parameters of the acknowledge message and notes that the server must not set these values to be larger than what the client requested.

I'm sending values of 8196 for both receiveBufferSize and sendBufferSize, with the intention of forcing messages to be chunked. The server is returning an Acknowledge message with 65536 for both sizes (and maxChunkCount=1, maxMessageSize=65536).

I'm not sure if you've intentionally limited the server to a max chunk count of 1, but if so I suggest you rethink that decision. It's understandable to limit the max message size to, say, 65536 (as you're doing), but you should allow that message to be assembled from multiple chunks.

ichrispa, 2015-05-06:
Hi Kevin,

the server does not support chunking yet. So in theory any Buffer value above 8k will result in an error during reception on the client side.

Stasik0, 2015-05-06:
The question is - do we need it for 0.1 release i.e. must we support it or not? Since we pass nano profile tests I suspect that chunking is not mandatory for nano i.e. the server is assumed to be more constrained than thr client.

So we need to look up the standard to decide if chunking is relevant for 0.1 milestone.

jpfr, 2015-05-06:
Imho, the server could very well report his true capabilities, as long as he respects the client's limitations. For that, we store the buffer sizes of server and client separately..
But since the spec says so, I just added the 4 lines of code that do the job.

Since we cap to the minimum of client and server anyway, we can simplify the connection struct have a single connectionconfig (post 0.1).

kevinherron, 2015-05-06:
The profiles don't dictate stack-level parameters like chunk size and chunk count. You can release without chunking support if you want, but you'll be hamstringing yourself if you encounter any "nano-sized" clients who can only send and receive small chunks that are intended to make up a slightly larger max message size.

jpfr, 2015-05-06:
Nano-sized clients will rather increase the max message size than implement chunking.
Since messages need to be cached and copied, chunking does not help at all if RAM is the limiting quantity.

The much bigger problem is that we can't transfer arrays > 65kb even if the client would support so.
So yes, this is definitely on the agenda.

Stasik0, 2015-05-06:
So its post 0.1? How hard is implementing chunks on client?

jpfr, 2015-05-06:
Can't be worse than what we already do to remedy the streamy-ness of TCP.
Yet another layer of caching...

Stasik0, 2015-05-06:
So go for it in 0.1?

Stasik0, 2016-02-09:
chunking is in #492 and PR #534

closed 2016-02-09 by Stasik0