open62541/open62541 #250: Conference Call: Subscriptions Architecture Discussion

jpfr, 2015-05-07:
Related to issue #245.

We have some good ideas for the subscriptions architecture and even the first working code.
Now we need to go and make some profound changes to existing code (e.g. the node objects themselves).

Let's have a conference call @ichrispa @Stasik0 and everybody who is interested.
http://doodle.com/ihazv9diuebnfkhf#table

jpfr, 2015-05-12:
According to the doodle, the telko will be held tomorrow at 10:00.
I will post the access link and phone number in this issue later on.
See you tomorrow!

jpfr, 2015-05-13:
The telko can be accessed at https://meetings.webex.com/collabs/meetings/join?uuid=M05GSUQW672RGMI92HKSL5KDD1-2U7N.

Stasik0, 2015-05-13:
and for the phone communication 
Telefonnummer: +49 241 80 28088 für deutsche Ansage bzw. 28087 für englische Ansage
Konferenz-Nummer:        97128

closed 2015-05-15 by jpfr